hoylabs/OpenDTU-OnBattery #58: Connection to Victron

frommelt5, 2022-11-01:
I know, it's not the right place for a question like that. But can you explain how to connect to a victron system with your software extension? I want to integrate the solar values of the hoymiles inverters into the vrm portal. Is this possible over the air without a cable connection? Do i need further wiring or dongles for the microcontroller?

helgeerbe, 2022-12-05:
@johannesosterberg Sorry for the late response. I use a serial connection (cable) to read the VeDirect protocol from my mppt battery charger.

I use a D1 mini esp32

````
[env:d1 mini esp32]
board = wemos_d1_mini32
build_flags = 
	${env.build_flags}
	-DHOYMILES_PIN_MISO=19
	-DHOYMILES_PIN_MOSI=23
	-DHOYMILES_PIN_SCLK=18
	-DHOYMILES_PIN_IRQ=16
	-DHOYMILES_PIN_CE=17
	-DHOYMILES_PIN_CS=5
	-DVICTRON_PIN_TX=21
	-DVICTRON_PIN_RX=22
````

berni2288, 2023-01-17:
Does the D1 mini esp32 work with 5V? Because the Victron MPPT uses 5V for the Ve.Direct port. This could damage the esp32 because the esp32 works with 3.3V.

Do you use a Logic level converter?

helgeerbe, 2023-01-17:
Hi @berni2288,

my official answer would be use a voltage divider or level shifter for the input pin on the D1 mini32.
Victron MPPT accepts officially  3.3 V as input, so there is no need for level shifting.

But, I don't use a level converter. (https://www.qworqs.com/2021/05/19/are-the-esp32-and-esp8266-5v-tolerant-yes-they-officially-are/). And yes, it seems to work fine for me. This works since more than 6 month perfectly. But I can't tell you something for a longer period first hand. 

If my D1 mini esp32 will die one day, I will replace it and will add a level shifter for input. I simply run out of level shifter, and now I'm too lazy to change something.

berni2288, 2023-01-17:
Thank you! That's interesting.

I tried to use a level shifter yesterday with my Azdelivery NodeMCU ESP32 yesterday, and I didn't get it working.
I've used the PINs 21 and 22 also, I'll try it with other pins the next time.
I'll risk it to connect it without logic level shifter, I have a second ESP32 if the first one breaks :)

berni2288, 2023-01-18:
It's working now without the logic level converter. Nice :)
Now I can enhance my dynamic power limiter feature by using the data from the Victron device also.

I've used the default GPIO21 and GPIO22 pins also. VICTRON_PIN_TX of the esp32 needs to be connected to the RX of the Victron.

The Victron MPPT Ve.direct pin order and voltages are the following:

**GND | RX (3.3V) | TX (5V) | 5V power**

Victron GND --> ESP32 GND
Victron RX --> ESP32 VICTRON_PIN_TX (not sure if this is even needed, as we only receive data
Victron TX --> ESP32 VICTRON_PIN_RX

helgeerbe, 2023-01-20:
Nice. I would always connect RX/TX. Maybe we want to make use of the hex protocol for Ve.direct. For that, you have to send commands to Victron.