hoylabs/OpenDTU-OnBattery #91: [Request] Show Victron UI if VeDirect enabled only

Adminius, 2023-02-03:
### Describe the solution you'd like

Hi, it would make sense to integrate VeDirect to main OpenDTU repository. 
But not every one needs VeDirect. Idee: Disable/Don't show any Victron related UI elements if VE Support is not enabled in the settings.

helgeerbe, 2023-02-03:
Hi, I wouldn't mind to create a PR for OpenDTU, if @tbnobody would accept it. But, this use case is quite special.

Adminius, 2023-02-03:
I would not say "quite special". You are not alone with this combination :)
Next step would be integration of @berni2288 's limiter fork in the main repository. 
It's really strange to work with a fork of fork... because will make another fork from the limiter fork 🤣

helgeerbe, 2023-02-03:
Oh, I'm highly interested in @berni2288's limiter implementation. We can join both repos together, and work as joined forces with the Victron integration. But other might use other products as well. Or we have to think of a general plugin structure, to extend OpenDTU in all ways. At the end it's just the loop part and some UI.

I pushed generell improvements, like the TLS implementation for mqtt, back to the main repro.

Adminius, 2023-02-03:
My initial idea was to use Epever XTRA MPPT Tracker. relatively cheap and with rs485 port. Now I'm using Victron 100/20 48V (it was only 20Euro more expensive...)
general MPPT structure is good approach, but HW should be available for implementation/test purposes, but it is not a case at the moment... 
lets start with small steps: Victron + Limiter + SDM72/630 power meter (I will implement it as an power input option for limiter -> one more fork ;) )

PR #131 (2023-03-09): Hide Victron Ve.direct UI elements from Live page when disabled
Resolves #91
closed 2023-03-09 by helgeerbe