hoylabs/OpenDTU-OnBattery #115: Update Readme

helgeerbe, 2023-02-23:


helgeerbe, 2023-02-23:
- Describe MQTT Messages
- Functional description power limiter
- describe that VeDirect requires cable

berni2288, 2023-03-27:
When I have time I would like to work on the README & documentation.

Should we rename the `README.md` of the parent project OpenDTU to something like `README-OpenDTU.md`?
It would probably make it easier for us because we would be more independent and avoidmerge conflicts with the parent OpenDTU project.

madmartin, 2023-04-02:
> Should we rename the `README.md` of the parent project OpenDTU to something like `README-OpenDTU.md`? It would probably make it easier for us because we would be more independent and avoidmerge conflicts with the parent OpenDTU project.

definitely yes.

MalteSchm, 2023-04-02:
I thought about taking a stab at this (probably over Easter).
IMHO the readme has become too large and should be broken down into individual documents. Can this be done after renaming README.md (to simplify merging)?

I thought about these changes:
- Merge "Features for end users" section with "Extensions to original openDTU" section
- Factor out Dynamic Power Limiter Interface into separate doc
- Merge MQTT information with already existing MQTT document
- Merge "REST-API (/api/verdirectlivedata/status):" into "Web-API Documentation"
- Create Hardware document, move Hardware information there, exend to cover the added peripherals
- Create getting started document, move getting started information there

berni2288, 2023-04-17:
Jesus, our project is growing and we get a lot of ⭐😅
I'll look into writing proper documentation this week and test how the GIT merging logic behaves when we rename the original README.

helgeerbe, 2023-04-17:
Beside readme, would be good if we check 
+ the actions, 
+ introduce our own release number
+ refactor System and About Page (keep the links to the original OpenDTU, but address our extensions).

helgeerbe, 2023-08-31:
Close this, as we moved the doc to the wiki a a living document.

closed 2023-08-31 by helgeerbe