hoylabs/OpenDTU-OnBattery #142: Enable and Disable "Dynamic Power Limiter" via MQTT

Napizaki, 2023-03-25:
### Describe the solution you'd like

It would be great if you could activate and deactivate the "Dynamic Power Limiter" externally via MQTT.
This would enable you to specify other values in Home Assistant, for example, without having to deactivate the DPL via the OpenDTU first.

helgeerbe, 2023-03-27:
Should be doable. 

PowerLimiter setting would be:
- Enable Powerlimiter
- SolarPassthrough enable
- Upper Limit
- Lower Limit
- SolarPasstrhough upper limit (not implemented yet)

berni2288, 2023-03-27:
What would be easier is to allow the updating of single values via the API endpoint /api/powerlimiter/config.
Updating the field "enabled" together with all other fields is already possible.
You can already do POST API requests from Home Assistant to OpenDTU. It's maybe not documented, but you can see the payload in the Browser developer console.

Adminius, 2023-04-12:
Feature Request:
- Upper power limit on battery:
if PV power < "Upper power limit on battery", then inverter limit <= "Upper power limit on battery" and not "Upper power limit"

dirk-schulz, 2023-08-06:
It would be great If the dynamic Power Limiter Config could bei updated via mqtt.
Your solution works absolutely great and I appreciate your greate development!

schlimmchen, 2024-03-18:
I'll close this now as the respective code changes to enable/disable the DPL using the mode switch are already released with 2024.03.07,  see #677. Changing settings other than the thresholds is tracked in #724.

[image]