hoylabs/OpenDTU-OnBattery #143: MQTT connect/disconnect loop in DEV branch

Adminius, 2023-03-25:
### What happened?

installed dev branch from today on factory resetted esp32.
only WiFi and MQTT are set.
MQTT connects and disconnects, see log

Tested with two different MQTT Servers (2x separate RasPis)

### To Reproduce Bug

enable MQTT (with credentials)

### Expected Behavior

MQTT stays connected

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

7d48e42

### Relevant log/trace output

```Shell
--- Terminal on COM6 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
22:03:23.776 > ets Jun  8 2016 00:22:57
22:03:23.776 > 
22:03:23.776 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:03:23.776 > configsip: 0, SPIWP:0xee
22:03:23.776 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:03:23.776 > mode:DIO, clock div:2
22:03:23.776 > load:0x3fff0030,len:1184
22:03:23.776 > load:0x40078000,len:13192
22:03:23.776 > load:0x40080400,len:3028
22:03:23.776 > entry 0x400805e4
22:03:23.776 > E (633) esp_core_dump_flash: No core du=�partition found!
22:03:23.776 > E (633) esp_core_dump_flash: No core dump partition found!
22:03:23.776 >
22:03:23.776 > Starting OpenDTU
22:03:23.776 > Initialize FS... done
22:03:23.776 > Reading configuration... done
22:03:23.776 > Reading PinMapping... [   164][E][vfs_api.cpp:105] open(): /littlefs/pin_mapping.json does not exist, no permits for creation
22:03:23.776 > using default config done
22:03:23.776 > Initialize Network... done
22:03:23.776 > Setting Hostname... Configuring WiFi STA using new credentials... done
22:03:23.776 > Initialize NTP... done
22:03:23.776 > Initialize SunPosition... done
22:03:23.776 > Initialize MqTT... done
22:03:23.776 > Initialize WebApi... done
22:03:23.776 > Initialize Display... done
22:03:23.776 > Check for default DTU serial... done
22:03:23.776 > Initialize Hoymiles interface... Connection successfull
22:03:23.776 >   Setting radio PA level...
22:03:23.776 >   Setting DTU serial...
22:03:23.776 >   Setting poll interval...
22:03:23.776 > done
22:03:23.776 > Initialize ve.direct interface...
22:03:23.776 > ve.direct rx = 16, tx = 17
22:03:23.776 > done
22:03:23.776 > Initialize Pylontech battery interface...
22:03:23.776 > Pylontech Battery rx = 27, tx = 14
22:03:23.776 > done
22:03:23.776 > Switch to WiFi mode
22:03:23.776 > Setting Hostname... done
22:03:23.776 > Configuring WiFi STA using new credentials... WiFi disconnected
22:03:23.776 > Try reconnecting
22:03:23.776 > [  1477][E][WiFiGeneric.cpp:113] set_esp_interface_ip(): Netif Set IP Failed! 0x5007
22:03:23.776 > done
22:03:23.776 > ConfiguringNetwork lost connection
22:03:23.776 >  WiFi STA DHCP IP... done
22:03:23.776 > E (6780) wifi:Association refused temporarily, comeback time 1022 mSec
22:03:23.776 > E (7824) wifi:Set status to INIT
22:03:23.776 > WiFi disconnected
22:03:23.776 > Try reconnecting
22:03:23.776 > Network lost connection
22:03:23.776 > WiFi connected
22:03:23.776 > WiFi got ip: 192.168.178.101
22:03:23.776 > Network connected
22:03:23.776 > Connecting to MQTT...
22:03:23.776 > Disconnected from MQTT.
22:03:23.776 > Disconnect reason:MQTT_NOT_AUTHORIZED
22:03:23.776 > Connecting to MQTT...
22:03:50.612 > Disconnected from MQTT.
22:03:50.614 > Disconnect reason:TCP_DISCONNECTED
22:03:52.611 > Connecting to MQTT...
22:04:22.643 > Disconnected from MQTT.
22:04:22.644 > Disconnect reason:TCP_DISCONNECTED
22:04:24.641 > Connecting to MQTT...
22:04:54.669 > Disconnected from MQTT.
22:04:54.671 > Disconnect reason:TCP_DISCONNECTED
22:04:56.670 > Connecting to MQTT...
22:05:26.695 > Disconnected from MQTT.
22:05:26.702 > Disconnect reason:TCP_DISCONNECTED
22:05:28.699 > Connecting to MQTT...
22:05:58.729 > Disconnected from MQTT.
22:05:58.732 > Disconnect reason:TCP_DISCONNECTED
22:06:00.728 > Connecting to MQTT...
22:06:18.732 > Admin mode disabled
22:06:30.755 > Disconnected from MQTT.
22:06:30.757 > Disconnect reason:TCP_DISCONNECTED
22:06:32.754 > Connecting to MQTT...
22:07:02.785 > Disconnected from MQTT.
22:07:02.785 > Disconnect reason:TCP_DISCONNECTED
22:07:04.770 > Connecting to MQTT...
22:07:34.801 > Disconnected from MQTT.
22:07:34.801 > Disconnect reason:TCP_DISCONNECTED
22:07:36.798 > Connecting to MQTT...
22:08:06.833 > Disconnected from MQTT.
22:08:06.848 > Disconnect reason:TCP_DISCONNECTED
22:08:08.832 > Connecting to MQTT...
22:08:38.864 > Disconnected from MQTT.
22:08:38.879 > Disconnect reason:TCP_DISCONNECTED
22:08:40.874 > Connecting to MQTT...
22:09:10.911 > Disconnected from MQTT.
22:09:10.911 > Disconnect reason:TCP_DISCONNECTED
22:09:12.913 > Connecting to MQTT...
22:09:42.948 > Disconnected from MQTT.
22:09:42.949 > Disconnect reason:TCP_DISCONNECTED
22:09:44.945 > Connecting to MQTT...
```

madmartin, 2023-03-26:
Looking at the log, I see
 
22:03:23.776 > Connecting to MQTT...
22:03:23.776 > Disconnected from MQTT.
22:03:23.776 > Disconnect reason:**MQTT_NOT_AUTHORIZED**

Does your mqtt broker wants username+password set or forces connects with SSL?

Adminius, 2023-03-26:
Both broker are mosquitto on differnet RasPi (one will be moved to a differnt appartment) both needs use+password but not ssl, both are worked with the dev version after my last power meter class PR+your fixes. 
Then I updated to lastest dev version and the problems begun... I tried to factory reset ESP but it doesn't help

madmartin, 2023-03-26:
If I were in your situation, I would "tcpdump" on the RasPi where the mqtt server is. If no SSL is in use, the mqtt communication should be readable and though possible to see what's the reason of "not authorized".

helgeerbe, 2023-03-27:
Can you try the original OpenDTU, do you have the same problem?

helgeerbe, 2023-03-27:
Check this https://github.com/tbnobody/OpenDTU/issues/744#issuecomment-1485271919

Adminius, 2023-03-27:
The fix in original OpenDTU is newer lib version 1.4.1 vs. 1.4.2 with this fix onBattery dev branch doesn't work also.
Original OpenDTU (with the "fix") works whithout problmes:

```
21:58:37.563 > WiFi disconnected
21:58:37.563 > Try reconnecting
21:58:37.563 > Network lost connection
21:58:39.185 > WiFi connected
21:58:39.215 > WiFi got ip: 192.168.178.101
21:58:39.215 > Network connected
21:58:39.215 > Connecting to MQTT...
21:58:39.266 > Connected to MQTT.
21:58:43.532 > Websocket: [/livedata][1] connect
21:58:47.578 > Websocket: [/livedata][1] disconnect
22:01:39.789 > Admin mode disabled
```

Adminius, 2023-03-28:
yes, wifi is stable (2m away from Accesspoint). I can use OpenDTU GUI, also GUI console.

Same HW, same place, with same settings: orinal OpenDTU works, onBattery not more...

helgeerbe, 2023-03-28:
I still see issues regarding MQTT in the original OpenDTU https://github.com/tbnobody/OpenDTU/discussions/750#discussioncomment-5452753

Logs look quite similar. My problem is, that I can not reproduce your issue. My systems connects without any issue. 
And at the moment, I don't have an idea what the problem might be.

Adminius, 2023-03-29:
the 750 issue is solved and was not related to my issue. I'll try to test if my "old" dev branch still working, I had no time last days :/

MalteSchm, 2023-03-29:
The issue is caused by empty MQTT topics in Power Meter
Fix is in: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/151

Adminius, 2023-03-29:
thanks! I can confirm, this solves the problem, but this code is better, because only one MQTT topic is also is valid setting:

```
    if (config.PowerMeter_Enabled && config.PowerMeter_Source == 0){
        if(strlen(config.PowerMeter_MqttTopicPowerMeter1) > 0){
            MqttSettings.subscribe(config.PowerMeter_MqttTopicPowerMeter1, 0, std::bind(&PowerMeterClass::onMqttMessage, this, _1, _2, _3, _4, _5, _6));
        }
        if(strlen(config.PowerMeter_MqttTopicPowerMeter2) > 0){
            MqttSettings.subscribe(config.PowerMeter_MqttTopicPowerMeter2, 0, std::bind(&PowerMeterClass::onMqttMessage, this, _1, _2, _3, _4, _5, _6));
        }
        if(strlen(config.PowerMeter_MqttTopicPowerMeter3) > 0){
            MqttSettings.subscribe(config.PowerMeter_MqttTopicPowerMeter3, 0, std::bind(&PowerMeterClass::onMqttMessage, this, _1, _2, _3, _4, _5, _6));
        }
    }
```

MalteSchm, 2023-03-30:
Yep, if this is the case I agree
I'm not really in the details here

berni2288, 2023-04-02:
@Adminius Do you still experience problems with the latest development changes?
If not I would close this issue :)

Adminius, 2023-04-03:
fixed 
#151

closed 2023-04-03 by Adminius