hoylabs/OpenDTU-OnBattery #147: [Request] Device Manager for serial interfaces

helgeerbe, 2023-03-28:
### Is your feature request related to a problem? Please describe.

There are a couple of devices (power meter, battery charger, bus) which requires serial interfaces. Each device has to be configured at compile time, or separately via the device manger (pin mapping).

I would like to make ist a little bit more manageable and easier to configure.

### Describe the solution you'd like

I propose to create an abstract serial device, which contains all serial interface pins.

- Pin setting is loaded via device map.
- It should only be possible to select two HW (1,2) serial interfaces at max.
- maximum number of interfaces should be defined at compile time

So at the device manger you will get a list of pins for the serial interfaces.

| id  | tx  | rx  | hw  |  label  |
|---|---|---|---|---|
| 0  | 20  | 21  | 1  | victron 1  |
|  1 |  08 |  15 |  0 | victron 2  |
|  ... |   |   |   |   1 |
|  5 |   |   |   |   |

To configure the serial devices you have to choose the interface by id/label to get the pins to initialize the device.

Adminius, 2023-03-28:
Is this really needed?
I mean, some things are known, like most of people will use one Victron only -> do it fix on HW serial x as today, pins are configurable via device manager
SDM/SML power meter, only one at a time, slow (9,6kbps), allways via SW Serial, Pins configurable via devicemanager
UART BMS (not implemented yet), more then one? theoreticaly yes but not sure... but fast (115,2 kbps)-> HW Serial y, pins via device manager

"Power User": more then one Victron: second and further -> sw serial

Thoreticaly case:
One or more Victrons + 2 BMS and + UART Power meter.
in this case we need 2x HW UARTs for BMS and 3x SW Serials and whole new structure for BMS handling -> not supported.
I know, some ready-to-use batteris can be extended, like Pylontech. But Pylontech is CAN based...
Other popular DIY BMS, like JBD/JK/DALI are not capable to talk with each other...

philippsandhaus, 2023-09-21:
`SoftwareSerial` and `HardwareSerial` both derive from `Stream`. I envision a class like `SerialProvider` which provides instances of `Stream` with for example `SerialProvider.getSerial(19200, SERIAL_8N1, rx, tx)`. 'SerialProvider' could then make the decision to hand an instance of `HardwareSerial` or `SoftwareSerial` based on baudrate and/or availability of hardware serial ports. But I guess the tricky parts are the details like how to get notified when a Hardware Serial port is freed up.

_Originally posted by @philippsandhaus in https://github.com/helgeerbe/OpenDTU-OnBattery/issues/452#issuecomment-1729801551_

schlimmchen, 2023-09-27:
Time to share my experience with SoftwareSerial for VE.Direct:

* It works for most messages.
* It works reliably in the sense that it does not break over time.
* Without increasing the default buffer size for SoftwareSerial, most messages cannot be decoded. That's because the VeDirectFrameHandler loop() is called too infrequently. The SoftwareSerial buffer is overrun before it was processed.
* One can increase the buffer size for SoftwareSerial such that a whole message fits into SoftwareSerial's buffer. More precisely: The buffer needs to be so large that a timestamp for every logic level change on the RX line can be recorded for a whole message. That's 10 Bits per character, 4 Bytes (timestamp) per Bit, and more than 200 characters. In my test setup I chose 256 characters. That's 10k of memory. Memory that is exclusively used for this one serial port. That's unacceptable.
* Still, some messages cannot be decoded successfully. My best guess is that in those moments an unsolicited HEX messages followed the unsolicited text message, such that the buffer is once again too small.

@MalteSchm gave me an idea in #454: The memory consumption could probably be drastically reduced if the logic level changes are translated to characters much more frequently. So, I am planning to implement a controller that has its own task that does nothing other than reading bytes from the underlying SoftwareSerial if there are some available, store them in a buffer (that will be read by VeDirectFrameHandler) and yield()s otherwise/afterwards. That way we only need a rather small overall memory footprint, because 40 Bytes of memory are interpreted to one character/byte each time one character/byte is read.

Actually, I have no use for this. My setup includes only one MPPT and the third serial is for the RS485 bus with the JK BMS. I don't need SoftwareSerial -- at least not right now. However, this is a fun exercise and I hope to try it soon.

fellpower, 2024-03-15:
I have the same "issue".

i have 2 victron mppt, 1 victron smart shunt

i put in the pimapping vitcron1 with pins, victron2 with pins - and smartshunt as battery rx/tx.

but the second victron mmpt is not shown.

i cant code, so i dont know, how to bring it in ^^

so such a kind of "Manager" would be nice ^^

schlimmchen, 2024-06-01:
Related to and mostly solved by #1018. PR #1018 is still lacking information for the user other than the lines printed on the serial console, which are hard to see given that they appear only only once on startup. You will have to be connected to the serial console when the device boots to capture them. It would be rather simple to implement a web interface in the status section which shows how the hardware UARTs are spent. It could also display whose requests could not be fulfilled.

I also thought that handing out intermediate classes which could either shim HardwareSerial or SoftwareSerial (and a SoftwareSerial which is serviced by a concurrent task) would be elegant. It seems overkill, though. The system is not that dynamic. This is a configuration problem, not a runtime problem. Instead, I think that components with serial interfaces that allow to use SoftwareSerial (low baud rate) should implement a toggle, allowing the user to specifically select to use SoftwareSerial for that component. Still, it should use one that services the SoftwareSerial in a separate task when this toggle is enabled. When initializing, that component would not ask for a hardware UART at the current SerialPortManager and the port stays free to be used by a component that must use a HardwareSerial.

schlimmchen, 2024-12-01:
This is still open because I think there should be a user interface showing how the hardware UARTs are spent. I recently created an issue to track that, which is why I am closing this request now.

See #1346.

As I don't currently see a need for a SoftwareUart implementation for devices that could probably be serviced by a software-based UART (and I think only the VE.Direct devices remain as options), I am not creating another issue to track the automatic downgrade to SoftwareUart for devices that support it. The ESP32-S3 is the de-facto standard SoC and that provides three freely usable hardware UARTs, which is probably sufficient for most users. As long as users don't ask to connect even more VE.Direct devices, I don't see a reason to put in the effort to implement a SoftwareUart-derived class with its own task and locking etc.

closed 2024-12-01 by schlimmchen