hoylabs/OpenDTU-OnBattery #158: The README is missing safety instructions and disclaimer for working with high Ampere batteries

berni2288, 2023-04-05:
### What happened?

People that use our project are maybe not aware that working with batteries can be highly dangerous and burn down your hose.
Especially when they don't have any knowledge about electrical engineering or electronics.

### To Reproduce Bug

Reproduce: Take a 1,00 mm2 cable and let 30 Ampere flow over it for an hour.

### Expected Behavior

The user should be aware that working with batteries can be highly dangerous. --> Fire

It's absolutely necessary that the README contains safety instructions at the top.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

any

madmartin, 2023-04-19:
Thumbs up for this addition!

I propose to add these links to an excellent e-book from Victron, which gave me VERY valuable information:

https://www.victronenergy.de/upload/documents/The_Wiring_Unlimited_book/43562-Wiring_Unlimited-pdf-en.pdf

https://www.victronenergy.de/upload/documents/The_Wiring_Unlimited_book/43562-Wiring_Unlimited-pdf-de.pdf

maybe more languages are available.

schlimmchen, 2024-12-01:
@madmartin Thanks for linking this guide, which seems to be quite extensive and valuable.

From my perspective, this issue is completed through https://github.com/hoylabs/OpenDTU-OnBattery-docs/commit/482724a34fcf2be08dc272aed96086f09e5d0a95. If something is missing on that new page, let me know.

[image]

closed 2024-12-01 by schlimmchen