hoylabs/OpenDTU-OnBattery #159: Dynamic Power Limiter - Solar Passthrough does not work

MalteSchm, 2023-04-05:
### What happened?

openDTU hangs when Solar Passthrough is enabled with the empty at Night option set. I looked through the code but cannot make much sense out of it.

I think that the code stays in the while(true) loop forever without reaching a return statement. I marked the conditions below that trigger this in my mind (without debugging further)

Now my main reason for this ticket is that I don't understand what the strategies "Empty at Night" and "Empty when Full" are supposed to do. This is also not really clear to me from looking at the code. From my perspective it would make sense to have a strategy that performs solar power passthrough independent of the current battery state and another strategy that does this only when the battery is fully charged.

Can somebody enlighten me?

            ```
case STATE_CONSUME_SOLAR_POWER_ONLY: {
                int32_t newPowerLimit = calcPowerLimit(inverter, true);
                if (isStopThresholdReached(inverter)) {
                    _plState = STATE_OFF;
                    break;
                }
                **if (isStartThresholdReached(inverter)) {
                    _plState = STATE_NORMAL_OPERATION;
                    break;
                }**

                if (!canUseDirectSolarPower()) {
                    if (config.PowerLimiter_BatteryDrainStategy == EMPTY_AT_NIGHT)
                        _plState = STATE_NORMAL_OPERATION;
                    else
                        _plState = STATE_OFF;
                    break;
                }
            case STATE_NORMAL_OPERATION: {
                int32_t newPowerLimit = calcPowerLimit(inverter, false);
                if (isStopThresholdReached(inverter)) {
                    _plState = STATE_OFF;
                    break;
                }
                **if (canUseDirectSolarPower() && (config.PowerLimiter_BatteryDrainStategy == EMPTY_AT_NIGHT)) {
                    _plState = STATE_CONSUME_SOLAR_POWER_ONLY;
                    break;
                }**
```

### To Reproduce Bug

Full battery, solar passthrough enabled with empty at night

### Expected Behavior

it should not hang

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

6f2901b

helgeerbe, 2023-04-05:
Hi @MalteSchm,

check the state diagram at readme. 

Of course there is a return. Otherwise the watchdog would reboot the ESP after a while.

Empty at night -> if there is not enough sun, battery is used until battery is empty or the sun shines again. (This is my default)
Empty when full -> Battery is used only, when the upper load threshold is reached. (@berni2288 is using this)

MalteSchm, 2023-04-05:
Hi @helgeerbe 

Empty at night and Empty when full are only shown in the UI when the "use solar passthrough" switch is flipped. From your response it seems that these options are general options that control the dynamic power limiter. Is this correct?

I know the state diagram but I'm not able to understand this.

helgeerbe, 2023-04-05:
No, these are options only for solar passthrough and describes what should happened if there is not enough sun anymore. Empty at night turns battery on, no matter how full it is. The other options turns the battery only on, when the upper charge threshold is reached.

Your use case is actual not supported. 1. prechare battery until full, 2. solar pathhrough only if battery full, 3. discharge at night? That cost you around 10% energy to charge and discharge the battery. I would prefer to consume directly the energy, and only if I can't consume the produced power, I will charge the battery. When sun is down, I will consume the saved battery power.

MalteSchm, 2023-04-05:
I think we have a terminology problem here. I'll switch to German. That might be easier

Wir haben doch die folgenden Fälle:
- Dynamic Power Limiter off --> Inverter wird gar nicht gesteuert.
- Solar Passthrough "off" --> Was passiert in diesem Fall? 
- Solar Passthrough "on" && Empty at night  --> In diesem Fall wird der Entladezyklus gestartet unabhängig vom Ladezustand der Batterie. Wenn der Inverter läuft wird er entsprechend dem Hausverbrauch gesteuert
- Solar Passthrough "on" && Empty when full --> In diesem Fall wird der Entladezyklus erst gestartet wenn die Batterie den threshold "full" erreicht hat. Wenn der Inverter läuft wird er entsprechend dem Hausverbrauch gesteuert

Solar Passthrough bedeutet für mich das die Leistung der Solarzellen an das Stromnetz durchgereicht wird. Mir scheint hier ist aber eher eine Nulleinspeisung gemeint.

Mein Fall ist eigentlich ganz einfach:
- Solange die Batterie nicht leer ist (*) soll der Inverter immer den Hausverbauch erzeugen.
- Wenn die Batterie voll ist soll der Inverter die volle Leistung aus den Solarzellen einspeisen (über den MPPT)

(*) ich habe noch eine 2. Anlage die den Eigenverbauch abdeckt. Diesen Fall könnte man aber generalisieren: Solange Strom von Solarzellen oder aus der Batterie vorhanden ist soll der Inverter den Hausverbauch abdecken.

MalteSchm, 2023-04-05:
> Ich finde dein Szenario aber ungünstig, da du immer die Ladeverluste hast. Ich bevorzuge den Strom sofort zu verbrauchen und nur dann zu speichern, wenn ich ihn nicht verbrauchen kann. Ich habe noch eine zweite ungeregelte Anlage. D. h. tagsüber wirkt openDTU nur unterstützend, der Großteil der Energie geht in die Batterie. Über Nacht verbrauche ich dann die Batterie. Damit reicht mir eine 2kWh Batterie und ich werde noch 2 weitere Pannels dazu stecken.

Da stimme ich zu. Speichern will ich den Strom nur wenn ich ihn nicht direkt verbrauchen kann. Das bringt mich zu einem weiterem Punkt: Warum haben wir diese Option überhaupt: Solar Passthrough "off" die scheint mir überflüssig zu sein?

Im Grunde ist das doch so das man alles selber verbrauchen will was geht (== Solar Passthrough = on) und nur unter umständen diese Batteriebedingung beachten möchte (also das der neue Zyklus erst losgeht wenn die einmal voll war)

berni2288, 2023-04-05:
> Da stimme ich zu. Speichern will ich den Strom nur wenn ich ihn nicht direkt verbrauchen kann. Das bringt mich zu einem weiterem Punkt: Warum haben wir diese Option überhaupt: Solar Passthrough "off" die scheint mir überflüssig zu sein?

Wahrscheinlich weil das nur funktioniert wenn ein Victron Laderegler an den ESP32 angeschlossen ist? Nur der wird grad unterstützt. Und keine Ahnung, vielleicht gibts andere Gründe warum jemand das nicht haben will... :)

MalteSchm, 2023-04-06:
Verstehe ich nicht.

Wenn kein Victron Laderegler angeschlossen ist funktioniert die Lösung mit Solar-Passthrough doch genauso wie die ohne?

Gab es mal jemand der das machen wollte?

berni2288, 2023-04-06:
> Wenn kein Victron Laderegler angeschlossen ist funktioniert die Lösung mit Solar-Passthrough doch genauso wie die ohne?

Der Power limiter muss wissen wieviel Sonnenleistung in Watt der Laderegler grad reinbekommt damit er das Limit dementsprechend setzen kann. Die Info bekommt er über Ve.Direct.

MalteSchm, 2023-04-06:
Ich versuche zu verstehen welchen Sinn die Einstellung "Solar Passthrough disabled" hat.
Du hast geschrieben "Solar Passthrough enabled" braucht ein Victron Laderegler. Daraus schließe ich: Wenn ich keinen habe muss ich Solar Passthrough disablen. Das widerspricht aber meinem Verständnis, weil ohne die infos von dem Laderegler das quasi automatisch disabled ist. Dann brauche ich da keinen Schalter für

Ich spreche über das:
[image]

Ich habe mal angefangen mich an diese Sache zu geben und den code etwas refactored. Feedback ist willkommen: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/161

helgeerbe, 2023-04-06:
Du kannst auch ein Victron haben, ohne Solar Passthrough nutzen zu wollen. Warum auch immer. Jeder hat da so bestimmte Vorstellungen und Vorlieben.

berni2288, 2023-04-06:
> Du kannst auch ein Victron haben, ohne Solar Passthrough nutzen zu wollen. Warum auch immer. Jeder hat da so bestimmte Vorstellungen und Vorlieben.

Erstens das, und nichts ist für User schlimmer als wenn wieder mal ein neues experimentelles Feature entwickelt wurde und dann funktioniert es nicht richtig und sie können es nicht abschalten.. und müssen auf den Bugfix warten.

MalteSchm, 2023-04-06:
Hm o.k.
Ich schätze da habe ich einfach eine andere Philosophie.

Important takeaways for me are:
- Solar Passthrough is an option to use solar power when available. This is especially relevant in situations where the battery is empty or not yet ready to be discharged (see point below)
- The Empty when full option fully charges the battery before consuming any power
- The Empty at night will discharge irrespective of the SoC
- The usecase I had in mind is not supported

closed 2023-04-06 by MalteSchm