hoylabs/OpenDTU-OnBattery #162: Lint error in PowerMeterAdminView.vue

helgeerbe, 2023-04-06:
### What happened?

Got Lint error in but don't know what's wrong here?

### To Reproduce Bug

See actions in GitHub

### Expected Behavior

no lint error

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

4eec055f84610d00a1c11161cd31297d6871b6fe

### Relevant log/trace output

```Shell
/home/runner/work/OpenDTU-OnBattery/OpenDTU-OnBattery/webapp/src/views/PowerMeterAdminView.vue
[8](https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/4627787519/jobs/8186115950#step:5:9)
  226:34  error  Parsing error: Expression expected
```

berni2288, 2023-04-07:
Looks like a problem with the eslint tool to me, not with PowerMeterAdminView.vue, the syntax should be correct and also compiles fine with `yarn build`.
I guess `yarn build` would throw an error if the expression would be invalid?

```
            testHttpRequestAlert: <{ message: string; type: string; show: boolean; }[]> [ ],
```

When I run `yarn --cwd webapp lint` locally I get this error:

```
WARNING: You are currently running a version of TypeScript which is not officially supported by @typescript-eslint/typescript-estree.

You may find that it works just fine, or you may not.

SUPPORTED TYPESCRIPT VERSIONS: >=3.3.1 <4.8.0

YOUR TYPESCRIPT VERSION: 5.0.2
```

Can we update the eslint maybe?

helgeerbe, 2023-04-11:
> Can we update the eslint maybe?

I think, we should request this in the original OpenDTU?

closed 2023-04-13 by helgeerbe