hoylabs/OpenDTU-OnBattery #163: [Request] Implement safety mechanism to protected the inverter against over current when connecting a 24V battery

berni2288, 2023-04-07:
### Is your feature request related to a problem? Please describe.

When you connected a 24V battery to Hoymiles inverter, you can easily exceed the Ampere limit of 10.5A / 11.5 / 12.25 that is different for Hoymiles models. When you exceed the limit you would run the inverter outside of the specifications of the manufacturer.

Exceeding the limit can cause overheating of the inverter and let the inverter report invalid values because they are not measured correctly anymore. --> Fire risk 🔥

Also See discussion in: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/156

### Describe the solution you'd like

While the user should be aware of this when he connects a 24V battery (documentation needed) and set this `Upper power limit` accordingly, it would also be good if we prevent exceeding the Ampere limit in the code with a safety mechanism.

**Proposal:**
New configuration option: Max Ampere per channel (Default 10.5A).

Formula for maximum power limiter per channel based on P = U * I:

Total channel power limit: **current_voltage x ampere_limit x num_dc_channels**
Example: 24V * 10.5A * 2 = 504W

### Describe alternatives you've considered

What do you think?

swingstate, 2023-04-08:
I also had this idea, however I am still wondering if code in the eeprom/memory of the inverter can act as fast and as early to prevent high current hitting the inverter when a battery is connected and switched on. I assume the limiting and levelling of the inverter happens only when the code is running and I have doubts that this is fast enough to prevent an inrush of high current. 

I decided to go the hardware route and to design a 24V PCB which physically prevents a high current rushing into the inverter. I am using what @madmartin has shared here in #145.

However, I would still support and appreciate if the inverter could limit itself.

berni2288, 2023-04-09:
This issue is NOT about inrush currents when connecting. This issue is about possible permanent over currents with 24V batteries.

helgeerbe, 2023-04-11:
Hm, is this really an issue? Setting my HM-300 to 300W max with a 24 V battery, my inverter limits itself to draw only 10,5 A. So I get something around 252 W max.

berni2288, 2023-04-11:
That's good to hear.
We would need someone with a model that has 4 DC inputs, because I'm not sure if they behave the same way like the 1 DC input models.
I also haven't observed this problem myself, I just got a bit alerted after what @MalteSchm has written in https://github.com/helgeerbe/OpenDTU-OnBattery/pull/156#issuecomment-1494841934.I don't really understand his test results tho. Maybe it's a > HM-1000 specific issue.
Those 4 DC-Input models only have 2 MPPT trackers, even tho they have 4 inputs, that could explain why the inverter does not limit the Ampere on just one input. But we don't know if the inverter takes damage in this case or not.

Maybe it's enough if we just mention this case in our documentation. It's better to connect a battery to all DC inputs anyway.

Manos1966, 2023-05-20:
> We would need someone with a model that has 4 DC inputs, because I'm not sure if they behave the same way like the 1 DC input models. 

I wanted to offer to test it myself, I have just changed from 1x25V LIPo Battery and HM-300 (delivering approx 250W max as mentionned already) to 2x25V Batteries (i.e. 50V) and HM-1500 (just testing posible configurations).

The HM-1500 has 4x11,5A input, working voltage 16V-60V thus, running 11,5A x 50V =  575W on two inputs could give me (2 x 575W) 1150W which would be great!

**QUESTION** regarding the 'Dynamic Power Limiter' setup, what does '**Channel ID**' mean and what do I choose when I have two or more inputs connected to the batteries on the HM-1500?  

Unfortunately I immediately got other problems as explained in a different post [https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/218](url) (Keine Anzeige in der Live Ansicht) so, right now I am struggling to understand what happened.

PS. Thanks for this great version of SW by the way!!!!! I have less than a week experimenting with OpenDTUonBattery. 
I found about it by coincidence, I was on the way to setup the same concept on my Homematic Server (FHEM) when I read about this SW and since I love it :-)

Manos1966, 2023-05-21:
Actually, although my two OpenDTUonBattery are not working, my OpenDTU is.
I can share with you my data:
Hoymiles HM-1500 at 100%
All four inputs connected to a 25,6V 100Ah PowerQueen LiPo Battery
Delivers 903,5W, still far from the 1500W max :-( 
String 1: 240,5W 25V 9,74A
String 2: 242,5W 25V 9,82A
String 3: 234,0W 25V 9,49A
String 4: 234,1W 25V 9,50A
So, it does not seem to me as if there is a problem of the Hoymiles going beyond its 11,5A limit or max. 15A limit.
[image]

schlimmchen, 2024-12-01:
I can see why this issue was opened in the first place. In the past I also argued that using a Hoymiles inverter on a 24V battery is unsafe. However, in practice, there seems to be no safety issues. Since I have been part of the project, I have not heard once that someone fried their inverter because a 24V battery was used. Indeed, the inverter's current sensing capabilities are saturated at some point and the inverter will draw and produce more power than it is able to measure. In reality, that seems to be the only problem.

Introducing some kind of new protection mechanism where OpenDTU-OnBattery observes the inverter's input voltage and current would (1) not work very well as the readings are simply maxed out and do not indicate clearly that there is an overload condition and (2) would cause complaints from users with 24V batteries which are accustomed to their working setup.

For the record: I do **not** endorse the usage of 24V batteries and I think that the user must configure an upper power limit that leads to currents that can for sure be safely handled by the inverter as per its datasheet.

Based on the aforementioned reasons, I am closing this issue as "not planned".

closed 2024-12-01 by schlimmchen