hoylabs/OpenDTU-OnBattery #171: Power Limiter Settings / cannot set permitted current below zero

sir-hackalot, 2023-04-12:
### What happened?

Hi! My configuration is a HM-1500 + Shelly 3EM + OpenDTU-OnBattery Version 616d042.

My real limited power mesured with a Shelly 3EM is always between 20 and 30 Watt, i am not able to come to round about 0 Watt. 

As you can see in screenshot 1 with LiveData, i have a difference between the non persistant power limit and the real value of the produced power of the HM-1500. Maybe the real power is used for the calculation of the new value of the non persistant limit.

Screenshot 2 show the debug console output, with the difference between non persistant power limit and the real value of the produced power.

In Screenshot 3 i tried to set the wanted power to "-30" to shift the offset, but i am not able to set a value below "0".

[image]
[image]
[image]

### To Reproduce Bug

Every time the power limit is set automaticly, the mesured value at the powermeter is 20 to 30 Watts away (to high) from zero. The power output of the Hoymiles Inverter is lower than the limit set.

### Expected Behavior

It would be great to be able to set the wanted power to a value below 0 (for example to -30 Watt), so i can make a manual calibration of false mesured values from the Hoymiles-inverter. My electricity grid operator allows a minimal amount of electricity to be fed into the grid.

Thanks for your great work! I love it! :-)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

616d042

helgeerbe, 2023-04-13:
> Maybe the real power is used for the calculation of the new value of the non persistant limit.

You are right. And I assume there will always be a slight difference between actual power and power limit.

I doubt, that your are able to trim 0. The points of measurements for inverter, power calculation and power meter are not synchronized to the same point in time. There might be 5 or more seconds between those measurements. Are you sure, that your shelly measures correctly? I have a deviation over time with my official power meter.

The wanted power is used to avoid power export on small and fast variations in the power consumption in-between the measurements. I set it to 50.

sir-hackalot, 2023-04-13:
I completely agree with you that it will never be possible to regulate the power consumption exactly to 0 and I am aware that there can be a small deviation between the smart meter and the Shelly 3EM.

Before discovering this great fork project (OpenDTU-OnBattery), I built my own controller with Shelly 3EM, ioBroker, MQTT and OpenDTU to control the inverter. But I like the possibility of this project without an intermediary ioBroker much better.

Initially, I also relied on the power displayed by the Hoymiles Inverter (in the OpenDTU Live display) for my control, but the deviation from the real value on the smart meter was always too large for me. So I subsequently took the "total_power" value from the PowerMeter (Shelly 3EM) and added or subtracted the difference from the zero point to the "limit_nonpersistent_absolute" value in order to use the result to set the new value for the "limit_nonpersistent_absolute".

limit_nonpersistent_absolute_new = limit_nonpersistent_absolute_old +/- total_power

As a result, I am not dependent on the correctness of the inverter values and control the required power using the power measurement on the PowerMeter.

I would be happy if there were the option of being able to set the desired power to a negative value (e.g. -30 watts), since my power grid operator allows a feed-in of up to 800W without a special approval procedure.

Unfortunately, my programming skills are not sufficient to actively participate in the coding of this project. But maybe my feedback will help to expand or improve this solution.

Thank you for the great work on this project!

closed 2023-04-19 by helgeerbe