hoylabs/OpenDTU-OnBattery #174: VE.Direct only receiving data when when MPPT 450/100 not producing power

SMEAC, 2023-04-14:
### What happened?

System configured in evening, VE.Direct cable connected to MPPT 450/100 with standard pin out, directly received data with correct Serial Number, Battery Voltage, Yield today and further.
In the morning, system stopped updating values.  Data Age continually increasing.  Despite any fault finding on cables.  No change.  
System started working again after complete sunset in evening.  Fault reproduced itself following two days.

### To Reproduce Bug

Connect MPPT 450/100 through VE.Direct cable, while system is producing power.

### Expected Behavior

Data should continue to be updated when MPPT producing power.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

1a0f44d

### Relevant log/trace output

```Shell
No output from VE.Direct is given to virtual console  :(
```

### Anything else?

Suspecting data format not handled correctly when certain values != 0.  
Would be good if Virtual Console included Data Stream from VE.Direct similar to Fetch Inverter data.

SMEAC, 2023-04-14:
Additional info: Readout of PID is 0xA110.  Which is not in the list of supported PIDs in VeDirectFrameHandler.cpp.

SMEAC, 2023-04-14:
Now looking at Data through and some sample data shows that the Firmware Tag, is not 'FW' but 'FWE' 
This should be the cause of the problem, but could be added to the code as an acceptable Name for setting the Firmware value in the frame.

Example Data:
22:18:43.274 > 13 10 80 73 68 9 48 88 65 49 49 48 13 10 PID: 0XA110
(Line with serial number removed, decoded fine though)
22:18:43.274 > 70 87 69 9 49 49 49 70 70 13 10 FWE: 111FF
22:18:43.274 > 67 83 9 48 13 10 CS: 0
22:18:43.274 > 69 82 82 9 48 13 10 ERR: 0
22:18:43.274 > 72 49 57 9 52 51 55 56 55 13 10 H19: 43787
22:18:43.274 > 72 50 48 9 49 50 49 51 13 10 H20: 1213
22:18:43.274 > 72 50 49 9 50 54 56 50 13 10 H21: 2682
22:18:43.274 > 72 50 50 9 55 53 51 13 10 H22: 753
22:18:43.274 > 72 50 51 9 51 51 52 54 13 10 H23: 3346
22:18:43.274 > 72 83 68 83 9 57 56 13 10 HSDS: 98
22:18:43.274 > 73 9 48 13 10 I: 0
22:18:43.274 > 80 80 86 9 48 13 10 PPV: 0
22:18:43.274 > 86 9 53

Stream ends here.  Last Name Value, V: 5 not decoded.

When it is light in the morning will send updated data.

SMEAC, 2023-04-15:
It is now 9:46, the data is just over 11500 old, which suggests it stopped working this morning sometime after 6am.  I guess around sun rise.
The data is now not being decoded properly.  Here is a view of the data being streamed:
09:43:56.667 > 58 
09:44:01.849 > 58 65 
09:44:01.908 > 56 70 69 68 48 48 51 49 48 
09:44:01.959 > 48 57 
09:44:07.208 > 13 10 
09:44:12.458 > 13 10 80 73 
09:44:17.660 > 58 65 
09:44:22.922 > 58 
09:44:28.210 > 13 10 
09:44:33.450 > 13 10 
09:44:38.664 > 58 65 
09:44:43.958 > 13 10 80 
09:44:49.255 > 56 50 
09:44:54.463 > 74 86 13 10 SER: HQ2224R2ZJV
09:44:54.463 > 70 
09:44:59.690 > 67 70 
09:45:04.953 > 13 10 80 
09:45:10.205 > 13 
09:45:15.616 > 58 
09:45:20.785 > 41 VE.Frame: [CHECKSUM] Invalid frame
09:45:20.785 > 58 65 66 67 69 68 48 
09:45:26.172 > 58 65 
09:45:26.239 > 52 69 48 51 48 48 48 48 70 
09:45:26.287 > 65 10 
09:45:26.556 > 58 
09:45:26.613 > 65 56 68 69 68 48 48 55 48 49 
09:45:26.662 > 52 52 

The Serial number was the only thing decoded properly.  Also getting the error invalid checksum.

helgeerbe, 2023-04-15:
I don't find a  MPPT 450/100  in the vedirect specs: https://www.victronenergy.com/upload/documents/VE.Direct-Protocol-3.32.pdf

450 V and 100 A? This seems not "Balkonkratwerk"

SMEAC, 2023-04-15:
Why the MPPT 450/100 is not in the VEDirect Specs is a good question.  It definitely has it as a communications standard included.
[https://www.victronenergy.de/upload/documents/Datasheet-SmartSolar-MPPT-RS-DE.pdf](https://www.victronenergy.de/upload/documents/Datasheet-SmartSolar-MPPT-RS-DE.pdf)
And it is still works as part of a "Balkonkraftwerk".  It is only the MPPT Battery charger.  From the Battery I have a Hoymiles HM-600 and give a maximal of 600W into the house net, 24/7.

SMEAC, 2023-04-15:
Not sure if this helps, but in the VE Direct spec document, BlueSolar HEX it is listed:
[https://www.victronenergy.com/upload/documents/BlueSolar-HEX-protocol.pdf](https://www.victronenergy.com/upload/documents/BlueSolar-HEX-protocol.pdf)

helgeerbe, 2023-04-15:
I haven't implemented the HEX protocol. Only Text. 

But you wrote, that you received the correct data?! Have you tried to reboot the esp? Maybe a hickup?

SMEAC, 2023-04-15:
Yes, I have done many reboots, and also updated the firmware a number of times, I have done a few hacks in the source code to get a view of the data being streamed on the virtual console, and it is the text VE.Direct, not a hex version.  So unfortunately nope, no hickup.  What I haven't been able to manage up until now though, is getting a complete look at the input buffer of the hardware serial, I was going to work on that next.  But am still trying to think of reasons that the system works in the night and not in the day, maybe the amount of data coming when the system is working is more, the the MPPT 450/100 having two trackers.  Any hints from the software, where I could temporarily increase the size of the Serial buffer to make sure it is not having any overruns?

berni2288, 2023-04-15:
What power supply are you using for the ESP32?

SMEAC, 2023-04-15:
Extern netzteil, 5V 3Amp Supply.

berni2288, 2023-04-15:
> Extern netzteil, 5V 3Amp Supply through USB input on ESP32

And the nrf24l01+ has a capacitor soldered? I just thought that the current/voltage maybe becomes unstable in the morning.

But a software failure like you said could also be possible of course.

SMEAC, 2023-04-15:
The NFR24L01+ doesn't have any additional external capacitors, but I wouldn'T beleive the problem could be caused by the wirless, the VE.direct is cable connected.  I can of course add one tomorrow and do a few measurements with the oscilloscope to see if there is any problem.

SMEAC, 2023-04-15:
One thought I am having.  If the hex mode is interupting the text mode.  At the end of the HEX Document, it states:
"For firmware version v1.52 and below, when no VE.Direct queries are sent to the device, the 
charger periodically sends human readable (TEXT) data to the serial port. For firmware 
versions v1.53 and above, the charger always periodically sends TEXT data to the serial 
port. See the "VE.Direct Protocol" document for a detailed description of the contents and 
availability of the information. "
At night, this periodially sending text is recieved by the openDTU with no problems.

However, in another section of the document, it says that Asynchronous commands are sent (HEX mode) whenever a change occurs.  When the system is producing power, and there are continual changes to power voltages and currents, the serial connection is almost predominantly HEX.

Going back through some of the logs, I could pull out this section of the data stream:
58 65 56 57 13 10 73 9 53 57 48 48 13 10 80 80 86 9 50 56 69 67 48 48 48 53 58 65 58 50 58 65 58 65 56 68 69 68 48 48 55 54 49 52 52 55 10 57 57 13 10 73 9 54 54 48 48 13 10 80 80 58 65 58 65 58 65 13 10 58 58 65 48 49 48 50 48 48 48 51 52 53 13 10 86 9 53 50 51 56 48 13 10 67 72 69 67 75 83 85 77 9 20

Decoded in char it is:
     :A89
     I	5900
     PPV	28EC0005:A:2:A:A8DED00761447
     99
     I	6600
     PP:A:A:A
     ::A0102000345
     V	52380
     CHECKSUM	'

Where we see a " :A " section, it is an asyncronous message, page from Page 27 of the HEX Spec Doc.

I couldn't think that this is the first device where this could be happening though.

berni2288, 2023-04-15:
Yeah, a software problem sounds very reasonable. Are you using a 3.3V/5V logic level shifter for the Ve.Direct? You could also play around with that (add one or remove it).

SMEAC, 2023-04-16:
Haven't had a chance to play with the logic level shifter yet, will grab one tomorrow.  But on the software decoding front, I am following the decoding of the Data in VeDirectFrameHandler.cpp, and I have got two problems I am following: 

Firstly on the Checksum Problem.
We move into the state RECORD_HEX when we receive a ':' character.  
When that happens, in case RECORD_HEX, we set the _checksum to 0. and then the state back to IDLE.
But what follows is that the rest of the HEX code is still received, and in each following loop, due to the lines:
```
	if (_state != RECORD_HEX) {
		_checksum += inbyte;
	}
```
the checksum continues to increment.
This will not be reset at all, until the next time we receive a checksum name, where we are checking the checksum value, which we now know is false, or another hex receive event, which further means the checksum is wrong.
Do we need to move the 'checksum += inbyte' somehow?  Or maybe staying in the RECORD_HEX state until we receive the '\n' to move the state RECORD_BEGIN.

Secondly, I am a bit confused with what is happening in the textRxEvent Function.

When the system is working fine, the name for example "PID" is decoded fine, and the value is saved.  However, if a HEX receive event occurs shortly before the string receive event, the strcmp does not work properly.  Even though I can follow that the data is received correctly.  Following is a log with some additional comments included:

```
22:36:44.952 > State: 0 - Received: 13 
22:36:44.952 > State: 0 - Received: 10 
22:36:44.952 > State: 1 - Received: 80 
22:36:44.952 > State: 2 - Received: 73 PI
22:36:44.952 > State: 2 - Received: 68 PID
22:36:44.952 > State: 2 - Received: 9 PID
22:36:44.952 > State: 3 - Received: 48 0
22:36:44.952 > State: 3 - Received: 88 0X
22:36:44.952 > State: 3 - Received: 65 0XA
22:36:44.952 > State: 3 - Received: 49 0XA1
22:36:44.952 > State: 3 - Received: 49 0XA11
22:36:44.952 > State: 3 - Received: 48 0XA110
22:36:44.952 > State: 3 - Received: 13 
22:36:44.952 > State: 3 - Received: 10 0XA110
22:36:44.952 > PID: strcmp('PID', name) = 0
22:36:44.952 > PID Recorded with Value: 0XA110
```

Worked fine, but then....

```
22:36:56.106 > State: 5 - Received: 58 Hex Receive Event
22:36:56.107 > State: 0 - Received: 65 State: 0 - Received: 56 State: 0 - Received: 68 State: 0 - Received: 69 State: 0 - Received: 68 State: 0 - Received: 48 State: 0 - Received: 48 State: 0 - Received: 68 State: 0 - Received: 70 State: 0 - Received: 49 State: 0 - Received: 51 State: 0 - Received: 68 State: 0 - Received: 70 State: 0 - Received: 10 22:36:56.990 > State: 5 - Received: 58 Hex Receive Event
22:36:56.990 > State: 0 - Received: 65 State: 0 - Received: 48 State: 0 - Received: 49 State: 0 - Received: 48 State: 0 - Received: 50 State: 0 - Received: 48 State: 0 - Received: 48 State: 0 - Received: 48 State: 0 - Received: 48 State: 0 - Received: 52 State: 0 - Received: 56 State: 0 - Received: 10 State: 5 - Received: 58 Hex Receive Event
22:36:56.990 > State: 0 - Received: 65 State: 0 - Received: 48 State: 0 - Received: 65 State: 0 - Received: 48 State: 0 - Received: 50 State: 022:36:57.524 > State: 1 - Received: 13 State: 2 - Received: 10 
22:36:57.524 > 
22:36:57.524 > State: 2 - Received: 80 
22:36:57.524 > P
22:36:57.524 > State: 2 - Received: 73 
22:36:57.524 > PI
22:36:57.524 > State: 2 - Received: 68 
22:36:57.524 > PID
22:36:57.524 > State: 2 - Received: 9 
22:36:57.524 > PID
22:36:57.524 > State: 3 - Received: 48 0
22:36:57.524 > State: 3 - Received: 88 0X
22:36:57.524 > State: 3 - Received: 65 0XA
22:36:57.524 > State: 3 - Received: 49 0XA1
22:36:57.524 > State: 3 - Received: 49 0XA11
22:36:57.524 > State: 3 - Received: 48 0XA110
22:36:57.524 > State: 3 - Received: 13 State: 3 - Received: 10 0XA110
22:36:57.524 > 
22:36:57.524 > PID: strcmp('PID', name) = 67
22:36:57.524 > No valid Tag received
```

So although, I can see that exactly the same data is received, the simple strcmp is producing the wrong answer.  I cant understand how the strcmp can be giving a value of 67.  You can see I am outputing the content of the name variable after each received value, and there is no additional 'C' there.  

Any suggestions what could be happening here?

helgeerbe, 2023-04-17:
From https://www.victronenergy.com/upload/documents/VE.Direct-Protocol-3.32.pdf chapter introduction:

There are **two** different implementations of the Text-mode and HEX-mode:
**Older implementations**
- On power up, a VE.Direct interface will always be in Text-mode, and continuously transmits all run-time fields. As soon as it receives a valid HEX-message, it will switch to HEX-mode. It will stay in HEX-mode as long as HEX-messages are frequently received. After a product has not received any valid HEX-messages for several seconds, it will switch back to Text-mode and start to auto transmit the run-time fields periodically again. **Some products will send Asynchronous HEX-messages, starting with “:A” and ending with a newline ‘\n’, on their own. These messages can interrupt a regular Text-mode frame.**

So my implementation here is not correct. Sorry for that, but I don't have such a device. I checked some openSource Code referenced by Victron directly: https://github.com/giacinti/VeDirectFrameHandler/blob/master/VeDirectFrameHandler.cpp

For such devices we have to implement hexRxEvent(uint8_t inbyte) reading stream until `‘\n’ and continue with Text mode after that. If you like you could use the linked code, but ignore the Hex Callback Function.

helgeerbe, 2023-04-17:
If you have time, I could do it. Otherwise I would be happy if you can provide a PR

SMEAC, 2023-04-17:
Okay, thanks for the link to the source code.  I should be able to send you a PR, could just take me a while to get it done.  I managed to do a very simple test with some of the code, using the push of the State variable when a HEX message starts, and pop at the end, I do manage to receive the text (almost) properly.  Unfortunately there is still a problem with the checksum, but ignoring that allows me to at least look at the bulk of the data, and I think I should be able to get it to work without a full implementation of the HEX stack.  Will probably have to wait a week or two for my implementation, not sure if you are able to get it implemented quicker than me.  Will try and keep you updated though.

helgeerbe, 2023-04-17:
I created the PR #177 
If you like you can test it, if check out branch 've.direct-hex'
What is the PID of you device? This is missing right now.

helgeerbe, 2023-04-19:
Should be fixed with 84a6ed540dfb693dc541c58c1178ea5dee20e913

SMEAC, 2023-04-19:
Just to confirm, it (pretty well) works with the patch.   I am still occasionally getting a failed checksum.  But it definately now workable giving me some good data.
PID was also correct.  Cheers

helgeerbe, 2023-04-20:
Thanks for the feedback. That you get occacsionally failed checksum reads is quite normal. Depending on the polling time we started to read from the serial interface. If we receive data in the middle of a frame, the first read will fail. Only if you receive multiple checksum failures in a row (same loop), you should check your cable connection (loose/bad connection).