hoylabs/OpenDTU-OnBattery #176: [Request] Control Victron LOAD output via Ve.Direct

Adminius, 2023-04-17:
### Is your feature request related to a problem? Please describe.

As allready known, if connecting Hoymils inverters to battery inrush current can set BMS to "overcurrent protection state".
The simplest solution is to precharge Hoymiles with a resitor in series and after some time connect Hoymiles directly to the battery.
Using big conntactors for this pupose is very inefficient, because of big coil power counsuption

### Describe the solution you'd like

**Given situation:**
BMS is turned off (the reason is mostly not relevant)
Inverter is turned off = not available because there is no DC voltage
BMS can't turn on because of inrush current.

**Pre requirement:**
Victron should work, that means, if there is no BMS Voltage, PV should produce enough power to turn Victron on (downside: we have to wait after sun is rising if BMS turned off in the night)
Connect a diode and a big resistor (for 48V Battery around 30 Ohm, it would limit the current to ~2A that Victron 48V can handle on LOAD output).

**Idea:**
1. Turn on LOAD output on victron: up to 2A@48V are going directly to inverter and it turns on.
2. Inverter is availble, no inrush current any more, BMS can turn on (some BMS do it automaticaly, some must be turned onmanually or via UART/CAN/RS485)
3. BMS turned on = victron sences battery voltage
4. turn off victron load output

helgeerbe, 2023-04-17:
Hm, 
my 100/30 does not have a Load-Output. I alway read this is a problem of **large** inverters. Are HM-300 / HM-600 large inverters? Is there anybody out there, who has this problem with this small inverters and their BMS?

To be on the safe side, you could use something like this: https://www.rec-bms.com/datasheet/UserManualPrecharge.pdf

Adminius, 2023-04-17:
I had this problem with HM600 yesterday. the only solution was: disconnect Hyomiles from the battery, wait until BMS turns on and turingn on the victron (it was at night) and than connecting Hoymiles again.
The reason for BMS "turn off" state was: one cell had more overvoltage state because of high power charging at the evening.

Yor linked solution needs an external contactor -> a lot of power los.
External contactor could be also controled via ESP32 directliy

helgeerbe, 2023-04-17:
> I had this problem with HM600 yesterday. the only solution was: disconnect Hyomiles from the battery, wait until BMS turns on and turing on the victron (it was at night) and than connecting Hoymiles again.

Thanks for letting me know. Hadn't expected that this a real problem.

Switching Victron load on/off will not work. (Maybe with VE.Direct Hex Protocol). Actual Text protocol is read only. And most MPPTs don't have a physical load output. So you might need to use a relais. 

But isn't this a one time task, if you power your system initially on? After startup battery is permanently connected to the inverter. I have a switchable DC circuit breaker, I would put another Circuit breaker in parallel connected to the resistor. So first turn on resistor connected circuit breaker. Wait 1 second. Turn on regular circuit breaker. Turn off resistor connectect circuit breaker.

Adminius, 2023-04-17:
I've found out, that Victron, directly connected to BMS is not a problem.
But it is a problem if Victron+Hoymiles will be connected simultaneously.

I conncected Hoymiles (both inputs in parallel) via DC ciruit breaker to BMS. Victron is always connected to Battery (Victron has it's own fuse)
Yesterday I turned whole system at first time:
1. DC breaker off
2. BMS on -> Victrons turns on (or was already on because of PV voltage)
3. DC breaker on
4. Hoymiles starts without problems
It worked some hours without problems.

After few hours, the battery was full and one cell was "overcharged" -> BMS turns off because of "overcharging protection".
Hoymiles turns off also, capacitors discharged... DC breaker still on.
Victron goes also off becuase it was dark at that time.
Then, the "overcharged" cell was ballanced -> BMS tried to turn on again -> inrush current -> bms turns off (Overdischarge protection)

Also by turning BMS manualy doesn't worked -> Overdischarge error
The only solution was: dc breaker off, BMS on, wait ~ 10 sek until Victron starts -> dc breaker on -> now hoymiles starts and BMS stays on.
It works until now, but I have to do something, because overcharging by some cells is something I can't realy influence

helgeerbe, 2023-04-17:
Hm, sounds more a problem of your victron setting. Victron should never overcharge the battery. If battery becomes full, victron reduces charge power. Check your settings.

Adminius, 2023-04-17:
I have not the best cells, so they are drifting out and one cell becomes "overcharged" (I think limit is set to 3.55V per cell in BMS as "overcharge protection").
Victron is set to 15x3.45V => 51,75 V

madmartin, 2023-04-17:
@Adminius what kind of battery do you use?

Adminius, 2023-04-17:
https://www.aliexpress.com/item/1005004221234736.html
+
JBD 40A SmartBMS

helgeerbe, 2023-04-18:
Hi @Adminius maybe I have a knot in my brain. But If you want to balance your cells, why don't you use the mechanism of the MPPT? If battery becomes full, MPPT is going in state Absorbtion and then Float. During this, BMS should do the cell balancing.

So easiest way is to turn off the inverter and wait for sun. The rest should be handled by your BMS and the MPPT

Adminius, 2023-04-18:
your knot is: you are assuming, that absorbtion state is achieved. that's works as long cell are good and well balanced.
Let's assume: 51,75V is charging voltage: 15S x 3.45V
After that is the battery lalmost fully chargeded => constant voltage
But, if some cells are not well balanced, it could happend, that 1-2 cells are already at 3,65V and others not:
51,75 - 3,65 -3,65 = 44,45 = 13S x 3,41V
That means 13 cells are not fully charged yet, and the constant voltage range is not achieved yet -> up to 20A are allowed (in my case) -> this two "bad" cells will be overcharged.
If I can reduce current (e.g. set Hoymiles limit higher) BMS would have more time to balance this two bad cells.

helgeerbe, 2023-04-18:
But isn't this what the MPPT is doing. You set a balancing voltage, which is slightly higher than the constant voltage. MPPT is charging in balancing state with balancing voltage and reduced current.

In "Expertmodus" you can set:
Zellenausgleich: Strom, Intervall, Stoppmodus und Dauer

Adminius, 2023-04-18:
There are two different scenarios to balance the cells:
- slightliy different voltage, but no problems
- slightliy different voltage, some cells are already charged (more than 3.45V in my case), but the battery pack is not at a "fully charged" voltage (51.75V in my case).

My settings are:
Absorbtion voltage: 51.75V = 15x 3.45V until this voltage there is CC and the voltage raises, after 51.75 is reached the current falls and voltage is constant (CV)
Floating voltage: 50.25V = 15x 3.35V this state is after Absorption voltage is reached and the Absorption time is over (i think I set it to 2h)

My problem is: 51.75 is not reached yet, max current is allowed, CC, voltage is still rising, but some cells are already at 3,65V (=overcharged).
MPPT doesn't know about those "overcharged" cells, because it sees pack voltage only and this is voltage is under 
51.75, MPPT allowes full current -> bad cells get more overcharged -> BMS turns off.

berni2288, 2023-04-18:
An MPPT is used for optimizing the energy gain by optimizing the input voltage, it's not for cell balancing. The MPPT doesn't see the individual cells of a battery pack, therefore it can't know if individual cells are unbalanced.

That's what the BMS is for. ONLY the BMS can balance cells.

Adminius, 2023-04-18:
that is my problem, that s why the idea: 
if I know (from BMS) that some cells are overcharged, I can say to MPPT: less current, please ;)
Setting less charging current is not so easy, but use more current by Hoymiles to achieve the same "less curent into the battery" is relatevely easy

helgeerbe, 2023-04-19:
I don't know how to solve this. Title is control Victron load, this is impossible. So it might be better to start a discussion until we have a valid proposal.

closed 2023-04-19 by helgeerbe