hoylabs/OpenDTU-OnBattery #181: [BUG] Hoymiles doesn't reset "yield today" over the night

Adminius, 2023-04-19:
### What happened?

If Hoymiles is connected to PV, there is no voltage in the night and Hoymiles reboots each night: yield today will be reseted
With the battery connected the Hoymiles does'n reboot at all, so the yield today couns more than "today":

### To Reproduce Bug

Connect Hoymiles to the battery (0 Watt Limit = StandBy is enogh) and wait over the night

### Expected Behavior

yield is reseted every night (if on battery = at 0:00)

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

b00ca02

Adminius, 2023-04-19:
yeah, my Grafana solution is now: last("yieldtoday") - first("yieldtoday")
hm... ok, your right, if Hoymiles is working at night, will not reboot it... hm (my battery was empty last night reboot...)

closed 2023-04-19 by Adminius

helgeerbe, 2023-04-19:
Try something like this:

Leistung = from(bucket:"home_assistant/autogen")
|> range(start: today())
|> filter(fn: (r) => 
  r._measurement == "kWh" and
  (
    r.entity_id == "energy_solar_sum" 
    
  )  and
  r._field == "value"
)
|> spread()