hoylabs/OpenDTU-OnBattery #187: Add Renogy MPPT Support [Request]

Yoeky, 2023-04-26:
### Describe the solution you'd like

It would be great if someone could add support for Renogy MPPTs, just like the Victron support.
They are also connected via serial interface (Modbus) and there is documentation available:
https://github.com/mickwheelz/NodeRenogy
I lack the programming knowledge to do this, but I think it seems not to be too difficult.
I am happy to do the testing if someone can do the programming!

helgeerbe, 2023-04-26:
@Yoeky I don't own such devices. But if someone willing to write the code, I would support it.

Yoeky, 2023-05-01:
If there is anyone with the programming knowledge around who is willing to do this, I would send over a charge controller for a while (even a cheap RENOGY WANDERER PWM should be enough, since the protocol is the same as far as I understand) within Germany (or even in Europe, if shipping is not getting too expensive...) if needed...

ChristianFFB, 2023-05-09:
Hi, maybe it is easier to be able to monitor the battery voltage via mqtt on such devices? I have a Renogy Rover which i have integrated via mqtt, also my Ecoflow Power Station.

Yoeky, 2023-05-09:
How do you get the Rover information to mqtt? with a separate ESP running NodeRenogy and via serial port? 
I'd love to have the logic of adjusting the Hoymiles power limit automated by the ESP32 (solar pass-through), without relying on my home assistant. It is running rock solid, but I just feel that keeping this simple with less components involved makes more sense...

ChristianFFB, 2023-05-09:
> How do you get the Rover information to mqtt? with a separate ESP and via serial port?
> I'd love to have the logic of adjusting the Hoymiles power limit automated by the ESP32 (solar pass-through), without relying on my home assistant. It is running rock solid, but I just feel that keeping this simple with less components involved makes more sense...

via the BT-1 bluetooth dongle & https://github.com/cyrils/renogy-bt1

Yoeky, 2023-05-26:
I do now have the Renogy monitoring running via esphome. It works perfectly and I get all the data in Home assistant.
I'd like to have the inverter go to 100% once the battery is full (sending it to the grid instead of wasting it... Solar Passthrough), could there be an api call added so I can push my solar charger information directly to opendtu-onbattery? Just like the data that can be pulled from the victrons. I do have the numbers on my esphome and could easily do a http request to transfer those datafields via json once every few seconds (mqtt should work, too, but that adds the mqtt server as another component needing to rely on). Or did I miss something and this is still possible? This functionality would make it really easy to use any kind of solar charge controller with opendtu-onbattery, as long as you can somehow get its status and send it to the network.

Another option would be to add more logic within home assistant to detect battery SOC 100%, then disable the the dynamic power limiter on opendtu and always set the power limit to the hoymiles according to the solar power coming in, also taking my power meter into account to decide when to go higher and use the battery, too. This would be reinventing the logic of opendtu-onbattery within home assistant - feels like the wrong way.

Yoeky, 2024-04-16:
I just wanted to bring this topic up again as it seems more people are active in this project by now. Is there anyone around with the programming skills to include existing code to monitor Renogy MPPTs into OpenDTU-OnBattery just like the Victron MPPT functionality?

it is a simple serial protocol and there already is code available to pull the data from the Renogy:
https://github.com/wrybread/ESP32ArduinoRenogy 
https://github.com/mickwheelz/NodeRenogy

I'd love to get that working without using external logic in home assistant to make something like solar passthrough working...

AndreasBoehm, 2025-01-31:
Since release [2025.01.23](https://github.com/hoylabs/OpenDTU-OnBattery/releases/tag/2025.01.23) you can use any solarcharger (MPPT) that is providing data via MQTT with OpenDTU-OnBatter.