hoylabs/OpenDTU-OnBattery #188: [Request] Reset Victron (MPPT) Values to 0 at night if Victron is off

Adminius, 2023-04-26:
### Is your feature request related to a problem? Please describe.

since yesterday Victron was off because of no solar and empty battery (bms turned off)
whole night, MQTT had (retained) H20=1,2 kWh ("kWh today" from yesterday). 
Since this morging around 6:00 Victron is up and running an produced 20+ Wh
Telegraf saved this from MQTT to InfluxDB and Grafana shows 1,2 kWh today (max()). 
last() has some troubles on last days that's why I'm using max().
Nevertheless, 1.2 kWh should be reseted an 0:00 if Victron is not available, so that values does make sense again.

### Describe the solution you'd like

H20 (or better all) should be reseted an 0:00 if Victron is not avaliable (e.g. last seen more then 10(?) minutes), 
so that values does make sense again.
0 Values an 0:00 make more sense than "old" values from the day before.

Squint5471, 2024-09-03:
Can this also be the reason for this problem in energy dashboard of Home Assistant that shows Victron MPPT data?
[image]

chrisBer31, 2024-09-08:
Victron mppt reset itself the value of H20 between midnight and 3 am it is variable. We need to reset by sending a command to the mppt via ve direct protocol.

schlimmchen, 2024-09-08:
This about values retained in OpenDTU-OnBattery internally, even though the charge controller is not sending data for hours. It makes sense that we need to reset them in those cases.

spcqike, 2024-09-20:
> you need to use victron panel yield total instead of victron yield panel day

off topic, but isn't this the way to go for all measurements? i never use daily values, but always total. it's much easier for visualizations and calculations. like, monthly is "last value of month - first value of month" and not "sum off all daily max (or last) values within range X" and you have no or less problems with rolling values / timezones. like some devices reset at 00:00 UTC, others at 00:00 local time and so on.

Squint5471, 2024-09-21:
> @Adminius @Squint5471 @schlimmchen you need to use victron panel yield total instead of victron yield panel day in home assistant in energy module and it solve the issue ![victron_mppt_ha](https://private-user-images.githubusercontent.com/62074043/369384278-426958f0-53bd-4154-97a4-1bc4df664ec5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY5MTIxMDQsIm5iZiI6MTcyNjkxMTgwNCwicGF0aCI6Ii82MjA3NDA0My8zNjkzODQyNzgtNDI2OTU4ZjAtNTNiZC00MTU0LTk3YTQtMWJjNGRmNjY0ZWM1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTIxVDA5NDMyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEwZjI0ZjJkMzY5OTVjY2RjOGQ3MDgxMzM1NWIyZTdkZWY5YTkwOTZjOTM5ZTcxNDFmMzk4OWEyY2M3NmRkZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.mfgZN3jqivndDsmVr7Ndc3SIWAogno1_OrWussI7JBY)

Thanks for the hint, configured it like this now 👍

closed 2025-02-19 by github-actions[bot]