hoylabs/OpenDTU-OnBattery #198: Enhancement - Simplifying the settings menu

swingstate, 2023-04-28:
### What happened?

not a bug - but suggestion to improve/simplify the GUI. 

The Settings menu is a bit redundant & inconsistent in terms of using of the word "settings".

### To Reproduce Bug

click on settings

### Expected Behavior

Since the menu is already named "Settings" the sub items can be listed by their names without showing the word settings. Like for AC Charger or Dynamic Power Limiter. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

7006055

helgeerbe, 2023-04-28:
I agree with you. Should be quite simple to do.

helgeerbe, 2023-04-28:
Should be done with 2d14d9f69e56df98940e87a7d633f5b1f18e198e

closed 2023-04-28 by helgeerbe