hoylabs/OpenDTU-OnBattery #200: Web GUI does no updates

Mogdar-M, 2023-04-28:
### What happened?

I've place the DTU next to the inverter to test if there is a stable wifi connection.

MQTT is enabled and delivers data
[image]
But the Web GUi is not opening
[image]

### To Reproduce Bug

I didn't had any issues testing it on the desk so it might be repated to the low wifi signal.
But my Ahoy DTU is working om the same dictance perfect

### Expected Behavior

GUI available

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

newest build from yeterday

Mogdar-M, 2023-04-28:
changed wifi network --> no change
changed from powerbank to AV adapter --> no change
[image]
I guess the issue is that it is turning on and off

helgeerbe, 2023-04-28:
What happens if you call the live data api.

Try: `http://opendtu/api/livedata/status` Replace opendtu with your device name or ip address.

Do you see the json document `{"inverters":[{"serial":"xyzzy","name":"Solar","data_age":7,"p.....`? or just null?

If you see null, there might be a memory fragmentation and openDTU was not capable to allocate a big enough memory block. 
Try a reboot, and check if the livedata is shown. 

I know the problem, but I don't have a solution for that. This has to be changed in the original openDTU, but this would be require a lot of work. Sometimes it's just a temporary problem (if you use e. g. prometheus api in parallel).

Mogdar-M, 2023-04-28:
finally managed to turn off MQTT and now everything is working again.
I've placed the DTU next to the inverter and I'm able to change configuration and save it.

So the issue is MQTT

Mogdar-M, 2023-04-28:
[image]

helgeerbe, 2023-04-28:
Hm, have you enabled any onBattery staff like veredirect.....? What happens if you disable everything related to onBattery. Does MQTT work?

Mogdar-M, 2023-04-28:
no, it was the lone ESP board, not even an inverter configured.

So I just got status, IP, uptime, rssi und hostname via MQTT

helgeerbe, 2023-04-28:
Maybe related to #203 

Does the esp reboot?

Mogdar-M, 2023-04-28:
Nö,

Der ESP lief ja weiter und hat auch weiter Daten geliefert.
Allerdings ja nur rudimentäre Daten, da ja nur der nackte ESP lief.

Sobald ich MQTT abschalten geht die GUi wieder.

Ja, geht an IO Broker, ich Check die Einstellung, ob ich da was anpassen kann

closed 2023-04-29 by helgeerbe