hoylabs/OpenDTU-OnBattery #203: MQTT messages for disabled components crashes openDTU-onbattery

helgeerbe, 2023-04-28:
OK, ich glaube zu verstehen, warum der ESP abstürzt, wenn er die Nachrichten für den Huawei empfängt. 

Ich vermute du hast für MQTT das Retain flag gesetzt? OpenDTU subscribt sich auf das topic und anscheinend legt der IO Broker das an. Da das retain flag gesetzt ist, sendet er komischerweise 0 auf dem topic (das macht anscheinend Mosquitto nicht???).

Da jetzt eine Nachricht empfangen wird, versucht der ESP den Wert auf eine nicht initialisierte Struktur zu setzen.

Lösung sollte relativ einfach sein. Alle Nachrichten ignorieren, wenn die Komponente disabled ist.

_Originally posted by @helgeerbe in https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/183#discussioncomment-5756259_

helgeerbe, 2023-04-28:
5d4c6866da968bda164ba99eb18bc1c8266bbcfd 
Ignore messages for Huawei and PowerLimiter if disabled

Waiting for feedback

Mogdar-M, 2023-04-28:
also bei mir ist in IO Broker kein Retain Flag gesetzt

helgeerbe, 2023-04-28:
OK, danke. Dann ist es bei dir was anderes.

Mogdar-M, 2023-04-28:
[image]
zumindest wurden diese Werte angelegt, obwohl die gar nicht konfiguriert und aktiviert sind

peff74, 2023-04-28:
> [5d4c686](https://github.com/helgeerbe/OpenDTU-OnBattery/commit/5d4c6866da968bda164ba99eb18bc1c8266bbcfd) Ignore messages for Huawei and PowerLimiter if disabled
> 
> Waiting for feedback

Wie flashe ich das?

helgeerbe, 2023-04-28:
Schau mal in die Actions in Github. z. B. hier https://github.com/helgeerbe/OpenDTU-OnBattery/actions/runs/4833525416.
Unter Artifacts das Zip-File runterladen. Entpacken und über Firware-Update in der openDTU das File opendtu-onbattery-generic.bin hochladen.

peff74, 2023-04-29:
Läuft!!!
WT32-ETH01 mit nRF24, Powermeter über HTTP, Dynamic Powerlimiter, MQTT und ioBroker

Mogdar-M, 2023-04-29:
Ich habe das Update auch eingespielt und habe nun auf dem Schreibtisch auch bereits 4 Minuten Uptime ohne Issue, MQTT aktiv in IO Broker.

Scheint also auch mein Problem gelöst zu haben

PR #204 (2023-04-28): fix: ignore MQTT messages for disabled components #203
fix #203
closed 2023-04-29 by helgeerbe