hoylabs/OpenDTU-OnBattery #215: [Request] Ausgang bei bestimmter Batterie Spannung schalten

eeprom23, 2023-05-01:
### Is your feature request related to a problem? Please describe.

Hallo, 

ersteinmal vielen Dank für das tolle opendDTU on Battery Projekt. Ist es möglich einen Ausgang am esp zu schalten und damit in Folge ein Relais wenn eine bestimmte Spannung vom mppt unter bzw überschritten ist ?

### Describe the solution you'd like

Menü mit Option für Ausgang ein bei Spannung > xxV Ausgang aus bei <xxV 

Ähnlich dem virtuellen lastausgang am victron mppt !

swingstate, 2023-05-01:
Damit könnte man dann auch ggs. ein 220V relay schalten, welches den WR AC Seitig (not)abschaltet, um die Batterie nicht leer zu saugen. 
Habe so etwas momentan bei mir laufen, allerdings noch ohne DTU/ESP integration, also ziemlich "dumm".

spcqike, 2023-09-07:
Was soll denn damit am Ende geschalten werden? LOW aktiv oder HIGH aktiv? Soll es eine Sicherheitsschaltung werden?

> wenn eine bestimmte Spannung vom mppt unter bzw überschritten ist

Solarspannung oder Batteriespannung? 

Im Code ist bestimmt noch Platz, aber ob noch genug freie Pins da sind, kann ich grad nicht beurteilen. 

Je nach konkretem Ziel würde ich es wohl aber mit einem zusätzlichen Gerät lösen und mich nicht auf die DTU verlassen.

alexseuf, 2024-01-16:
Wenn du nur einen 230v Ausgang abhängig vom Batterie SoC schalten willst, empfehle ich dir einen Shelly. Über die  Shelly intern laufenden scripts kannst du mittels mqtt  (evtl geht auch http get)  den 230 v Ausgang vom Shelly schalten.

stefan123t, 2024-10-29:
Ähnlicher Request hier auf der [AhoyDTU Wishlist](https://github.com/lumapu/ahoy/issues/1199#issuecomment-1873326989), mit einem Schaltplan / PSPICE mit Optokoppler und Solid State Relay:

> Wunsch: Sunset und Sunrise inkl. einstellbarem Offset an GPIO33 als low/high oder anderem PIN ausgeben. 
> Bei Zeiten zwischen Sunrise +/- Offset und Sunset +/- Offset = `low`,
> Bei Zeiten zwischen Sunset +/- Offset und Sunrise +/- Offset = `high`, 
> Diese astronomische Uhr möchte ich als Indikator für eine Grundlast-Nachteinspeisung nehmen.
> Das Thema kommt ohnehin zunehmend. Wäre für mich ein Grund von openDTU hier her zu wechseln.
> 
> [image]