hoylabs/OpenDTU-OnBattery #219: HM-1500 allows only 300W, if 600W is set

Frogowitsch, 2023-05-04:
### What happened?

I set the limit to 600W and my HM-1500 only allows 300W. As soon as I set the limit to 1200W it goes up to 520W, which are the peak atm. How can I change this ? Modules are at input 1 and 2.

### To Reproduce Bug

Set limit to 600W and only 300W will be there, even if the Modules create more power. Occurs if Modules are at input 1 and 2.

### Expected Behavior

At the moment there should be 520W, if I set the limit to 600W. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

 569edbe

Frogowitsch, 2023-05-04:
I found it out by myself. What a joke. If I knew, that its an Input 1+2 and 3+4 limit, then I would not have bought the HM-1500. Its impossible this way to limit it right. Bad implementation hardware-wise... :(

bf4ed, 2023-05-04:
It's not a bug, it's a feature.

You need 4 panels or connect all 4 inputs together.

helgeerbe, 2023-05-04:
I close this., as it is not an openDTU-onBattery issue.

closed 2023-05-04 by helgeerbe