hoylabs/OpenDTU-OnBattery #230: [Request] multiple Inverter at Dynamic Power Limiter

felix-m-p, 2023-05-10:
### Describe the solution you'd like

Hi,
I currently have 2 HM-600 installed. It would be great if it's possible to add multiple inverters to the dynamic power limiter.

Regards

### Additional context

Thanks for such a helpful project!

Yoeky, 2023-05-21:
I would like to second this request. Especially since I can only get around 500W out of my HM-600 from my 24V battery, so I'd like to add a second HM-600 that I have ready to be added.

I think instead of selecting a specific inverter, just adding checkboxes for the inverters to use for DPL and then just evenly distribute the load over all of them would do the trick at least for a start. Because of the limits, this would only make sense with identical inverters. A second step could be having all the limits set separate for each inverter.
There might be some potential in first using the first inverter, and then start increasing the power on the second one if the first one reached its upper limit to get a better efficiency since the inverters are less efficient at certain load settings as far as I know.

helgeerbe, 2023-05-22:
Each additional inverter will influence the response time to limiter changes. Worst case is: number of inverters * poll rate.

- 1 inverter * 5 sec poll rate -> 5 sec adjustment time
- 10 inverter * 5 sec poll rate -> 50 sec adjustment time

Have in mind, that you are not allowed to produce more than 600W (in future maybe 800W). This is not the same as export to grid. 

On a software failure you will be on the safe side, but if you try this with 4 HM-1500 you might burn your house down.

So my advice would be, before you increase the number of inverters, take a more powerful inverter. Go for 48V (instead of 24V) to achieve the full power of an inverter.

Yoeky, 2023-05-24:
I thought about your (valid!) points a bit, but I still think at least a basic extension to 2 or 3 inverters makes sense:

Response times:
If the logic would only start using the second inverter once the first inverters upper limit is exceeded, this should not be a problem. All power requests up to 500w would be fulfilled by inverter 1, once that limit is exceeded inverter 2 is the one that needs the updates for fine adjustments, inverter 1 then stays at 500w and needs no updates. 

Is 5 seconds the fastest poll rate a hoymiles can do? (just wondering since that is also the lowest setting in opendtu and ahoydtu)

I do not see the danger of burning a house down. This has something to do with correct cables, diameters and fuses. It got nothing to do with opendtu or other controllers. I have all my solar systems on separate circuits...

Getting a hm-1500 would be an option, but just a really expensive one when you have other inverters lying around....

helgeerbe, 2023-06-02:
I agree that supporting a second Inverter might be suitable for some cases.

I would propose to use the first inverter as the main inverter, the second would be a standby and only used, if the max limit exceed the capacity of the first inverter. That should it make easy to implement.

Yoeky, 2023-06-02:
> I would propose to use the first inverter as the main inverter, the second would be a standby and only used, if the max limit exceed the capacity of the first inverter. That should it make easy to implement.

I totally agree. That would be an easy to understand, simple and efficient logic

neolauren-cmd, 2023-07-09:
Halle @helgeerbe  gibt es bei diesem Projekt schon Neuigkeiten? Ich möchte mir ungern eine zweite OpenDTU kaufen. Gruß  neo

martingro, 2023-11-13:
I've installed a 600 and run it with fixed limit and the 1500 is working with power limit (6 panels, 3 MPPT 100/20). In the summer time I could set the 600 to 200W without draining the battery until next day. On very cloudy days, the BMS switched off until the panels recharched. As the 40A CB got triggered to often, the 1500 is now limited at 1200. 
Now during winter, the 600 runs with 3% and the 1500 is limited to 600W. It barely manages to cover our standby throughout the day/high tarif time from 6 to 21h, I cut off the system with the shelly 1PM during night time.

jladefoged, 2023-12-18:
I only have HM-300 inverters (1 inverter per panel) so only supporting 1 inverter makes the dynamic power limiter feature not really working.

breuerm710, 2024-01-23:
@helgeerbe very unfortunately I can't program stuff but I own a HM300 and two HMS1600 and would be willing to test code.

gitisgreat2023, 2024-04-28:
@genmad has a solution based on a shelly, capable of API 2 or higher, creating virtual power meters that multiple DTUs (on Battery) can use.

See here for the code https://github.com/genmad/shellyAPIGen2AndAboveScripts

> I agree that supporting a second Inverter might be suitable for some cases.
> 
> I would propose to use the first inverter as the main inverter, the second would be a standby and only used, if the max limit exceed the capacity of the first inverter. That should it make easy to implement.

closed 2024-11-17 by schlimmchen