hoylabs/OpenDTU-OnBattery #240: [Request] Automatischer Neustart des WR an der Batterie

nocopy074, 2023-05-24:
### Is your feature request related to a problem? Please describe.

Erst einmal vielen vielen Dank für das großartige Projekt!

Die Anzeige "Inverter Gesamtertrag Heute" beeinhaltet den Tagesertrag des WR an der Batterie aus der vorangenagenen Nachteinspeisung. 

### Describe the solution you'd like

Ich hätte gerne einen automatischen Neustart des Wechselrichters an der Batterie zu einer konfigurierbaren oder festen Zeit (z.B. 12 Uhr), damit sich der Tagesertrag des WR an der Batterie auf 0 setzt und damit der Wert "Inverter Gesamtertrag Heute" in der Übersicht aktualisiert und der Ertrag aus der Nacht nicht mehr hinzu addiert wird.

helgeerbe, 2023-05-24:
Persönlich nutze ich die Tageszähler nicht, sondern nur den total Wert. In Grafana berechne ich darüber die Tageswerte. Das hat zusätzlich den Vorteil, dass nach unbeabsichtigten Neustarts, die Daten immer noch stimmen.

Die Tageszähler vom Wechselrichter und Ladegerät sind eigentlich nicht zu gebrauchen.

spcqike, 2023-05-24:
eine alternative Lösung wäre, den Tagesertrag des Wechselrichters gar nicht zu nutzen, sondern den Gesamtertrag.

der Tagesertrag setzt sich auch im Laufe des Tages zurück, wenn der WEchselrichter mal abgeschaltet oder neugestartet wird, oder wenn die Paneele stark verschattet sind sodass kurzzeitig keine Produktion stattfindet.

Meiner Meinung nach wäre es sinnvoller, den Tagesertrag in der openDTU zu berechnen, anhand des vortlaufend hochzählenden Gesamtertrags. 

den Gesamtertragswert eines jeden Inverters müsste man zu einer festen Zeit (0 Uhr? mit angedachtem DeepSleep der openDTU über Nacht ggf. beim Aufwachen?) zwischenspeichern und dann entsprechend als Referenz nehmen.

gleichzeitig könnte man die Ausgabe des Tageswerts der des Gesamtwerts anpassen. der eine wird ja in kWh übergeben, der andere in Wh. Das macht Aufzeichnungen nämlich auch blöd :D 

ich betrachte bisher auch nur den Gesamtzähler und werte diesen mit Influx/Grafana aus

madmartin, 2023-05-24:
Soll ich dazu was programmieren?

helgeerbe, 2023-05-24:
@madmartin und was machen die Kollegen, die ihre openDTU in den DeepSleep schicken?  Wenn du eine Lösung findest, mit der alle (na ja 80%) zufrieden sind, gerne.

nocopy074, 2023-05-24:
@madmartin Ich wäre ja für ein Neustart gegen 12:00 Uhr Mittags, da ich so den Ertrag aus der Batterie in der Nacheinspeisung nach nachvollziehen könnte, währen eine Neustart in der Nacht mir den Wert kaputt machen würde. 

Genrell stimme ich euch allen zu, dass eine genaue Dokumentation auf einem weiteren Gerät mit Graphana o.ä. passieren sollte.

helgeerbe, 2023-05-24:
In #233 haben wir einen Fall, wo der Wechselrichter wohl nicht mit Dauerbetrieb klar kommt. Meiner hatte selbst einmal ein komisches Verhalten, wo die gesetzen Limit Werte nicht stimmen. Erst ein Neustart hat geholfen. So wäre ein Zwangsneustart ggf. schon sinnvoll.

nocopy074, 2023-05-24:
Ein Neustart würde ja auch dem "natürlichen Verhalten" eines WR ohne Batterie entsprechen.

spcqike, 2023-05-24:
was spricht gegen einen Neustart zur bereits vorhandenen nautischen Morgendämmerung (NTP)? oder dann, wenn die ersten wenigen Watts produziert werden?

Wann weckt sich eine DeepSleep openDTU denn auf?

madmartin, 2023-05-24:
> @madmartin und was machen die Kollegen, die ihre openDTU in den DeepSleep schicken? Wenn du eine Lösung findest, mit der alle (na ja 80%) zufrieden sind, gerne.

Zum Thema DeepSleep sage ich mal: verträgt sich konzeptionell nicht mit der Regelung der Batterie-Leerung zur Nacht. Was soll ich denn da mit DeepSleep? Ich will doch daß die Regelung auch nachts arbeitet.
Da ich ein Herz für Usecases ungleich meinen eigenen habe - denke ich mal an diejenigen die nur den Powerlimiter zur NullEinspeisung nutzen wollen. Bei denen wird der WR nachts ja abgeschaltet, also interessiert es nicht.
Hab ich noch was vergessen? 
Ach ja - überhaupt mal nach DeepSleep im Projekt gesucht und dazu https://github.com/tbnobody/OpenDTU/discussions/265 und https://github.com/tbnobody/OpenDTU/issues/397 gefunden. Ergebnis: ist nur in nicht aktuellen Forks enthalten.
Also aus meiner Sicht ist DeepSleep hier nicht relevant. Bitte korrigiert mich wenn ich was übersehen habe.

So schlage ich eine WR-Neustart-Implementierung mit wählbarer Uhrzeit vor, die als default abgeschaltet ist.

madmartin, 2023-06-04:
Siehe #255

helgeerbe, 2023-06-06:
Sollte mit 2023.06.06 implementiert sein.