hoylabs/OpenDTU-OnBattery #251: Support of Shelly pro 3em (digest authentification)

helgeerbe, 2023-06-02:
### Is your feature request related to a problem? Please describe.

Based on the discussion #247 the Power Meter Module has to be extended to support beside basic authentication also digest authentication, to support the new Shelly pro 3em. 

### Describe the solution you'd like

extend the http + json Konfiguration to use digest authentication if it receives a 401 error.

swingstate, 2023-06-19:
This would be a massive improvement and make other home automations, at least partially, obsolete.

helgeerbe, 2023-06-27:
Unfortunately the HTTClient lib on esp32 doesn't support digest authentication.  So I asked chatGPT to generate me a code example, that should run on a esp32.

The code itself looks promising, but I can't test it. @swingstate if I create a test branch with a fixed user and password. Would you be able to compile and test it?

````c++
#include <WiFi.h>
#include <HTTPClient.h>
#include <MD5Builder.h>

const char* ssid = "YourWiFiSSID";
const char* password = "YourWiFiPassword";
const char* url = "http://example.com/api/endpoint";  // Replace with your API endpoint

// Digest authentication credentials
const char* username = "YourUsername";
const char* realm = "YourRealm";
const char* plaintextPassword = "YourPlaintextPassword";
char passwordDigest[33];  // Buffer to store the password digest (MD5 hash)
bool authenticationCompleted = false;

void setupWiFi() {
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi...");
  }
  Serial.println("Connected to WiFi");
}

void generatePasswordDigest() {
  MD5Builder md5;
  md5.begin();
  md5.add(username);
  md5.add(':');
  md5.add(realm);
  md5.add(':');
  md5.add(plaintextPassword);
  md5.calculate();
  md5.getBytes(passwordDigest);
}

bool performHttpRequest(HTTPClient& httpClient, const char* url, int attempt = 0) {
  httpClient.begin(url);

  if (attempt > 0) {
    // Add Authorization header with the digest response
    char authorizationHeader[256];
    snprintf(authorizationHeader, sizeof(authorizationHeader), "Digest username=\"%s\", realm=\"%s\", nonce=\"%s\", uri=\"%s\", response=\"%s\"",
             username, realm, httpClient.authNonce().c_str(), url, passwordDigest);
    httpClient.setAuthorization(authorizationHeader);
  }

  int httpCode = httpClient.GET();

  if (httpCode == HTTP_CODE_OK) {
    String response = httpClient.getString();
    Serial.println("Response: " + response);
    return true;
  } else if (httpCode == HTTP_CODE_UNAUTHORIZED && attempt == 0) {
    // Handle authentication challenge
    if (httpClient.hasHeader("WWW-Authenticate")) {
      String authHeader = httpClient.header("WWW-Authenticate");
      if (authHeader.indexOf("Digest") != -1) {
        authenticationCompleted = false;
        return performHttpRequest(httpClient, url, 1); // Retry with authentication
      }
    }
  } else {
    Serial.printf("HTTP request failed with error code %d\n", httpCode);
  }

  return false;
}

void performDigestAuthentication() {
  // Create an HTTPClient object
  HTTPClient http;

  while (!authenticationCompleted) {
    if (performHttpRequest(http, url)) {
      authenticationCompleted = true;
    }
  }

  http.end();
}

void setup() {
  Serial.begin(115200);

  setupWiFi();

  generatePasswordDigest();

  performDigestAuthentication();
}

void loop() {
  // Your main code here
}

````

helgeerbe, 2023-06-27:
Ich habe noch einen Shelly Plus 1PM. Der soll auch digest authentication machen. Dann kann ich mir selber ein Testprogramm schreiben.

helgeerbe, 2023-07-04:
[image]

I don't own a Shelly pro 3em, so I could only test with a Shelly 1 plus pm.

Please let me know, if this is working now. (I consider this feature as experimental).

helgeerbe, 2023-07-04:
Should be done with release 2023.07.04.post1

closed 2023-07-04 by helgeerbe

swingstate, 2023-07-04:
Sorry, I didn't catch your ask for help.
Do you still need support here @helgeerbe ?

helgeerbe, 2023-07-04:
@swingstate Should be working with the latest release. Feel free to test.