hoylabs/OpenDTU-OnBattery #252: adding POST on web api calls for /api/vedirectlivedata/status and api calls to configure and turn the dynamic power limiter on/off

Yoeky, 2023-06-02:
### Is your feature request related to a problem? Please describe.

For non-Victron users, it would be really helpful if there was a way to somehow inject the MPPT charger live data into 
OpenDTU-OnBattery every few seconds in order to be able to use solar passthrough. 

I would also love to be able to remotely (via web API) configure and turn the dynamic power inverter on/off, so I could have Home Assistant turn it off and manually set the power limit on the inverter for a few hours for whatever reasons (or change settings from time to time based on the weather forecast)...

### Describe the solution you'd like

for the MPPT data: an API call that could easily be made by some esphome or Home Assistant installation would be great. I assume the only relevant values needed would be "V" and "I" as shown by /api/vedirectlivedata/status, so if a new API POST call would allow only to update those two fields, the whole existing logic would "just work" like it does with Victrons. Is that assumption correct?

for the dynamic power limiter, the api call to /api/powerlimiter/status should just be writable

helgeerbe, 2023-06-05:
Hi @Yoeky ,

there is a PR [#232](https://github.com/helgeerbe/OpenDTU-OnBattery/pull/232) pending, that would enable/disable solar pass through via mqtt. Maybe this would be helpfull for you.

At the moment I don't have any plans to implement a remote interface for MPPTs. If someone is going to implement this, I would support the integration.

schlimmchen, 2024-04-12:
> configure and turn the dynamic power inverter on/off, so I could have Home Assistant turn it off and manually set the power limit on the inverter for a few hours

You can disable the DPL using the mode switch (MQTT), see the [Wiki](https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/MQTT#power-limiter-topics). HomeAssistant knows this switch. You can then set a limit on the inverter using MQTT/HomeAssistant.

> For non-Victron users, it would be really helpful if there was a way to somehow inject the MPPT charger live data

That would be done in #457 using MQTT.

Is there something else that you are asking for which is not covered by the DPL mode switch and the MQTT solar power input?

Yoeky, 2024-04-15:
> > For non-Victron users, it would be really helpful if there was a way to somehow inject the MPPT charger live data
> 
> That would be done in #457 using MQTT.
> 
> Is there something else that you are asking for which is not covered by the DPL mode switch and the MQTT solar power input?

I think that should work. I have Renogy MPPTs and get their data in home assistant (using esphome), ideally I could just push those values to opendtu directly from my esphome once the interface exists.

schlimmchen, 2024-04-15:
Yeah, I tried setting up my smart switches this way as well, so I would not need a broker. Turns out the user experience is much better with MQTT, as HTTP requires a lot of back and forth, while MQTT is made for such environments and is very lightweight.

schlimmchen, 2024-12-01:
* We do allow disabling the DPL using MQTT already.
* Disabling the DPL using HTTP is doable: Request the current config, patch the JSON to only disable the DPL keeping the other settings as they are, and POST'ing that patched JSON to apply the settings.
* We have #457 and that will eventually be available.
* We may implement some kind of HTTP(S)+JSON solar charge provider in the future, although nobody asked for it yet. However, implementing a POST endpoint for that and defining an API is tedious and less flexible. I don't support adding POST endpoints to "connect" to peripherals. That would be a new scheme of talking to peripherals, and I maintain that MQTT is the better option.

@AndreasBoehm Would you like to keep this open? I tend to close it because I think that such a paradigm shift (POST'ing to OpenDTU-OnBattery suddenly becomes a thing) is not the right choice for OpenDTU in general.

AndreasBoehm, 2024-12-04:
I agree that we should not add POST endpoints to inject data into OpenDTU. 

In my opinion we can close this.

closed 2024-12-04 by schlimmchen

stefan123t, 2024-12-04:
While I agree that it would require to have a separate REST API for ODOB which you woukd have to support theb.

I would like to point out that there already are a small set of endpoints upstream, eg at leadt for APL ActivePowerLimit and Basic Start/Stop/Restart.

i have documented the REST API upstream using a OpenAPI/Swagger file. https://github.com/tbnobody/OpenDTU/issues/378#issuecomment-2479470412

There is also a version for ODOB if you want to use it for the ODOB changes to the upstream endpoints.

schlimmchen, 2024-12-04:
Hmmmmmm, I don't think that's quite on the nose. You see, those endpoints/APIs are already there because they are part of the web UI. And the web UI API is **not** stable, see the docs:

[image]

Adding an endpoint to receive POST data from/for a peripheral is a completely different beast.