hoylabs/OpenDTU-OnBattery #259: Restart Hoymiles if Efficiency drops below 70% percent[Request]

Manos1966, 2023-06-13:
### Is your feature request related to a problem? Please describe.

I notice that sometimes, the inverter will get "stuck" producing with efficiency of only 50% percent (normal is somewhere between 94% and 95% percent).
The only way to get the inverter out of this loop is by re-starting it.

I can not reproduce the effect. It usually happens sometime during the day when:
1. Using two 25V batteries connected at 50V
2. the batteries are "drained out" from the previous night and did not have time to recharge itself. 
3. The VICTRON is charging the batteries thus delivering high voltage (which the OpenDTU **without CANbus** interprets as "battery fully charged").
4. There is high electricity consumption in the house thus, OpenDTU gives the command to the inverter to start producing electricity at max.
5. The electricity provided my VICTRON is lower than the electricity the inverter is trying to deliver thus the (empty) batteries is further drained... even below the DC Voltage - Stop threshold. 
6. The BMS of the batteries stops them (to protect them), usually this means one battery will stop earlier than the other thus voltage drops suddenly from 48V down to 24V
7. OpenDTU stops the inverter (DC Voltage - Stop threshold reached)
8. With Inverter stopped, VICTRON continues to produce, thus the situation goes back to point #3 and the vicious "loop" starts again.

### Describe the solution you'd like

Proposed solution:
1. Add the option to "restart the inverted" at a selected time, once per day. This will get the inverter out of the loop if it has happened.

2. Have an option to monitor the efficiency % of the inverter and if it identifies that over a longer period the inverter is:
     - producing electricity (i.e. inverter is not idle)
     - with efficiency below 70%  
    restart the inverter

spcqike, 2023-06-13:
Proposed solution 1 is already build in openDTU-onBattery. In power limiter config page at the bottom. You can select one hour to restart your inverter. It was implemented to reset the daily production counter to 0 and in general to give the inverter the change to power cycle/reboot.

Manos1966, 2023-06-13:
> Proposed solution 1 is already build in openDTU-onBattery. In power limiter config page at the bottom. You can select one hour to restart your inverter. It was implemented to reset the daily production counter to 0 and in general to give the inverter the change to power cycle/reboot.

Thanks @spcqike ! THAT would be great! (and sufficient to stop the problem).
I am running May 23, 2023 Firmware, was not there, let me update and check

helgeerbe, 2023-06-13:
Drücke mal  die Reload Page im Browser. Ich vermute mal, die Seite ist noch im Browser cache. Der Restart-Parameter ist ganz unten.

Manos1966, 2023-06-13:
> Drücke mal die Reload Page im Browser. Ich vermute mal, die Seite ist noch im Browser cache. Der Restart-Parameter ist ganz unten.

Genau, das war es! DANKE!

closed 2023-06-13 by Manos1966