hoylabs/OpenDTU-OnBattery #267: Huawei AC Charger only accepts integer numbers

indie89, 2023-06-20:
### What happened?

I want to set the voltage of the Huawei AC Charger via MQTT (same in webinterface) to e.g. 53.3 V. Unfortunately only integer numbers (like 53 or 54) are accepted an transmitted. Why is that and could it be possible to transfer a number with at least one decimal place?

### To Reproduce Bug

Same behaviour whether I try it via mqtt or via web interface...

### Expected Behavior

I want to set voltage values with at least one decimal place.

Thanks for the project! I just start using opendtu on battery for controlling my Huawei charger and my hoymiles inverter to charge/discharge my battery :-)

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

a6f5e8a

MalteSchm, 2023-06-27:
I'm on this
I actually fixed this as part of some other features related the the Huawei PSU that I'm testing right now

MalteSchm, 2023-06-27:
I actually re-based my dev branch with the development branch today. 
https://github.com/MalteSchm/OpenDTU-OnBattery/tree/huawei_auto_power

All my testing looks good so far.

MalteSchm, 2023-07-01:
Tested in context of https://github.com/helgeerbe/OpenDTU-OnBattery/pull/288 and I think this works

indie89, 2023-07-07:
> Tested in context of #288 and I think this works

mhhh... I tested it with the precompiled v2023.07.04 and unfortunately still the same problem in the webinterface (tried it with "," and "." as separator):
[image]

MalteSchm, 2023-07-09:
Fix is in https://github.com/helgeerbe/OpenDTU-OnBattery/pull/302

helgeerbe, 2023-07-09:
PR #302 is included in release 2023.07.09

closed 2023-07-09 by helgeerbe

indie89, 2023-07-10:
Thanks for taking care! Unfortunately, my inital issue was not solved by Fix #302 

[image]

MalteSchm, 2023-07-10:
This is weird.
Which version do you use?

indie89, 2023-07-10:
The most recent one from last night (1d559c1). 
I's not very important for me, that it works, now that I have it connected to my mqtt server, but still: if you fix it, you wanna know it works properly ;-).

bf4ed, 2023-07-10:
For me the floating point is working in the latest version. Try to erase flash on the ESP32. I use erase flash command in terminal on linux.

I have problems with saving the settings on newest builds. I have 3 OpenDTU's, the older one always with updated versions is ok. The new with fresh install is not stable.

MalteSchm, 2023-07-10:
@indie89 
It could be that your browser has the HTML / JS code cached. I noticed this a few times in the past when developing
Try clearing the cache

indie89, 2023-07-10:
> 

Thanks, that indeed worked fine! Thanks for the hint!