hoylabs/OpenDTU-OnBattery #268: Inverter hängt, DTU wird nicht aktiv

uli-bs, 2023-06-21:
### What happened?

Der Inverter (HM-600) und openDTU (onBattery) läuft zunächst einwandfrei und wie konfiguriert.
Nachts (0:00) wird der Inverter über die DTU neu gestartet, was auch im Log vermerkt ist.
Danach treten zeitlich versetzt lt.. Log Fehler (DTU command failed / Error 2) auf (15x)
Am nächsten Morgen, genügend Ladung ist bereits im Akku, startet, trotz entsprechender Anforderung (480W), der Inverter nicht.

Um wieder normalen Betrieb zu erreichen muss sowohl der Inverter, wie auch openDTU, über die Oberfläche neu gestartet werden.
Der Neustart nur einer der beiden Komponente löst das Problem nicht.

### To Reproduce Bug

Tritt nicht jeden Tag auf, daher eher schlecht zu reproduzieren.

### Expected Behavior

event eine Art Watchdog implementieren?

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

	fd58ad2

spcqike, 2023-06-21:
Ich weiß nicht, obs zu deinem Problem passt, aber ich hab in letzter Zeit auch vermehrt DTU command failed. heute bereits mindestens 15 mal. (zumindest war das das letzte, was ich gesehen habe... )
bei mir zeigt sich das teilweise auch in Neustarts der DTU, in fehlenden Daten in der Aufzeichnung und, wie jetzt grade, im nicht laufenden Webinterface
[image]

[image]
[image]

wobei die webkonsole aktuell geht
[image]

Dabei habe ich weder einen Akku, noch habe ich aktuell den DPL aktiv. im Moment läuft die DTU nur zum auslesen des Inverters.

uli-bs, 2023-06-21:
Den Aufbau des Logs habe ich noch nicht verstanden.
Anscheinend werden max. 15 Einträge gespeichert, aber es sind weder die ersten 15, noch die letzten 15?
Es sind ohnehin nur "Inverter starts" , "...commnd failed" und "shut down by remote.." enthalten, kein "Power up/down" o.ä
Der "Inverter starts" scheint permanent im Log zu sein, die anderen haben anscheinend irgendwelche "TTLs"

In der Konsole werde ich mein Problem vermutlich nicht finden, da ja aktuelle Daten und keine History

uli-bs, 2023-06-21:
Auf welcher Basis beruht eigentlich die aktuelle Version (23.06.14.post2)?

Hier [https://github.com/tbnobody/OpenDTU/issues/1032](url) wird nämlich ein ähnliches Problem beschrieben, was angeblich mit der 23.6.14 nicht mehr auftrat.

helgeerbe, 2023-06-21:
23.6.13 Findest du auch immer in der Readme als Badge. Ich habe da eine Action laufen, die immer in der Readme die Version der originalen OpenDTU für die aktuelle Version angibt

helgeerbe, 2023-06-21:
Werde mal schauen, ob ich Zeit habe gleich mal zu mergen und eine neue Realease freizugeben

uli-bs, 2023-06-21:
Erstmal vielen Dank für die schnelle Reaktion und das neue Build

Als erste Feststellung an den "missing" in der Konsole hat sich nichts geändert, oder muss das so?
Die DTU (Leistung auf "min.") hängt ca. 30cm neben dem WR, ein Übertragungsproblem dürfte da ja wohl nicht sein

Ob sich an dem Kernproblem etwas geändert hat wird sich erst in ein paar Tagen zeigen.

helgeerbe, 2023-06-21:
Also das "missing" habe ich auch. Habe mir den Code dazu aber nie angeschaut, da ich funktionell keine Probleme habe. Will sagen, ich habe keine Ahnung was diese Logausgabe wirklich aussagt.

helgeerbe, 2023-06-21:
Das die DTU neustartet sollte eigentlich nicht sein. Das deutet auf irgendwelche Bugs hin. Bei mir lief jetzt alles über 6 Tage total stabil.

Welche Komponenten sind bei euch aktiviert?
Bei mir 
- VeDirect an
- PowerMeter (https) an
- PowerLimiter (solar passthrough) an
- Batterie aus
- AC Charger aus

spcqike, 2023-06-21:
In meinem Fall aktuell nur MQTT, powermeter (https) und das oled. 

Ich werde meine DTU morgen updaten und dann beobachten

uli-bs, 2023-06-21:
MQTT ohne HA
Powermeter (http)
Limiter (ohne pt)
Display

VE -off-
Batterie -off-
AC-Lader -off-

uli-bs, 2023-06-22:
Bei mit hat das System den WR geplant um 0:00 nu gestartet, Error 2 ist bisher noch nicht aufgetreten, allerdings kommt aktuell auch keine Leistung in den Akku, sodass er auch nicht freigegeben wird.

helgeerbe, 2023-06-22:
Bei mir sieht alles gut aus

Geplanter Neustart des Inverters um 23 Uhr und Abschalten des Invertes, wenn das andere Kraftwerk genug Leistung hat, dass keine Batterieunterstützung benötigt wird.

[image]

Ich habe aber auch gar keine Idee, was Fehler 2 sein soll.

helgeerbe, 2023-06-22:
> ich sehe den Log vom Inverter, 15 Fehler. (kann es sein, dass nicht mehr angezeigt oder ausgelesen werden können?)

Ich glaube größer ist der interne Fehlerspeicher des Inverters nicht. Und er überschreibt auch Meldungen mit unterschiedlichen Prios. Soweit ich mich erinnere.

uli-bs, 2023-06-22:
Error 2 war "...Command failed", flutete in der Vorgängerversion auch mein Log

spcqike, 2023-06-22:
> > ich sehe den Log vom Inverter, 15 Fehler. (kann es sein, dass nicht mehr angezeigt oder ausgelesen werden können?)
> 
> Ich glaube größer ist der interne Fehlerspeicher des Inverters nicht. Und er überschreibt auch Meldungen mit unterschiedlichen Prios. Soweit ich mich erinnere.

Macht Sinn. Denn der Start (Code 1) ist ja noch vorhanden.

Bisher ist auch nichts weiter passiert.

Ich hab auch keine Ahnung, welcher Command da hängen geblieben sein soll. Die openDTU sollte bei mir ja aktuell nichts schicken. Oder werden die aktuellen Leistungsdaten ebenfalls gepollt?

die Konsole sieht aktuell wie folgt aus:

```
12:31:49.917 > [PowerLimiterClass::loop] ******************* ENTER **********************
12:31:49.917 > [PowerLimiterClass::loop] ******************* PL disabled
12:31:54.197 > PowerMeterClass: TotalPower: 69.00
12:31:54.197 > Fetch inverter: xxxx
12:31:54.302 > TX RealTimeRunData Channel: 61 --> 15 83 72 27 49 80 17 12 76 80 0B 00 64 94 23 1A 00 00 00 00 00 00 00 00 B2 7A F3 
12:31:54.412 > Interrupt received
12:31:54.610 > RX Channel: 3 --> 95 83 72 27 49 83 72 27 49 02 77 5D 00 00 E6 D7 01 3E 02 2E 09 3F 13 89 06 2B 1E | -80 dBm
12:31:54.711 > Interrupt received
12:31:54.814 > RX Channel: 75 --> 95 83 72 27 49 83 72 27 49 83 00 00 00 43 03 E8 01 69 00 54 6D 03 EC | -30 dBm
12:31:54.898 > RX Period End
12:31:54.898 > Middle missing
12:31:54.898 > Request retransmit: 1
12:31:54.898 > TX RequestFrame Channel: 75 --> 15 83 72 27 49 80 17 12 76 81 F8 
12:31:54.989 > Interrupt received
12:31:55.076 > RX Channel: 40 --> 95 83 72 27 49 83 72 27 49 01 00 01 01 83 00 BB 02 D2 01 8A 00 ED 03 A4 00 00 BD | -80 dBm
12:31:55.222 > RX Period End
12:31:55.222 > Success
```
```
12:32:21.456 > [PowerLimiterClass::loop] ******************* ENTER **********************
12:32:21.456 > [PowerLimiterClass::loop] ******************* PL disabled
12:32:21.456 > Fetch inverter: xxxx
12:32:21.626 > TX RealTimeRunData Channel: 23 --> 15 83 72 27 49 80 17 12 76 80 0B 00 64 94 23 35 00 00 00 00 00 00 00 00 83 A2 35 
12:32:21.728 > Interrupt received
12:32:21.829 > RX Channel: 3 --> 95 83 72 27 49 83 72 27 49 01 00 01 01 83 00 BC 02 D8 01 8A 00 ED 03 A6 00 00 B2 | -80 dBm
12:32:21.933 > Interrupt received
12:32:22.027 > RX Channel: 75 --> 95 83 72 27 49 83 72 27 49 02 77 5E 00 00 E6 D8 01 3F 02 2F 09 49 13 8A 06 33 7F | -80 dBm
12:32:22.132 > Interrupt received
12:32:22.205 > RX Channel: 61 --> 95 83 72 27 49 83 72 27 49 83 00 01 00 43 03 E8 01 69 00 54 52 0C DD | -30 dBm
12:32:22.344 > RX Period End
12:32:22.344 > Success
```
gelegentlich mit "Middle missing", aber auch oft ohne. die Kanäle ändern sich gelegentlich.

helgeerbe, 2023-06-22:
Die Daten werden immer gepollt. Allerdings habe ich mir das genaue Protokoll nicht angeschaut. 
Das "Missing" habe ich auch gelegentlich, meine Annahme bisher, das ggf. wegen Störungen usw. etwas nicht korrekt empfangen werden kann. Es wird ja auch ein retransmit angestoßen. Sieht für mich aber ganz normal aus.

spcqike, 2023-06-23:
Kurzes Update. Bisher läuft es dann ohne Auffälligkeiten. Keine Fehler Meldungen, keine erkennbaren Lücken in der Aufzeichnung.

Laufzeit 21:11 h. 
Heap sagt 50%. (137, 134, 271 kB)

spcqike, 2023-06-28:
Update:
Heute war wieder so ein Tag. 
nach knapp 24h (genaue Laufzeit weiß ich nicht) gab es wieder diesen Fehler 2 im log. 15 Einträge. Das werden ja wahrscheinlich nicht alle gewesen sein. 

Danach war die DTU irgendwann völlig weg. Ich bin nicht drauf gekommen. Kein Ping, keine IP, kein MQTT, keine api. 

Zwischenzeitlich kam ich auf die api. Es gab aber immer nur null als Antwort. 

Jetzt komme ich wieder auf die Info Seite. Die DTU läuft aktuell seit 1 Tag 3 Stunden. Sie hat sich also nicht neu gestartet. MQTT geht wieder. Ebenso die api. 

Heap steht bei 69% (82/186/267 kB)

kann man irgendwie auslesen / herausfinden, was den heap füllt?

helgeerbe, 2023-06-29:
@spcqike doofe Frage: Netzwerkprobleme kannst du ausschliessen? Kein Mesh-Netzwerk oder ähnliches?
Da würde mir jetzt nur ein Log über USB einfallen. Dann kann man genau sehen, was der esp treibt, wenn er nicht verfügbar ist.

spcqike, 2023-06-29:
100%ig auschließen kann man sowas nie :) Mein Heimnetzwerk ist schon .... größer und komplexer als eine FritzBox :) aber da ich alle anderen Geräte (und ihre Daten) aus dem IoT Netzwerk einwandfrei erreichen konnte (bzw. die anderen Daten ohne Lücken protokolliert wurden, also bspw. Tasmota Sensoren die ebenfalls MQTT nutzen), würd ich sagen da lag kein Problem vor.

spätestens, als ich die OpenDTU wieder per Ping erreicht habe, und auch per "http://<IP>/api/livedata/status", dort aber nur "null" zurück kam, würd ich sagen das Problem liegt bei der DTU.

wie gesagt, später ging dann die API wieder (der Neustart allerdings nicht. der gab weiterhin im Webinterface ein "Parsen nicht möglich" und direkt per API aufruf ein "null") ebenso hatte sich MQTT von alleine gefangen. Dabei war der HEAP dann recht belegt, siehe oben. 

zum HEAP, da du es in einem anderen Post meintest, dass das Maximum sich nicht ändern sollte. Ich kann sagen, es ändert sich. wie geschrieben, gestern im "halb aufgehangenen Zustand" lag er bei 267kB Max. nach dem Neustart (am Netzteil) ging es hoch auf 271, aktuell sind es 270.

Meinst du, es würde zum Debuggen etwas bringen, den Seriellen Log mal ein paar Tage aufzuzeichnen? Dann würd ich mir mal was einfallen lassen, wie ich das hinbekomme....

helgeerbe, 2023-06-29:
Bringen würde es mit sicherheit etwas. Ich würde erwarten, dass wenn der liveview nicht funktioniert du im Log sowas sehen würdest: "Call to /api/livedata/status temporarely out of resources."

Generell tippe ich auf ein Speicherproblem. LiveView ist immer das, wo es am schnellsten auffällt. Dort wird versucht den größten zusammenhängenden Memory-Block zu bekommen (für den Status-Report). In Summe mag zwar genug Speicher verfügbar sein, aber durch Fragmentierungen im Speicher gibt es keinen zusammenhängenden Block mehr. Das kann passieren, wenn kleinere Speicher-Blöcke nicht sauber freigegeben werden. Dann fallen irgendwann weitere Funktionen aus und am Ende bootet der ESP neu.
Meine Installation läuft über stabil und ich habe die Probleme nicht. Da es ja anscheinend immer bei euch auftritt, müsst ihr irgendetwas konfiguriert haben, was die Probleme verursacht, und was die die meisten anderen so nicht verwenden.

Mein Problem ist es, dass ich es nicht nachstellen kann.

spcqike, 2023-06-29:
na, ich gucke mal und lasse mir was einfallen. wird schon irgendwie gehen mit dem Log :) 

zu meiner config:
esp32 dev board, oled mit sh1106, nrf24l01+ mit externer Antenne, gelötet auf nen perfboard (keine extra Kondensatoren, die sollten ja aber den Speicher nicht beeinflussen, oder?)

-das Modul ist im WLAN
-Dammerung standard (90.8)
-MQTT standard ohne HA oder AUthentifizierung, aller 5 sekunden updates
-ein HM-600 (nachts keine befehle, sonst alle Felder am Wechselrichter befüllt, also Leistung und Namen der Module 
-DTU 5s intervall, Sendeleistung minimum
-VE.direct aus
-Powermeter HTTP+JSON eines Tasmota Geräts, 1000ms timeout, nur Phase 1 als Summe aller
-DPL aktuell inaktiv
-keine Batterie
-kein AC Charger
-hardware profil standard aus dem Repo, generic esp32 mit sh1106
-display mit powersave und screensaver, deutsch, keine rotation, 51% kontrast

> Dann fallen irgendwann weitere Funktionen aus und am Ende bootet der ESP neu.

das ist gestern nicht passiert. er hat sich wieder halbwegs gefangen. MQTT ging wieder, ebenso einige API aufrufe (neustart bspw. nicht), LiveView weiterhin nicht. Das wird schon am belegten HEAP liegen, es waren dann ja nur 82kB frei.

Ich habe noch 2 ähnliche Systeme installiert bei Bekannten, allerdings ohne MQTT und PM (und auch ohne DPL, den ich ja seit ner Woche zum beobachten auch nicht nutze). die laufen unauffällig. (da wird aber auch weniger oft drauf geguckt.... vielleicht kommt das HEAP Problem ja erst durch das Öffnen des LiveView? Mir kommt es auch so vor, als käme das Problem bevorzugt, wenn ich mit dem iPhone drauf gucke. Auf der anderen Seite läuft es auch Tage stabil, und irgendwann häufen sich diese komischen Fehler 2 im InverterLog. Und irgendwann steigt das WebUI aus. die zwei PRobleme müssen ja aber nicht zwingend zusammen hängen (ich kann aber schlecht überprüfen ob VOR dem vollen HEAP und den Abstürzen Fehler da waren. gestern jedenfalls war es so))

uli-bs, 2023-06-30:
Bisher läuft die Version f018a01 vom Jun 21, 2023 relativ problemlos, 
"Relativ" bedeutet, der Inverter wird _nicht immer_ zeitnah zum Bedarf gestartet/gesteuert (trotz 5 sec Polltime), die ominösen Error2 (DTU...) tauchen sporadisch auf, und anscheinend startet auch die DTU ab und an unmotiviert neu. Da Teile dieser Fehler auch im originalem openDTU berichtet werden, dürfte das Problem von dort stammen.

Was mich aber jetzt zusätzlich irritiert, sind die neu verfügbaren Versionen mit "ESP32" im Namen?
Ich ging bisher immer davon aus, dass ein ESP32 die Grundanforderung des Programmes ist?!
Laufen tut hier ein ESP32-D0WD-V3, welche Version wäre hier nun richtig, und kann die entsprechende *.bin einfach als Update geflashed werden?

uli-bs, 2023-07-02:
Heute war der Fehler leider wieder da :(
Im Akku ist genug Power, Startbedingungen erfüllt, Leistungsanforderung: 128W (40W unter akt.Verbrauch) - nur der Inverter startet nicht

[image]

schlimmchen, 2023-07-02:
@uli-bs Scheinbar hast du noch kein Log (Console) geteilt. Kannst du einmal einen Auszug kopieren und uns zeigen? Gerade bist du ja scheinbar im Fehlerzustand, da wäre der Log vermutlich sehr hilfreich.

Was hast du im Kontext das Power Limiter als "lower power limit" für den inverter eingestellt?

uli-bs, 2023-07-02:
Wie das immer so ist... aktuell liefert er wieder

Eingestellt ist:
[image]

Und 250W pro Kanal beim WR

Das Log, nur jetzt kein akuter Fehler:
```
19:29:42.066 > Fetch inverter: 114182922450
19:29:42.137 > TX RealTimeRunData Channel: 75 --> 15 82 92 24 50 80 18 30 88 80 0B 00 64 A1 B4 06 00 00 00 00 00 00 00 00 19 BB 0F 
19:29:42.200 > Interrupt received
19:29:42.301 > RX Channel: 75 --> 95 82 92 24 50 82 92 24 50 02 49 A9 00 00 4A 6C 00 B0 00 B1 09 29 13 86 09 59 B5 | -80 dBm
19:29:42.616 > RX Period End
19:29:42.616 > Last missing
19:29:42.616 > Request retransmit: 3
19:29:42.616 > TX RequestFrame Channel: 3 --> 15 82 92 24 50 80 18 30 88 83 D2 
19:29:42.686 > Interrupt received
19:29:42.815 > RX Channel: 40 --> 95 82 92 24 50 82 92 24 50 83 00 00 00 66 03 E8 01 82 00 27 75 A5 EF | -80 dBm
19:29:42.902 > RX Period End
19:29:42.902 > Middle missing
19:29:42.902 > Request retransmit: 1
19:29:42.902 > TX RequestFrame Channel: 23 --> 15 82 92 24 50 80 18 30 88 81 D0 
19:29:42.980 > Interrupt received
19:29:43.050 > RX Channel: 75 --> 95 82 92 24 50 82 92 24 50 01 00 01 01 04 01 E3 04 E4 01 04 01 E3 04 E5 00 00 94 | -80 dBm
19:29:43.125 > RX Period End
19:29:43.125 > Success
19:29:43.329 > [PowerLimiterClass::loop] ******************* ENTER **********************
19:29:43.329 > [PowerLimiterClass::loop] dcVoltage: 26.00 Voltage Start Threshold: 24.60 Voltage Stop Threshold: 24.10 inverter->isProducing(): 1
19:29:43.329 > [PowerLimiterClass::loop] reusing old limit
19:29:43.329 > [PowerLimiterClass::loop] Status: SolarPT enabled 0, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 1
19:29:43.329 > [PowerLimiterClass::loop] Status: StartTH 1, StopTH: 0, loadCorrectedV 26.239300
19:29:43.329 > [PowerLimiterClass::loop] Status Batt: Ena: 0, PowerMeterClass: TotalPower: 40.00
19:29:47.324 > Fetch inverter: 114182922450
19:29:47.376 > TX RealTimeRunData Channel: 40 --> 15 82 92 24 50 80 18 30 88 80 0B 00 64 A1 B4 0B 00 00 00 00 00 00 00 00 89 E3 CA 
19:29:47.504 > Interrupt received
19:29:47.631 > RX Channel: 40 --> 95 82 92 24 50 82 92 24 50 83 00 00 00 66 03 E8 01 82 00 27 77 CF 87 | -80 dBm
19:29:47.857 > RX Period End
19:29:47.857 > Middle missing
19:29:47.857 > Request retransmit: 1
19:29:47.857 > TX RequestFrame Channel: 61 --> 15 82 92 24 50 80 18 30 88 81 D0 
19:29:48.035 > RX Period End
19:29:48.035 > Middle missing
19:29:48.035 > Request retransmit: 1
19:29:48.035 > TX RequestFrame Channel: 75 --> 15 82 92 24 50 80 18 30 88 81 D0 
19:29:48.096 > Interrupt received
19:29:48.146 > RX Channel: 3 --> 95 82 92 24 50 82 92 24 50 01 00 01 01 04 01 E3 04 E5 01 04 01 E3 04 E5 00 00 95 | -80 dBm
19:29:48.197 > RX Period End
19:29:48.197 > Middle missing
19:29:48.197 > Request retransmit: 2
19:29:48.197 > TX RequestFrame Channel: 3 --> 15 82 92 24 50 80 18 30 88 82 D3 
19:29:48.246 > RX Period End
19:29:48.246 > Middle missing
19:29:48.246 > Request retransmit: 2
19:29:48.246 > TX RequestFrame Channel: 23 --> 15 82 92 24 50 80 18 30 88 82 D3 
19:29:48.318 > Interrupt received
19:29:48.368 > RX Channel: 3 --> 95 82 92 24 50 82 92 24 50 02 49 A9 00 00 4A 6C 00 B0 00 B1 09 2A 13 85 09 59 B5 | -80 dBm
19:29:48.488 > RX Period End
19:29:48.488 > Success
19:29:52.569 > PowerMeterClass: TotalPower: 38.00
19:29:52.569 > Fetch inverter: 114182922450
19:29:52.627 > TX RealTimeRunData Channel: 40 --> 15 82 92 24 50 80 18 30 88 80 0B 00 64 A1 B4 10 00 00 00 00 00 00 00 00 79 5D 9F 
19:29:52.696 > Interrupt received
19:29:52.843 > RX Channel: 75 --> 95 82 92 24 50 82 92 24 50 01 00 01 01 04 01 DD 04 D5 01 04 01 DD 04 D5 00 00 95 | -80 dBm
19:29:52.960 > Interrupt received
19:29:53.049 > RX Channel: 75 --> 95 82 92 24 50 82 92 24 50 83 00 02 00 65 03 E8 01 83 00 27 EF 71 A1 | -80 dBm
19:29:53.253 > RX Period End
19:29:53.253 > Middle missing
19:29:53.253 > Request retransmit: 2
19:29:53.253 > TX RequestFrame Channel: 61 --> 15 82 92 24 50 80 18 30 88 82 D3 
[... 2086 lines cut ...]
19:37:42.867 > Middle missing
19:37:42.867 > Request retransmit: 2
19:37:42.867 > TX RequestFrame Channel: 61 --> 15 82 92 24 50 80 18 30 88 82 D3 
19:37:42.937 > Interrupt received
19:37:43.003 > RX Channel: 23 --> 95 82 92 24 50 82 92 24 50 02 49 B9 00 00 4A 7C 00 C0 00 C1 09 23 13 86 07 6F 87 | -80 dBm
19:37:43.072 > RX Period End
19:37:43.072 > Success
19:37:47.528 > PowerMeterClass: TotalPower: 40.00
19:37:47.528 > Fetch inverter: 114182922450
19:37:47.585 > TX RealTimeRunData Channel: 75 --> 15 82 92 24 50 80 18 30 88 80 0B 00 64 A1 B5 EB 00 00 00 00 00 00 00 00 88 7B B2 
19:37:47.651 > Interrupt received
19:37:47.777 > RX Channel: 3 --> 95 82 92 24 50 82 92 24 50 01 00 01 01 05 01 80 03 E5 01 04 01 80 03 E5 00 00 94 | -80 dBm
19:37:47.849 > Interrupt received
19:37:47.913 > RX Channel: 40 --> 95 82 92 24 50 82 92 24 50 02 49 B9 00 00 4A 7C 00 C0 00 C1 09 21 13 86 07 6F 85 | -80 dBm
19:37:48.078 > Interrupt received
19:37:48.141 > RX Channel: 3 --> 95 82 92 24 50 82 92 24 50 83 00 00 00 51 03 E8 01 82 00 27 35 E4 D9 | -80 dBm
19:37:48.258 > RX Period End
19:37:48.258 > Success
19:37:48.318 > [PowerLimiterClass::loop] ******************* ENTER **********************
19:37:48.318 > [PowerLimiterClass::loop] dcVoltage: 26.10 Voltage Start Threshold: 24.60 Voltage Stop Threshold: 24.10 inverter->isProducing(): 1
19:37:48.318 > [PowerLimiterClass::loop] reusing old limit
19:37:48.318 > [PowerLimiterClass::loop] Status: SolarPT enabled 0, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 1
19:37:48.318 > [PowerLimiterClass::loop] Status: StartTH 1, StopTH: 0, loadCorrectedV 26.290300
19:37:48.318 > [PowerLimiterClass::loop] Status Batt: Ena: 0, ``
```

uli-bs, 2023-07-04:
Jetzt mit weniger Müll und mit Fehler:

[image]

`
```shell
Websocket: [/livedata][20] disconnect
18:03:16.822 > Websocket: [/vedirectlivedata][20] disconnect
18:03:16.822 > Websocket: [/huaweilivedata][20] disconnect
18:03:16.976 > Websocket: [/batterylivedata][20] disconnect
18:03:18.611 > RX Period End
18:03:18.611 > All missing
18:03:18.611 > Nothing received, resend whole request
18:03:18.611 > TX PowerControl Channel: 40 --> 51 82 92 24 50 80 18 30 88 81 00 00 B0 01 25 
18:03:18.732 > Interrupt received
18:03:18.798 > RX Channel: 3 --> D1 82 92 24 50 82 92 24 50 81 00 00 00 00 24 00 74 | -80 dBm
18:03:19.117 > PowerMeterClass: TotalPower: 136.00
18:03:20.628 > RX Period End
18:03:20.628 > Success
18:03:20.763 > TX ActivePowerControl Channel: 61 --> 51 82 92 24 50 80 18 30 88 81 0B 00 03 E8 00 00 10 81 E5 
18:03:20.880 > Interrupt received
18:03:20.968 > RX Channel: 3 --> D1 82 92 24 50 82 92 24 50 81 00 00 0B 00 14 07 48 | -80 dBm
18:03:22.699 > RX Period End
18:03:22.700 > Success
18:03:22.700 > Fetch inverter: 114182922450
18:03:22.811 > TX RealTimeRunData Channel: 75 --> 15 82 92 24 50 80 18 30 88 80 0B 00 64 A4 42 CA 00 00 00 00 00 00 00 00 0C 4A D4 
18:03:22.927 > Interrupt received
18:03:23.015 > RX Channel: 3 --> 95 82 92 24 50 82 92 24 50 83 00 00 00 00 00 00 01 28 00 39 BA 93 2F | -80 dBm
18:03:23.217 > RX Period End
18:03:23.217 > Middle missing
18:03:23.217 > Request retransmit: 1
18:03:23.217 > TX RequestFrame Channel: 3 --> 15 82 92 24 50 80 18 30 88 81 D0 
18:03:23.324 > RX Period End
18:03:23.324 > Middle missing
18:03:23.324 > Request retransmit: 1
18:03:23.324 > TX RequestFrame Channel: 23 --> 15 82 92 24 50 80 18 30 88 81 D0 
18:03:23.442 > RX Period End
18:03:23.442 > Middle missing
18:03:23.442 > Request retransmit: 1
18:03:23.442 > TX RequestFrame Channel: 40 --> 15 82 92 24 50 80 18 30 88 81 D0 
18:03:23.545 > Interrupt received
18:03:23.631 > RX Channel: 23 --> 95 82 92 24 50 82 92 24 50 01 00 01 01 09 00 03 00 07 01 08 00 03 00 07 00 00 94 | -80 dBm
18:03:23.683 > RX Period End
18:03:23.683 > Middle missing
18:03:23.683 > Request retransmit: 2
18:03:23.683 > TX RequestFrame Channel: 61 --> 15 82 92 24 50 80 18 30 88 82 D3 
18:03:23.747 > RX Period End
18:03:23.747 > Middle missing
18:03:23.747 > Request retransmit: 2
18:03:23.747 > TX RequestFrame Channel: 75 --> 15 82 92 24 50 80 18 30 88 82 D3 
18:03:23.835 > RX Period End
18:03:23.835 > Middle missing
18:03:23.835 > Retransmit timeout
18:03:24.358 > PowerMeterClass: TotalPower: 139.00
18:03:26.367 > [PowerLimiterClass::loop] ******************* ENTER **********************
18:03:26.367 > [PowerLimiterClass::loop] ******************* PL inverter updates PM: 4747, Inverter: -385 
18:03:27.679 > Fetch inverter: 114182922450
18:03:27.873 > TX RealTimeRunData Channel: 3 --> 15 82 92 24 50 80 18 30 88 80 0B 00 64 A4 42 CF 00 00 00 00 00 00 00 00 5C 75 BE 
18:03:28.045 > Interrupt received
18:03:28.132 > RX Channel: 23 --> 95 82 92 24 50 82 92 24 50 01 00 01 01 09 00 03 00 07 01 08 00 03 00 07 00 00 94 | -80 dBm
18:03:28.198 > Interrupt received
18:03:28.254 > RX Channel: 3 --> 95 82 92 24 50 82 92 24 50 83 00 00 00 00 00 00 01 28 00 39 BF 5E E7 | -80 dBm
18:03:28.309 > RX Period End
18:03:28.309 > Middle missing
18:03:28.309 > Request retransmit: 2
18:03:28.309 > TX RequestFrame Channel: 23 --> 15 82 92 24 50 80 18 30 88 82 D3 
18:03:28.381 > Interrupt received
18:03:28.543 > RX Channel: 40 --> 95 82 92 24 50 82 92 24 50 02 4D 09 00 00 4D D3 00 15 00 15 09 36 13 8F 00 00 EE | -80 dBm
18:03:28.608 > RX Period End
18:03:28.608 > Success
[... 41 lines cut ...]
18:03:40.965 > TX RealTimeRunData Channel: 61 --> 15 82 92 24 50 80 18 30 88 80 0B 00 64 A4 42 DC 00 00 00 00 00 00 00 00 6C AC 44 
18:03:41.141 > Interrupt received
18:03:41.232 > RX Channel: 40 --> 95 82 92 24 50 82 92 24 50 01 00 01 01 09 00 03 00 07 01 08 00 03 00 07 00 00 94 | -80 dBm
18:03:41.460 > RX Period End
18:03:41.460 > Last missing
18:03:41.460 > Request retransmit: 2
18:03:41.460 > TX RequestFrame Channel: 75 --> 15 82 92 24 50 80 18 30 88 82 D3 
18:03:41.591 > RX Period End
18:03:41.591 > Last missing
18:03:41.591 > Request retransmit: 2
18:03:41.591 > TX RequestFrame Channel: 3 --> 15 82 92 24 50 80 18 30 88 82 D3 
18:03:41.743 > RX Period End
18:03:41.743 > Last missing
18:03:41.743 > Request retransmit: 2
18:03:41.743 > TX RequestFrame Channel: 23 --> 15 82 92 24 50 80 18 30 88 82 D3 
18:03:41.808 > Interrupt received
18:03:41.946 > RX Channel: 40 --> 95 82 92 24 50 82 92 24 50 02 4D 09 00 00 4D D3 00 15 00 15 09 37 13 8F 00 00 EF | -80 dBm
18:03:41.999 > RX Period End
18:03:41.999 > Last missing
18:03:41.999 > Request retransmit: 3
18:03:41.999 > TX RequestFrame Channel: 40 --> 15 82 92 24 50 80 18 30 88 83 D2 
18:03:42.073 > Interrupt received
18:03:42.152 > RX Channel: 75 --> 95 82 92 24 50 82 92 24 50 83 00 00 00 00 00 00 01 29 00 39 7F CE B6 | -80 dBm
18:03:42.220 > RX Period End
18:03:42.220 > Success

schlimmchen, 2023-07-04:
Hm... Ganz schön zerhackstückelt. Da kannst du natürlich nichts dafür.

Also immerhin sehen wir, dass er den Wechselrichter einschalten möchte: "Starting up inverter...". Danach sehen wir ein "TX ActivePowerControl", aber keine Indikation, dass ein PowerControl (nicht Active-) gesendet wird. Allerdings weiß ich auch nicht sicher, ob man da so sieht, wie ich es jetzt erwarte. Zu wenig Erfahrung...

Es ist natürlich möglich, dass versucht wird, den Inverter zu starten (Hoymiles Bibliothek), aber dass das Kommando nicht ankommt (zeitweise Probleme auf der Funkstrecke). Wenn deine Hysterese groß ist (30 W ist eher viel) und dein Netzbezug eher "ruhig", dann kann es sein, dass kein weiteres Einschalten versucht wird.

Auf welchem Release bist du denn gerade unterwegs, und könntest du auf das heutige aktualisieren? Da kenn ich mich jedenfalls besser aus :grin: Wenn du aktualisierst, dann lass nach Möglichkeit ansonsten mal alles, wie es ist, wenns du es einrichten kannst.

uli-bs, 2023-07-04:
Version https://github.com/helgeerbe/OpenDTU-OnBattery/commit/f018a0136ea7ab12c96b70110150ba1be8f23344 vom Jun 21, 2023 
Anfassen möchte ich heute allerdings nichts mehr, die Koffer stehen hier schon rum... ;)

Mittlerweile produziert er auch wieder, ohne dass ich etwas gemacht habe.
Event. gibt es aber eine "Vorgeschichte", dass der WR möglicherweise "beleidigt" war - im Log standen innerhalb weniger Minuten mehrere Stop-Vorgänge, durch entsprechende Lastwechsel, ich weiß nicht, ob es ein (zeitabhängiges?) Limit gibt, bei der Anzahl der Kommandos, bis ggf. ein Überlauf stattfindet.

schlimmchen, 2023-07-28:
@uli-bs Kann das Issue geschlossen werden, oder hast du noch Probleme, die damit zusammenhängen?

uli-bs, 2023-07-28:
An den Problem hat sich nichts geändert, mittlerweile läuft wieder die aktuellste Version.

the-lonely-one, 2023-07-30:
Hallo, habe ein ähnliches Problem und hoffe das ich bei der Fehlersuche behilflich sein kann!

Meine Daten:
Firmwareversion: 43e836a
Chip-Modell: ESP32
Display: SSD1306
Inverter: Hoymiles HM-800

MQTT: aktiviert
DTU-Abfrageintervall: 5 Sekunden
VE.Direct: aktiviert
Power Meter: aktiviert HTTP(S) + JSON
Dynamic Power Limiter: deaktiviert
Batterie CAN Bus Schnittstelle: deaktiviert
AC Ladegerät: deaktiviert

Ich lasse über MQTT ca. alle 20 bis 30 Sekunden ein Limit für den Inverter mit dem Topic "limit_nonpersistent_absolute" setzen.
Die OpenDTU-OnBattery empfängt auch den Wert (ersichtlich über Debug-Konsole) und gibt diesen an den Inverter weiter (ebenfalls ersichtlich über Debug-Konsole).

Irgendwann (nach 1, 2, 10, 15, ... Stunden Laufzeit, ganz unterschiedlich) wird dieser Wert aber nicht mehr an den Inverter weitergeben!
Die OpenDTU-OnBattery empfängt zwar den Limit-Wert über MQTT, gibt diesen aber dann nicht mehr an den Inverter weiter!

Auch ein manuelles Limit-setzen über die Live-Ansicht: -> Limit-Einstellungen funktioniert dann nicht mehr!

Das aktuell gemeldete Limit vom Inverter bleibt unverändert!

Verbindungsprobleme zum Inverter bestehen eigentlich nicht, da ich die aktuellen Werte vom Inverter in der OpenDTU-OnBattery angezeigt bekomme!

Hier ein Auszug aus der Debug-Konsole, wenn alles funktioniert:

```
19:23:13.656 > Fetch inverter: 1234567890
19:23:13.708 > TX RealTimeRunData Channel: 3 --> 15 84 23 75 07 80 18 02 96 80 0B 00 64 C3 F9 80 00 00 00 00 00 00 00 00 E8 97 E6 
19:23:13.767 > Interrupt received
19:23:13.865 > RX Channel: 40 --> 95 84 23 75 07 84 23 75 07 01 00 01 01 2C 00 43 00 C8 01 F4 01 3E 06 32 00 01 CC | -80 dBm
19:23:13.924 > Interrupt received
19:23:13.972 > RX Channel: 61 --> 95 84 23 75 07 84 23 75 07 02 34 C3 00 02 3A 95 00 0E 00 7E 08 F8 13 87 06 A4 7B | -80 dBm
19:23:14.022 > Interrupt received
19:23:14.076 > RX Channel: 40 --> 95 84 23 75 07 84 23 75 07 83 00 01 00 4A 03 E8 01 6E 00 01 40 A6 3E | -80 dBm
19:23:14.172 > RX Period End
19:23:14.172 > Success
19:23:22.879 > Limit Non-Persistent: 300 W
19:23:22.929 > TX ActivePowerControl Channel: 23 --> 51 84 23 75 07 80 18 02 96 81 0B 00 0B B8 00 00 61 83 53 
19:23:22.977 > Interrupt received
19:23:23.031 > RX Channel: 40 --> D1 84 23 75 07 84 23 75 07 81 00 00 0B 00 14 07 48 | -80 dBm
19:23:23.662 > PowerMeterClass: TotalPower: 186.00
19:23:24.947 > RX Period End
19:23:24.947 > Success
19:23:28.643 > Fetch inverter: 1234567890
19:23:28.713 > TX RealTimeRunData Channel: 40 --> 15 84 23 75 07 80 18 02 96 80 0B 00 64 C3 F9 8F 00 00 00 00 00 00 00 00 18 D6 58 
19:23:28.815 > Interrupt received
19:23:28.917 > RX Channel: 40 --> 95 84 23 75 07 84 23 75 07 01 00 01 01 2C 00 44 00 CB 01 F4 01 3E 06 32 00 01 C8 | -80 dBm
19:23:28.964 > Interrupt received
19:23:29.122 > RX Channel: 40 --> 95 84 23 75 07 84 23 75 07 02 34 C3 00 02 3A 96 00 0E 00 7F 08 F6 13 86 06 A7 75 | -80 dBm
19:23:29.177 > Interrupt received
19:23:29.233 > RX Channel: 40 --> 95 84 23 75 07 84 23 75 07 83 00 01 00 4A 03 E8 01 6D 00 01 12 91 58 | -80 dBm
19:23:29.283 > RX Period End
19:23:29.283 > Success
19:23:34.151 > PowerMeterClass: TotalPower: 180.00
19:23:43.643 > Fetch inverter: 1234567890
19:23:43.705 > TX RealTimeRunData Channel: 61 --> 15 84 23 75 07 80 18 02 96 80 0B 00 64 C3 F9 9E 00 00 00 00 00 00 00 00 48 16 D9 
19:23:43.868 > Interrupt received
19:23:43.918 > RX Channel: 40 --> 95 84 23 75 07 84 23 75 07 83 00 00 00 4B 03 E8 01 6E 00 01 19 17 D6 | -80 dBm
19:23:44.452 > PowerMeterClass: TotalPower: 185.00
19:23:44.453 > RX Period End
19:23:44.453 > Middle missing
19:23:44.453 > Request retransmit: 1
19:23:44.453 > TX RequestFrame Channel: 75 --> 15 84 23 75 07 80 18 02 96 81 4D 
19:23:44.508 > Interrupt received
19:23:44.688 > RX Channel: 75 --> 95 84 23 75 07 84 23 75 07 01 00 01 01 2C 00 45 00 CE 01 F4 01 43 06 47 00 01 C4 | -80 dBm
[... 212 lines cut ...]
18:47:30.570 > Limit Non-Persistent: 300 W
18:47:30.711 > TX ActivePowerControl Channel: 40 --> 51 84 23 75 07 80 18 02 96 81 0B 00 0B B8 00 00 61 83 53 
18:47:30.813 > Interrupt received
18:47:30.915 > RX Channel: 75 --> D1 84 23 75 07 84 23 75 07 81 00 00 0B 00 14 07 48 | -80 dBm
18:47:32.676 > RX Period End
18:47:32.676 > Success
18:47:34.629 > Fetch inverter: 1234567890
18:47:34.680 > TX RealTimeRunData Channel: 61 --> 15 84 23 75 07 80 18 02 96 80 0B 00 64 C6 94 26 00 00 00 00 00 00 00 00 C7 C7 57 
18:47:34.909 > Interrupt received
18:47:34.963 > RX Channel: 40 --> 95 84 23 75 07 84 23 75 07 01 00 01 01 33 00 6E 01 50 01 F4 01 3F 06 32 00 01 66 | -80 dBm
18:47:35.009 > Interrupt received
18:47:35.120 > RX Channel: 40 --> 95 84 23 75 07 84 23 75 07 02 3B 1F 00 02 4D 54 04 CE 07 1A 08 FE 13 86 07 26 3D | -80 dBm
18:47:35.216 > Interrupt received
18:47:35.320 > RX Channel: 75 --> 95 84 23 75 07 84 23 75 07 83 00 00 00 50 03 E8 01 A9 00 0F 4F A7 E2 | -80 dBm
18:47:35.421 > RX Period End
18:47:35.421 > Success
```

Die Meldungen sind ähnlich zur OpenDTU-OnBattery!
Aufgefallen ist mir aber, das hier zusätzlich noch die Meldung: "Received MQTT message on topic: solar/1234567890/cmd/limit_nonpersistent_absolute" vorhanden ist,
die es bei OpenDTU-OnBattery nicht gibt!

Bevor ich es noch vergesse!

Finde das Projekt: "OpenDTU-OnBattery" echt genial und ziehe meinen Hut!

helgeerbe, 2023-07-31:
@schlimmchen kann das irgendwas mit #287 zu tun haben, wenn sich die originalDTU und -onBattery unterschiedlich verhalten? Allerdings ist bei mir noch nicht aufgetreten.

uli-bs, 2023-07-31:
Mir sieht das nach einer Art "Überlauf" aus:
Nach dem Start funktioniert alles und nach unterschiedlicher Zeit eben nicht mehr, bis es irgendwann, nach Neustart, oder auch ohne Eingriff, wieder funktioniert.

schlimmchen, 2023-07-31:
> Hallo, habe ein ähnliches Problem und hoffe das ich bei der Fehlersuche behilflich sein kann!

@the-lonely-one Danke für deine ausführliche Beschreibung.

Mir fäll auf, dass der `espMqttClient` scheinbar in einem eigenen Task läuft, weil wir den Konstruktor ohne Argumente verwenden (kann man im Code nachschauen). Das würde bedeuten, dass der MQTT Client asynchron zur `loop()` läuft. Das heißt wohl wiederum, dass alle Subscriber asynchron zur `loop()` laufen. Dass das immer reibungslos ist, wage ich anzuzweifeln.

Mein Gefühl ist es, dass meine Änderung https://github.com/helgeerbe/OpenDTU-OnBattery/pull/287/commits/8b233246930de31f3816f3d967f4ac74b64f4fd2 womöglich Teil des hier diskutierten Problems ist. Das würde allerdings bedeuten, dass bereits ein anderes Problem vorliegt, nämlich dass `CMD_PENDING` gesetzt wird *nachdem* das Kommando bereits verarbeitet wurde.

Das ist theoretisch möglich: Der MQTT Task benutzt `sendActivePowerControlRequest` und wird unterbrochen, nachdem das Kommando in der Queue gelandet ist und bevor `setLastLimitCommandSuccess(CMD_PENDING)` im Kontext des MQTT Task aufgerufen wird. In dieser Unterbrechung läuft die `loop()` und verarbeitet das Kommando (oder wirft es weg, weil die Seriennummer noch nicht gesetzt wurde...) und die Antwort des Inverters wird notiert: `CMD_OK` (oder `CMD_NOK`). Das Scheduling wechselt wieder und der MQTT Task setzt in seinem Kontext `setLastLimitCommandSuccess(CMD_PENDING)`. Dann ist `sendActivePowerControlRequest` für immer gesperrt, denn es gibt kein weiteres Kommando in der Queue, das `CMD_PENDING` zurücksetzen würde und es kann auch keins mehr reinkommen weil `CMD_PENDING` gesetzt ist.

Man könnte ein Pflaster basteln, aber das macht natürlich keinen Sinn, weil meiner Einschätzung nach alle Leute immer davon ausgehen, dass ihr Code synchron in der `loop()` läuft, was wegen des MQTT Clients aber nicht der Fall zu sein scheint.

Der MQTT Client darf keinen internen Task benutzen, um seine `loop()` aufzurufen. Stattdessen muss die MQTT `loop()` synchron im Kontext der main `loop()` aufgerufen werden.

Eine enstprechende Änderung habe ich vorbereitet und grundsätzlich auf meiner Platform verifiziert (MQTT funktioniert wie gewohnt). @the-lonely-one welchen Typ Firmware brauchst du? Ich würde dir gerne eine Vorabversion bauen, die du testen könntest. Wenn ich es richtig verstanden habe, tritt das Problem bei dir ja recht zuverlässig auf, da könnten wir nach 24h ohne dein Problem schon davon ausgehen, dass es in der Tat behoben wurde.

Es macht übrigens Sinn, dass viele Leute dieses Problem nie sehen, weil sie selten oder gar nicht MQTT nutzen, um den Inverter zu steuern. Es braucht schon meine oben verlinkte Änderung *und* ein häufiges Steuern des Inverters durch MQTT, um diese Race Condition zu treffen.

schlimmchen, 2023-07-31:
> kann das irgendwas mit #287 zu tun haben, wenn sich die originalDTU und -onBattery unterschiedlich verhalten? Allerdings ist bei mir noch nicht aufgetreten.

@helgeerbe Benutzt du MQTT, um deine Inverter zu steuern?

Meiner Einschätzung nach reden wir in diesem Issue jetzt von mindestens zwei Problemen. Wenn ich @the-lonely-one richtig verstanden habe, dann hat meine Änderung, nur Commands vom selben Typ in die Queue zu lassen, wenn noch keins drin ist, dazu geführt, dass der MQTT Client Task zu einem realen Problem wird. Das hab ich ja im vorherigen Kommentar ausführlich beschrieben.

@uli-bs schreibt aber davon, dass er lauter `DTU command failed / Error 2` sieht und den Inverter neu starten muss. Damit hab ich hoffentlich nichts zu tun. Also ich hoffe es nicht nur, sondern ich hab da schon drüber nachgedacht, und ich sehe es nicht, dass ich da helfen kann.

the-lonely-one, 2023-07-31:
@schlimmchen Die Antwort ging ja echt schnell

Ich habe einen ESP32 und hatte zuletzt die OpenDTU-OnBattery Firmwareversion: 43e836a drauf, bei der aktuell auch das Problem vorhanden ist.
Somit bräuchte ich eine neu "opendtu-onbattery-generic_esp32.bin" oder eine neue "opendtu-onbattery-generic_esp32.factory.bin"

schlimmchen, 2023-07-31:
@the-lonely-one Dann freue ich mich, wenn du für uns einmal die Firmware aus https://github.com/schlimmchen/OpenDTU-OnBattery/releases/tag/mqtt-no-task ausprobieren könntest und berichtest.

the-lonely-one, 2023-07-31:
@schlimmchen Test läuft

Schaut aktuell gut aus! Meldungen in der Debug-Konsole sind jetzt ähnlich zur originalen OpenDTU!
Lasse das Ganze mal einen Tag durchlaufen und melde mich dann morgen Abend nochmal!

the-lonely-one, 2023-08-01:
@schlimmchen Und läuft und läuft und läuft!!!

Nach dem ich gestern Abend meine DTU mit der von dir bereitgestellten Firmwareversion "aca46f2" aktualisiert habe, ist das von mir beschriebene Problem nicht mehr aufgetreten! 

Aktuell läuft meine DTU seit ca. 22 Stunden durchgehend ohne Probleme!

Die Randbedingungen waren in diesem Zeitraum gleich:
Habe über MQTT alle 20 Sekunden ein Limit für den Inverter mit dem Topic "limit_nonpersistent_absolute" setzen lassen.

Danke dir nochmal, für die schnelle Bearbeitung!

schlimmchen, 2023-08-01:
@the-lonely-one Sehr gerne, war ein spannender Fehler.

Also nehmen wir mal an, dass es in der Tat ein Problem/Fehler war und dass das Problem bei dir nicht mehr auftritt, weil du es bisher mindestens nach 15h beobachtet hast. Ich erstelle dann noch einen Pull-Request gegen das Upstream-Projekt und verlinkte deine ausführliche Fehlerbeschreibung.

the-lonely-one, 2023-08-05:
@schlimmchen  Limit Vorgabe für meinen Inverter über MQTT funktioniert bisher immer noch ohne Probleme!

Mir ist nun aber etwas anderes aufgefallen.
Und zwar wird mir der Wert: "Inverter Gesamtertrag Heute" mehrmals am Tag auf 0 zurückgesetzt!

Siehe Chart:
(Y-Achse sind "Wh")
[image]

Vom 31.07. bis zum 02.08. hatte ich die von dir bereitgestellte Firmware-Version zum testen installiert. Aktuell ist die Firmware: 81864b3 installiert, bei der aber ebenfalls das Problem besteht!
An den bisher genannten Einstellungen an der OpenDTU-OnBattery habe ich nichts geändert!
Selben Bedingungen wie bisher!

Laut dem Webinterface von meiner OpenDTU-OnBattery kann ich weder Neustarts von meiner DTU selbst noch von meinem Hoymiles-Inverter nachvollziehen.
Wird eigentlich der Wert: "Inverter Gesamtertrag Heute" in der OpenDTU berechnet oder vom Hoymiles-Inverter bereitgestellt?

Ich vermute mal, das mein Problem etwas mit der Web-API bzw. mit der Webserver-Implementierung zu tun, das aktuell bei OpenDTU als Pull Request [#1201](https://github.com/tbnobody/OpenDTU/pull/1201) diskutiert wird!

schlimmchen, 2023-08-05:
> Limit Vorgabe für meinen Inverter über MQTT funktioniert bisher immer noch ohne Probleme!

Das freut mich zu lesen!

Kopierst du deine Frage nach dem Zurücksetzen der Inverter-Tagesleistung bitte in ein neues Issue?

closed 2023-08-09 by helgeerbe

schlimmchen, 2023-12-28:
@the-lonely-one Ich würde mich freuen, wenn du uns erfahren lässt, ob dein Problem mit dem Release [aus Dezember 2023](https://github.com/helgeerbe/OpenDTU-OnBattery/releases/tag/2023.12.27.post1) oder neuer wieder auftritt. Halt da doch mal ein Auge drauf, bitte.

commit 9c7dc681 (2022-10-26): Fix #268: Allow12 digit dtu serial numbers

They will also be generated by OpenDTU itself