hoylabs/OpenDTU-OnBattery #277: DPL MQTT full solar passthrough is unconditional

schlimmchen, 2023-06-26:
### What happened?

I noticed (in the code) that if the DPL is set into mode `PL_MODE_SOLAR_PT_ONLY` using MQTT, the power limit is unconditionally matched to the solar charger's output power. That's *full* solar passthrough.

I do not use the feature, but this implementation is surprising and I am guessing that it is a bug? If not, the documentation could be updated to explain that this mode really makes all solar energy disappear into the grid :wink: 

### To Reproduce Bug

Set DPL into mode `PL_MODE_SOLAR_PT_ONLY` using MQTT -- or look at the code.

### Expected Behavior

Non-full battery may still charge even when `PL_MODE_SOLAR_PT_ONLY` is set using MQTT.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2223afa

MalteSchm, 2023-06-27:
Hi @schlimmchen 
I'm using this feature and this is intensional.

You're the 2nd person that asked this question. I commented / explained the rationale here: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/232#issuecomment-1583297708

In essence the feature turns the whole setup into a regular inverter that provides all solar power to the home. In my case this is done when the car is charging 

But it seems that there is some unclarity around how this can be used so I think the documentation can be improved.

helgeerbe, 2023-06-29:
Can I close this issue?

schlimmchen, 2023-06-29:
Yes, preferably after merging #282 :wink:

schlimmchen, 2023-07-04:
> Can I close this issue?

Done :wink:

PR #282 (2023-07-02): explain DPL modes set through MQTT in more detail
Closes #277
closed 2023-07-04 by schlimmchen