hoylabs/OpenDTU-OnBattery #281: Pin mapping missing after upgrading to 2023.06.23

dszam, 2023-06-28:
### What happened?

I downloaded the opendtu-on-battery-generic.bin to upgrade from release 2023.06.14 to 2023.06.23.
After the upgrade I saw 

`[Pylontech] Twai driver get status - invalid state`

messages in the console. After investigating, I found that the pin mapping in "Hardware -> Standardprofil" was missing/incorrect for ve.direct and battery.
Attached is a screenshot of the wrong settings after the upgrade.
After downgrading to 2023.06.14 again, all pin mappings were correct again.
[image]

To get it working again, I needed to create a custom pin_mapping.json and upload it.

### To Reproduce Bug

- upgrade from 2023-06-14 to 2023-06-23 by using the "generic" bin file.
- look at the console
- look at the hardware settings -> standard profile

Components are Smartsolar MPPT 100/20 and a pylontech compatible battery using can.

### Expected Behavior

the pin settings stay as they are, after upgrading the firmware

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2223afa

Manos1966, 2023-06-28:
Thanks for posting this.
Upgraded this morning and changed HW.
I almost went crazy trying to understand what went wrong....

madmartin, 2023-06-28:
I also upgraded two days ago to https://github.com/helgeerbe/OpenDTU-OnBattery/commit/2223afac627585a175c78f425cf43a8a17dd1b6f and did NOT have this problem.

BUT i compiled it myself, did not use a downloaded binary.

Manos1966, 2023-06-28:
> I compiled it myself, did not use a downloaded binary.

Angeber!  :-p
Was it "opendtu-onbattery-generic.bin", update by OTA
Upgraded three systems (testing various setups in parallel) by OTA, all had the same result (Victron pins became 255).
This was the Firmware Page I got them from this morning:
[image]

helgeerbe, 2023-06-29:
It seems, that this is the intended behavior. With version v23.6.21 in the original OpenDTU pre-compiled bins don't have any pin config. If there is not pin_mapping.json the defaults values are used, which are -1 for OpenDTU-onBattery extensions.

If you are using the pre-compiled bin, you have to prepare one  pin_mapping.json and import it once into openDTU. After that, all following updates should work as expected, without uploading a pin_mapping.json again.

All who are compiling their own bin, don't have to do this, because pins default values are defined in platformio.ini.

But anyhow. Pin type changed from unit8_t to int8_t. That is the reason why 255 is shown instead of -1. Will fix it.

schlimmchen, 2023-06-29:
Is it possible that @helgeerbe previously uploaded a firmware built with the `d1_mini_esp32` environment? I don't see how else people got a working board without uploading a `pin_mapping.json` and without building their own firmware. The `d1_mini_esp32` is the only environment that defines OpenDTU-OnBattery-specific pins, and the issue reporter did mention that previously both VE.Direct and Pylontech CAN were working without a `pin_mapping.json`.

fwscholz, 2023-06-30:
Hello.
Sorry for the question.
How does such a `pin_mapping.json` file look like, containing the additional OnBattery pins only? (VE.Direct pins would be enough for me.)
Thanks!

schlimmchen, 2023-06-30:
@fwscholz Have a look at https://github.com/helgeerbe/OpenDTU-OnBattery/blob/master/docs/DeviceProfiles.md It should make it clear.

helgeerbe, 2023-07-07:
Not a bug

closed 2023-07-07 by helgeerbe