hoylabs/OpenDTU-OnBattery #284: Dynamic Power Limiter: Upper Inverter Limit not working

Manos1966, 2023-06-30:
### What happened?

Sorry for being a pain.... 
I have a problem with the Power Limiter settings. The Solar Pass-through is disabled and it seems that the Upper Watt Limit for the inverter is not working.
Solar Pass-through: DISABLED
Inverter Limit at Power Limiter set to: 900W
Despite this, the inverter is at its max 1500W

[image]
[image]
[image]
[image]

### To Reproduce Bug

Aktiviere Solar Pass-through: DISABLED
Oberes Leistungslimit: enter a limit lower than the max power of the inverter.

### Expected Behavior

Normally the inverter should not go above the set limit (in my pictures 900W).
Despite that it goes to the max of 1500W (it is a HM-1500).

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2223afa

schlimmchen, 2023-06-30:
Thanks for providing expresiv screenshots.

Can you please look into your console's output, there should be a line `[PowerLimiterClass::setNewPowerLimit] ... channels total, ... producing channels, scaling power limit`. What exactly does it say?

Does this problem stay for a longer period of time, or just for short periods (half a minute), after which the limit is dialed back again?

schlimmchen, 2023-06-30:
Oh, understood. I can reproduce this immediately and the console output makes it obvious: Since your grid consumption is even larger than 1500W, the power limit is calculated to be 4400W in the case of your screenshot: Grid consumption plus power generated by the inverter.

And then there's the problem: The `newPowerLimit` is indeed limited to the configured upper limit, but the result is assigned to a new variable `effPowerLimit`. However, the old `newPowerLimit` (4400W) is scaled, rather than `effPowerLimit`.

If this is indeed the cause I can provide a merge request later today.

Manos1966, 2023-06-30:
Hallo @schlimmchen  
You were faster than me copying the results of the console :-p

and it seems you already identified the cause, which is great!

Best regards

schlimmchen, 2023-06-30:
> and it seems you already identified the cause, which is great!

I put a firmware on my device with the fix proposed in #286 and the problem went away. Sorry for introducing this regression. @Manos1966 Are you able to build your own firmware or are willing to wait until a new release is published that includes the fix?

Manos1966, 2023-06-30:
Hi @schlimmchen 
First of all, it can wait.
I do not know how to build Firmware. Is there somewhere an explanation on how to do that? 
I begin to see there is a distinct advantage in doing that.
RGDS
Manos

schlimmchen, 2023-07-01:
> I begin to see there is a distinct advantage in doing that.

There should only be an advantage if you want bleeding edge changes.

Please have a look at the README.md of this project and maybe also its upstream project for instruction on how to build your own firmware. Yes, you named the correct section.

Manos1966, 2023-07-05:
With the new Firmware the problem is solved - Many thanks for the quick fix guys! 👍 
<html>
<body>

Firmwareversion / git Hash | 344498d
-- | --

</body>
</html>

[image]
Firmwareversion / git Hash	[344498d](https://github.com/helgeerbe/OpenDTU-OnBattery/commits/344498d)

PR #286 (2023-07-02): DPL: fix the configured upper limit not being respected
the requested `newPowerLimit` was indeed limited to the configured maximum inverter output and the result was stored in `effPowerLimit`. later, however, `effPowerLimit` must be scaled up (if necessary), not `newPowerLimit`. the latter variable is not respecting the configured maximum inverter output.

Closes #284.
closed 2023-07-05 by Manos1966