hoylabs/OpenDTU-OnBattery #293: [Request] MQTT topic subscribe Battery Level for DPL

niceborn, 2023-07-05:
### Is your feature request related to a problem? Please describe.

I don't have a Pylontech battery but a Junteks KH-110F bluetooth battery coulometer (http://www.junteks.com/productinfo/977969.html) to get detailed battery information. which can be additional read by a serial adapter. I connected it to a second ESP32 and submit all received data to my MQTT broker.
Included in the information is also the battery level. 
Right now only the battery DC voltage in Dynamic Power limiter Settings can be used.

### Describe the solution you'd like

I would like to have an option to use an MQTT battery level for the setting in DPL. Setting could be the same as for power meter settings, where also an MQTT option exists.

madmartin, 2023-07-16:
@helgeerbe I think this is a useful enhancement.

I write a PR for this.

schlimmchen, 2023-07-17:
@madmartin I implemented support for the Jikong (JK) BMS as the battery interface. The software is already in use for roughly a week now.

I assume that part of my work could be beneficial to this feature (assuming you have not already implemented that parts that I can contribute). This feature request is asking for a new battery interface, which will use MQTT as the protocol to retrieve supported battery stats rather than CAN/UART/RS485. If you agree, then you might want to base your work on https://github.com/schlimmchen/OpenDTU-OnBattery/commits/additional-battery-interfaces where I already made the respective changes to (1) enable the Battery admin web interface to present a list of supported communication protocols, (2) changed texts so that it is clear that not only Pylontech CAN is supported as battery interface and (3) changed the Pylontech CAN code such that it will only run if it is selected as the battery interface protocol.

I am happy to exchange notes on this topic.

madmartin, 2023-07-17:
@schlimmchen got it... I was just designing the Web UI part, so I take a look at your fork before I continue.

TheGrumpyBadger, 2023-07-18:
I still like to see a MQTT connection for SOC. In my case a Victron SmartShunt is connected to a Raspberry with VenusOS.

swingstate, 2023-08-21:
I'll support this request too. Receiving SOC data as universal MQTT messages would make onBattery even more open to more battery configuration. 

I have a pending request open to support the SOC reading through VE.Direct from a Victron Shunt (Since this project has already integrated parts of Victron hardware), however the MQTT approach would be more universal and I suggest to prioritise the MQTT extension over Ve.Direct for SOC data (can come later). 

Having worked with my setup for a while now, I am clearly missing the feature that onBattery can use SOC information. using the voltage is maximum an indicator.

schlimmchen, 2023-08-29:
@madmartin #319 is ready for review and merging. Feel free to start implementation of the feature described in this issue and let me know if you need any info on something I wrote for the JK BMS support. I would be happy to see that you find your way around my code easily so that implementing the MQTT battery provider is fun :blush:

88markus88, 2023-11-18:
Ist ja schon eine Weile offen .... Ist absehbar daß Input von Batteriedaten via MQTT kommen wird?

DerWe, 2023-11-23:
I'd support this feature as well. Is there anyone working on this Feature currently? Is it foreseeable to see it in any further release?

PR #589 (2024-01-04): Feature: implement MQTT-driven battery provider
this battery provider implementation subscribes to a user-configurable MQTT topic to retrieve the battery SoC value. the value is not re-published under a different topic. there is no card created in the web app's live view, since the SoC is already part of the totals at the top of the live view. that is the only info this battery provider implements.

closes #293.
relates to #581.
commit e7a00583 (2024-01-04): Feature: implement MQTT-driven battery provider (#589)

this battery provider implementation subscribes to a user-configurable
MQTT topic to retrieve the battery SoC value. the value is not
re-published under a different topic. there is no card created in the
web app's live view, since the SoC is already part of the totals at the
top of the live view. that is the only info this battery provider
implements.

closes #293.
relates to #581.
closed 2024-01-04 by helgeerbe