hoylabs/OpenDTU-OnBattery #295: [Request] add/enable "power control" for Hoymiles via mqtt

indie89, 2023-07-07:
### Is your feature request related to a problem? Please describe.

I want to use the "openDTU on Battery" project for automating battery charging via Huawei R4850 and discharging via Hoymiles hm-600. I was confident, that the "power control" mqtt topics were available, as they are in the original openDTU project, but unfortunately they weren't.

I use the inverter completely different than intended in the "openDTU on battery" project, not as pv inverter but as battery inverter, so I need the "power control" ability.

### Describe the solution you'd like

Please add/enable the "power control" topics for the Hoymiles inverter, as they are enabled in the original project. Thanks a lot!

### Describe alternatives you've considered

If this function is not enabled, I have to use another ESP32 to run the original project in order to control the HM-600 as a battery inverter. This is exactly, what I wanted to avoid by using the "openDTU on battery" project.

MalteSchm, 2023-07-07:
What about: cmd/power?

See:
https://github.com/helgeerbe/OpenDTU-OnBattery/blob/master/docs/MQTT_Topics.md

indie89, 2023-07-07:
sorry, my bad, again: I don't have an inverter configured yet but as the ac and dc folders were already existing, I thought the "Inverter limit specific topics" were completely missing in this project, but they are not...probably just will appear when I configure an inverter 🤦‍♂️. I'm sorry!

closed 2023-07-07 by indie89