hoylabs/OpenDTU-OnBattery #298: Stability issues with development branch

MalteSchm, 2023-07-07:
### What happened?

Hi,

I'm unsure if it is only me or not. I compiled from latest development branch and seeing significant stability issues. 

Just erased flash and performed a clean install. Configured the network, inverter, enabled VE.Direct and the Power Meter. No other features are enabled at this point. Right now this seems to work. 
I tested this twice today: when I enable the Power Limiter the ESP seems to stall after some time. No response, no gui

Wondering if it's only me or if other users have these problems as well.

The system has been stable for almost three hours now with the PL disabled. I will now re-enable and capture the logs. 

### To Reproduce Bug

Enable power limiter

### Expected Behavior

No crashes

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

344498d

MalteSchm, 2023-07-07:
`22:41:52.183 > RX Period End
22:41:52.183 > Last missing
22:41:52.183 > Request retransmit: 8
22:41:52.183 > TX RequestFrame Channel: 40 --> 15 83 07 46 82 80 14 75 68 88 54 
22:41:52.183 > Interrupt received
22:41:52.184 > RX Channel: 3 --> 95 83 07 46 82 83 07 46 82 08 00 15 41 FD 42 3C 00 00 00 00 B0 02 00 16 42 4F E3 | -80 dBm
22:41:52.184 > RX Period End
22:41:52.184 > Last missing
22:41:52.184 > Request retransmit: 9
22:41:52.184 > TX RequestFrame Channel: 61 --> 15 83 07 46 82 80 14 75 68 89 55 
22:41:52.185 > RX Period End
22:41:52.185 > Last missing
22:41:52.185 > Retransmit timeout
22:41:52.185 > [PowerLimiterClass::loop] ******************* ENTER **********************
22:41:52.185 > [PowerLimiterClass::loop] dcVoltage: 49.60 Voltage Start Threshold: 50.00 Voltage Stop Threshold: 49.00 inverter->isProducing(): 1
22:41:52.185 > [PowerLimiterClass::loop] newPowerLimit: 412
22:41:52.185 > [PowerLimiterClass::setNewPowerLimit] 4 channels total, 4 producing channels, scaling power limit
22:41:52.185 > [PowerLimiterClass::setNewPowerLimit] reusing old limit: 446 W, diff: 34, hysteresis: 50
22:41:52.185 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, DraiRX Period End
22:41:57.488 > Middle missing
22:41:57.488 > Request retransmit: 3
22:41:57.488 > TX RequestFrame Channel: 40 --> 15 83 07 46 82 80 14 75 68 83 5F 
22:41:57.637 > RX Period End
22:41:57.638 > Middle missing
22:41:57.638 > Retransmit timeout
22:41:57.794 > TX AlarmData Channel: 61 --> 15 83 07 46 82 80 14 75 68 80 11 00 64 A8 78 93 00 00 00 00 00 00 00 00 0A D8 B8 
22:41:57.945 > Interrupt received
22:41:58.011 > RX Channel: 40 --> 95 83 07 46 82 83 07 46 82 01 00 01 80 01 00 01 6A 19 6A 19 00 00 00 00 40 8F DA | -80 dBm
22:41:58.073 > Interrupt received
22:41:58.203 > RX Channel: 40 --> 95 83 07 46 82 83 07 46 82 06 3F 7C 00 00 00 21 70 7C 00 12 3D 1E 41 E9 00 00 64 | -80 dBm
22:41:58.314 > Interrupt received
22:41:58.458 > RX Channel: 3 --> 95 83 07 46 82 83 07 46 82 8C 47 F6 00 00 00 05 B1 8D 91 | -80 dBm
22:41:58.661 > RX Period End
22:41:58.661 > Middle missing
22:41:58.661 > Request retransmit: 2
22:41:58.661 > TX RequestFrame Channel: 75 --> 15 83 07 46 82 80 14 75 68 82 5E 
22:41:58.840 > RX Period End
22:41:58.840 > Middle missing
22:41:58.840 > Request retransmit: 2
22:41:58.840 > TX RequestFrame Channel: 3 --> 15 83 07 46 82 80 14 75 68 82 5E 
22:41:58.962 > Interrupt received
22:41:59.024 > RX Channel: 40 --> 95 83 07 46 82 83 07 46 82 02 00 04 6A 21 6A 65 00 00 07 A3 40 91 00 05 6A 22 EF | -80 dBm
22:41:59.179 > RX Period End
22:41:59.179 > Middle missing
22:41:59.179 > Request retransmit: 3
22:41:59.179 > TX RequestFrame Channel: 23 --> 15 83 07 46 82 80 14 75 68 83 5F 
22:41:59.404 > Interrupt received
22:41:59.611 > RX Channel: 40 --> 95 83 07 46 82 83 07 46 82 03 6A 65 00 00 12 C0 40 7C 00 07 6D EF 73 62 00 00 E3 | -80 dBm
22:41:59.711 > RX Period End
22:41:59.711 > Middle missing
22:41:59.711 > Request retransmit: 4
22:41:59.711 > TX RequestFrame Channel: 40 --> 15 83 07 46 82 80 14 75 68 84 58 
22:42:00.018 > Interrupt received
22:42:00.096 > RX Channel: 61 --> 95 83 07 46 82 83 07 46 82 04 00 00 50 7C 00 0D 73 83 2C 7E 00 00 00 00 70 7C 1E | -80 dBm
22:42:00.162 > RX Period End
22:42:00.162 > Middle missing
22:42:00.162 > Request retransmit: 5
22:42:00.162 > TX RequestFrame Channel: 61 --> 15 83 07 46 82 80 14 75 68 85 59 
22:42:00.303 > RX Period End
22:42:00.303 > Middle missing
22:42:00.303 > Retransmit timeout
22:42:00.403 > [  10450.884] DPL: waiting for sufficiently recent inverter data
22:42:00.606 > PowerMeterClass: TotalPower: 313.08
22:42:01.503 > Fetch inverter: 116183074682
22:42:01.557 > TX RealTimeRunData Channel: 75 --> 15 83 07 46 82 80 14 75 68 80 0B 00 64 A8 78 98 00 00 00 00 00 00 00 00 E0 B0 2B 
22:42:01.733 > Interrupt received
22:42:01.836 > RX Channel: 23 --> 95 83 07 46 82 83 07 46 82 01 00 01 01 F0 00 D5 00 D2 04 1D 04 10 00 02 39 8B DE | -80 dBm
22:42:01.939 > Interrupt received
22:42:02.011 > RX Channel: 40 --> 95 83 07 46 82 83 07 46 82 84 13 8A 0F 8A 00 D3 00 AC 03 E7 01 9F 00 24 AA BE 38 | -80 dBm
22:42:02.067 > RX Period End
[... 542 lines cut ...]
22:42:57.281 > RX Period End
22:42:57.281 > Last missing
22:42:57.281 > Request retransmit: 4
22:42:57.281 > TX RequestFrame Channel: 75 --> 15 83 07 46 82 80 14 75 68 84 58 
22:42:57.382 > RX Period End
22:42:57.382 > Last missing
22:42:57.382 > Request retransmit: 4
22:42:57.382 > TX RequestFrame Channel: 3 --> 15 83 07 46 82 80 14 75 68 84 58 
22:42:57.543 > RX Period End
22:42:57.543 > Last missing
22:42:57.543 > Request retransmit: 4
22:42:57.543 > TX RequestFrame Channel: 23 --> 15 83 07 46 82 80 14 75 68 84 58 
22:43:01.803 > Interrupt received
22:43:08.968 > [PowerLimiterClass::loop] ******************* ENTER **********************
22:43:08.968 > [PowerLimiterClass::loop] dcVoltage: 49.60 Voltage Start Threshold: 50.00 Voltage Stop Threshold: 49.00 inverter->isProducing(): 1
22:43:08.968 > [HttpPowerMeter] Getting the power of phase 1 failed. Error: Error(http://shellyem3-34945474a444/status): read Timeout
22:43:08.968 > [HttpPowerMeter] Couldn't find a value with Json query "total_power"
22:43:08.968 > [PowerLimiterClass::loop] newPowerLimit: 424
22:43:08.968 > [PowerLimiterClass::setNewPowerLimit] 4 channels total, Interrupt received
22:43:09.247 > RX Channel: 3 --> 95 83 07 46 82 83 07 46 82 8C 47 F6 00 00 00 05 B1 8D 91 | -80 dBm
22:43:11.324 > RX Period End
22:43:11.325 > Middle missing
22:43:11.325 > Request retransmit: 2
22:43:11.325 > TX RequestFrame Channel: 3 --> 15 83 07 46 82 80 14 75 68 82 5E 
`

MalteSchm, 2023-07-07:
It stalled again. What is interesting is that the uptime counter did not indicate a reset.
Logs below

23:31:37.433 > [PowerLimiterClass::loop] ******************* ENTER **********************
23:31:37.433 > [PowerLimiterClass::loop] dcVoltage: 49.80 Voltage Start Threshold: 50.00 Voltage Stop Threshold: 49.00 inverter->isProducing(): 1
23:31:37.433 > [PowerLimiterClass::loop] useFullSolarPassthrough: FullSolarPT Start 100.000000, FullSolarPT Stop: 100.000000, dcVoltage: 49.799999
23:31:37.433 > [PowerLimiterClass::loop] newPowerLimit: 192
23:31:37.433 > [PowerLimiterClass::setNewPowerLimit] 4 channels total, 4 producing channels, scaling power limit
23:31:37.433 > [PowerLimiterCInterrupt received
23:31:37.924 > RX Channel: 40 --> 95 83 07 46 82 83 07 46 82 02 00 02 31 C7 01 33 01 2F 01 F2 00 57 00 55 01 B3 3C | -80 dBm
23:31:38.156 > RX Period End
23:31:38.156 > Middle missing
23:31:38.156 > Request retransmit: 3
23:31:38.156 > TX RequestFrame Channel: 23 --> 15 83 07 46 82 80 14 75 68 83 5F 
23:31:38.165 > RX Period End
23:31:38.165 > Middle missing
23:31:38.165 > Request retransmit: 3
23:31:38.165 > TX RequestFrame Channel: 40 --> 15 83 07 46 82 80 14 75 68 83 5F 
23:31:38.174 > RX Period End
23:31:38.174 > Middle missing
23:31:38.174 > Retransmit timeout
23:31:38.401 > TX AlarmData Channel: 61 --> 15 83 07 46 82 80 14 75 68 80 11 00 64 A8 84 36 00 00 00 00 00 00 00 00 A1 DA 48 
23:31:38.620 > Interrupt received
23:31:38.850 > RX Channel: 75 --> 95 83 07 46 82 83 07 46 82 04 00 00 50 7C 00 0D 73 83 2C 7E 00 00 00 00 70 7C 1E | -80 dBm
23:31:38.863 > Interrupt received
23:31:39.086 > RX Channel: 40 --> 95 83 07 46 82 83 07 46 82 06 41 E9 00 00 00 00 70 7C 00 15 41 FD 42 3C 00 00 E0 | -80 dBm
23:31:39.095 > Interrupt received
23:31:39.305 > RX Channel: 23 --> 95 83 07 46 82 83 07 46 82 09 7B 72 FF FF FF FA B0 02 00 26 7B 72 7B 72 00 00 04 | -80 dBm
23:31:39.531 > Interrupt received
23:31:39.762 > RX Channel: 75 --> 95 83 07 46 82 83 07 46 82 0B 00 28 7B B4 7B B4 00 00 00 05 B0 02 00 29 7C 50 04 | -80 dBm
23:31:40.000 > [PowerLimiterClass::loop] ******************* ENTER **********************
23:31:40.000 > [PowerLimiterClass::loop] dcVoltage: 49.80 Voltage Start Threshold: 50.00 Voltage Stop Threshold: 49.00 inverter->isProducing(): 1
23:31:40.000 > [PowerLimiterClass::loop] useFullSolarPassthrough: FullSolarPT Start 100.000000, FullSolarPT Stop: 100.000000, dcVoltage: 49.799999
23:31:40.000 > [PowerLimiterClass::loop] newPowerLimit: 190
23:31:40.000 > [PowerLimiterClass::setNewPowerLimit] 4 channels total, 4 producing channels, scaling power limit
23:31:40.000 > [PowerLimiterCRX Period End
23:31:40.010 > Last missing
23:31:40.010 > Request retransmit: 12
23:31:40.010 > TX RequestFrame Channel: 3 --> 15 83 07 46 82 80 14 75 68 8C 50 
23:31:40.236 > RX Period End
23:31:40.236 > Last missing
23:31:40.236 > Request retransmit: 12
23:31:40.236 > TX RequestFrame Channel: 23 --> 15 83 07 46 82 80 14 75 68 8C 50 
23:31:40.246 > Interrupt received
23:31:40.260 > RX Channel: 75 --> 95 83 07 46 82 83 07 46 82 8C 7C 50 FF FF FF F6 EE AE 7C | -80 dBm
23:31:40.490 > RX Period End
23:31:40.491 > Middle missing
23:31:40.491 > Request retransmit: 1
23:31:40.491 > TX RequestFrame Channel: 40 --> 15 83 07 46 82 80 14 75 68 81 5D 
23:31:40.503 > Interrupt received
23:31:40.512 > RX Channel: 75 --> 95 83 07 46 82 83 07 46 82 01 00 01 80 01 00 01 6A 19 6A 19 00 00 00 00 40 8F DA | -80 dBm
23:31:40.734 > RX Period End
23:31:40.734 > Middle missing
23:31:40.734 > Request retransmit: 2
23:31:40.734 > TX RequestFrame Channel: 61 --> 15 83 07 46 82 80 14 75 68 82 5E 
23:31:40.745 > Interrupt received
23:31:40.985 > RX Channel: 75 --> 95 83 07 46 82 83 07 46 82 02 00 04 6A 21 6A 65 00 00 07 A3 40 91 00 05 6A 22 EF | -80 dBm
23:31:40.995 > RX Period End
23:31:40.995 > Middle missing
23:31:40.995 > Retransmit timeout
23:31:41.230 > [PowerLimiterClass::loop] ******************* ENTER **********************
23:31:41.230 > [PowerLimiterClass::loop] dcVoltage: 49.80 Voltage Start Threshold: 50.00 Voltage Stop Threshold: 49.00 inverter->isProducing(): 1
23:31:41.230 > [PowerLimiterClass::loop] useFullSolarPassthrough: FullSolarPT Start 100.000000, FullSolarPT Stop: 100.000000, dcVoltage: 49.799999
23:31:41.230 > [PowerLimiterClass::loop] newPowerLimit: 190
23:31:41.230 > [PowerLimiterClass::setNewPowerLimit] 4 channels total, 4 producing channels, scaling power limit
23:31:41.230 > [PowerLimiterC[PowerLimiterClass::loop] ******************* ENTER **********************
23:31:41.452 > [PowerLimiterClass::loop] dcVoltage: 49.80 Voltage Start Threshold: 50.00 Voltage Stop Threshold: 49.00 inverter->isProducing(): 1
23:31:41.452 > [PowerLimiterClass::loop] useFullSolarPassthrough: FullSolarPT Start 100.000000, FullSolarPT Stop: 100.000000, dcVoltage: 49.799999
23:31:41.452 > [PowerLimiterClass::loop] newPowerLimit: 190
23:31:41.452 > [PowerLimiterClass::setNewPowerLimit] 4 channels total, 4 producing channels, scaling power limit
[... 268 lines cut ...]
23:32:06.082 > Interrupt received
23:32:06.307 > RX Channel: 75 --> 95 83 07 46 82 83 07 46 82 0B 00 28 7B B4 7B B4 00 00 00 05 B0 02 00 29 7C 50 04 | -80 dBm
23:32:06.533 > RX Period End
23:32:06.533 > Last missing
23:32:06.533 > Request retransmit: 12
23:32:06.533 > TX RequestFrame Channel: 40 --> 15 83 07 46 82 80 14 75 68 8C 50 
23:32:06.775 > Interrupt received
23:32:07.030 > RX Channel: 61 --> 95 83 07 46 82 83 07 46 82 8C 7C 50 FF FF FF F6 EE AE 7C | -80 dBm
[... cut ...]
23:43:32.402 > [PowerLimiterClass::loop] dcVoltage: 49.90 Voltage Start Threshold: 50.00 Voltage Stop Threshold: 49.00 inverter->isProducing(): 1
23:43:32.402 > [PowerLimiterClass::loop] useFullSolarPassthrough: FullSolarPT Start 100.000000, FullSolarPT Stop: 100.000000, dcVoltage: 49.900002
23:43:32.402 > [PowerLimiterClass::loop] newPowerLimit: 187
23:43:32.402 > [PowerLimiterClass::setNewPowerLimit] 4 channels total, 4 producing channels, scaling power limit
23:43:32.402 > [PowerLimiterCFetch inverter: 116183074682

spcqike, 2023-07-08:
Sounds like a heap problem which was already mentioned here 
https://github.com/helgeerbe/OpenDTU-OnBattery/issues/268#issuecomment-1611727536

is it possible that you have (had?) some of these „error 2“ errors in your inverter log? 

I habe the feeling that this is related somehow. 

It looks like Whenever „error2“ occurs, I get a problem with heap, to that point that the webUI can’t be shown and that the esp seems to stop working.

yesterday I had 15 of said errors until 8:14am, I checked and had a nearly full heap (like 70% or so) but still could reach and control the DTU. I still could restart it and after that the heap was normal again. No new errors occurred.

MalteSchm, 2023-07-08:
I'm not getting error 2
Right now I'm seeing esp restarts. My guess would be every minute or so.
I'll now swap the nrf tranmitter (I anyhow wanted to do that) and then try if I can isolate it somehow

MalteSchm, 2023-07-08:
Looking through the logs I'm also seeing this:

19:28:26.277 > [PowerLimiterClass::setNewPowerLimit] 4 channels total, 3 producing channels, scaling power limit

I have all channels connected to the same battery. This is another bug

MalteSchm, 2023-07-09:
Disabled debug messages and also the Hoymiles printouts and things seem to work stable (for the time I tested ~14h)
I created a smaller PR here: https://github.com/helgeerbe/OpenDTU-OnBattery/pull/301

helgeerbe, 2023-07-09:
I checked the log and see an unhandled exception, when printing a debug message. But the exception occurs in hoymiles in an abstract class. I don't know right now, why a frequent call to the inverter statistics results in a crash.

````cmd
12:52:56.006 > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
12:52:56.017 > 
12:52:56.017 > Core  1 register dump:
12:52:56.017 > PC      : 0x401d2577  PS      : 0x00060430  A0      : 0x800e2706  A1      : 0x3ffb2130  
12:52:56.023 > A2      : 0x00000000  A3      : 0x0000000a  A4      : 0x3ffc8e80  A5      : 0x000001f4  
12:52:56.034 > A6      : 0x00000003  A7      : 0x00060423  A8      : 0x80084184  A9      : 0x3ffb20c0  
12:52:56.040 > A10     : 0x3ffcf850  A11     : 0x3ffdddd4  A12     : 0x00000076  A13     : 0x3ffb1f50  
12:52:56.051 > A14     : 0x00000076  A15     : 0x3ffb1f50  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
12:52:56.056 > EXCVADDR: 0x00000234  LBEG    : 0x4008c069  LEND    : 0x4008c079  LCOUNT  : 0xffffffff  
12:52:56.062 > 
12:52:56.067 > 
12:52:56.067 > Backtrace: 0x401d2574:0x3ffb2130 0x400e2703:0x3ffb2150 0x400e297a:0x3ffb2180 0x400e388d:0x3ffb21c0 0x40100150:0x3ffb2270 0x401275a5:0x3ffb2290
12:52:56.233 > 
12:52:56.233 >   #0  0x401d2574:0x3ffb2130 in InverterAbstract::Statistics() at lib/Hoymiles/src/inverters/InverterAbstract.cpp:125
12:52:56.233 >   #1  0x400e2703:0x3ffb2150 in PowerLimiterClass::getLoadCorrectedVoltage(std::shared_ptr<InverterAbstract>) at src/PowerLimiter.cpp:539
12:52:56.233 >   #2  0x400e297a:0x3ffb2180 in PowerLimiterClass::isStartThresholdReached(std::shared_ptr<InverterAbstract>) at src/PowerLimiter.cpp:566
12:52:56.233 >   #3  0x400e388d:0x3ffb21c0 in PowerLimiterClass::loop() at src/PowerLimiter.cpp:281
12:52:56.233 >   #4  0x40100150:0x3ffb2270 in loop() at src/main.cpp:206
````

As a workaround disable the debug messages might work. But I wonder, what the root cause is.

schlimmchen, 2023-07-09:
I followed this closely but did not have a chance to look at it in detail until now. Thanks for finding the exception, @helgeerbe. 

[removed rumbling about not being able to make sense of the exception]

The exception shows that the error occured in the DPL loop, but only *after* a new power limit was calculated. And indeed, the only assignment `_inverter = nullptr` is in `shutdown()`, and `shutdown()` might be called when a new power limit is calculated (DPL `Status::LowerLimitUndercut`).

My best guess is that the inverter was not producing, a new limit was calculated, the new limit was still less than the lower power limit, so `shutdown(Status::LowerLimitUndercut)` was called, `_inverter` was reset and afterwarts, the nullptr is used when printing the debug messages.

Turns out my system uptime is only a couple of minutes. Probably the error did occur on my system as well, but I did not notice.

MalteSchm, 2023-07-09:
Only slightly related: I asked for guidance regarding how to get a better message output https://github.com/tbnobody/OpenDTU/discussions/1130
since the Hoymiles messages and the DPL messages always interfere.

MalteSchm, 2023-07-09:
Thanks @schlimmchen , @helgeerbe 

Seeing the fix now could mean that I only accidentally solved this by no longer executing the code when disabling the debug messages

schlimmchen, 2023-07-09:
> I asked for guidance regarding how to get a better message output https://github.com/tbnobody/OpenDTU/discussions/1130

Thanks, I joined the discussion.

> Seeing the fix now could mean that I only accidentally solved this by no longer executing the code when disabling the debug messages

Exactly.

> I still think its good to have them off by default

Yes. Users should be able to enable the messages easily, though. See my comment in #301.

helgeerbe, 2023-07-09:
Should be fixed with PR #303 in release 2023.07.09

closed 2023-07-09 by helgeerbe