hoylabs/OpenDTU-OnBattery #300: Wechselrichter Neustart - Stunde fuer Neustart - funktioniert nicht?

Manos1966, 2023-07-09:
### What happened?

I have the suspicion, the Restart of the Inverter does not happen as planned.
**Can somebody test whether this is a bug or not?**

Initially I had the Inverter Restart programmed at 00:00
I decided to change it early this morning (at 01:00) to 08:00

My pictures show: 
Restart planned for 08:00
Live Daten today (11:30) show a Daily Production of 16,43 kWh which is NOT POSSIBLE 
Inverter log shows events as of 00:00 thus no restart happened at 08:00 (events before 08:00 would have been deleted).

[image]
[image]

### To Reproduce Bug

set Inverter Restart at 08:00 and see if it happens.
(I presume that you could try also other times).

### Expected Behavior

Restart of the Inverter should happen at the defined time.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

344498d

schlimmchen, 2023-07-09:
This might be easy to explain, thanks to your detailed description: The restart was setup for 0:00 and the DPL calculated a timestamp at which it would reboot the inverter. In your case: midnight the next night. Before that event, you changed the restart time to 8:00. The DPL already had calculated the timestamp for the next restart, so your new setting was overlooked. The DPL will restart your inverter at midnight, then recalculate the next inverter restart timestamp, and the inverter will be restarted again at 8:00 tomorrow, eight hours later.

You are asking for the DPL to make sure that a changed setting is respected immediately (without reboot). That's fair.

Please observe your setup. If my predictions turn out to be true, then your issue is indeed that the setting is not picked up on change.

Manos1966, 2023-07-10:
Thanks @schlimmchen you were right!

I had changed the restart for 13:00 during my tests, 24hrs later I see the restart happening at 13:00 👍 
 
[image]

closed 2023-07-10 by Manos1966

1-am-r00t, 2023-07-10:
I have the same behavior since I've upgraded to one of the recent releases (344498d). I had the restart set to 04:00 ever since I started using this fork of OpenDTU and it has been working fine.

madmartin, 2023-07-10:
Hi,
I have written this feature, and I can tell you that the "time-to-next-restart" is calculated
- on ESP startup, after system time has been set from NTP server
- after saving the DPL parameters (e.g. when you change the restart hour)
- after the inverter restart

I assume @Manos1966 your observation was a single fault, due to whatever. 
If you continue to observe this, please record the console log, filter it for `[PowerLimiterClass::loop]` and post it here.

1-am-r00t, 2023-07-11:
> Hi, I have written this feature, and I can tell you that the "time-to-next-restart" is calculated
>
>     * on ESP startup, after system time has been set from NTP server
>
>     * after saving the DPL [...]

thanks for the follow up! I tried to set the restart time to 3:00 and observed the console log within OpenDTU GUI - no PowerLimiterClass entries can be seen and the restart did not happen. The NTP time is synchronized.

find below the relevant timeframe from the console log:

02:59:53.660 > PowerMeterClass: TotalPower: 212.00
02:59:53.660 > Fetch inverter: 114174206224
02:59:53.660 > Request device info
02:59:53.697 > TX RealTimeRunData Channel: 23 --> 15 74 20 62 24 80 17 14 32 80 0B 00 64 AC A9 89 00 00 00 00 00 00 00 00 EC 30 09 
02:59:53.711 > Interrupt received
02:59:53.718 > RX Channel: 61 --> 95 74 20 62 24 74 20 62 24 01 00 01 00 ED 00 03 00 06 00 ED 00 02 00 06 00 06 92 | -30 dBm
02:59:53.791 > Interrupt received
02:59:53.798 > RX Channel: 40 --> 95 74 20 62 24 74 20 62 24 02 A2 B4 00 03 67 B9 0E 67 0D A9 09 1B 13 88 00 00 18 | -80 dBm
02:59:53.840 > Interrupt received
02:59:53.849 > RX Channel: 23 --> 95 74 20 62 24 74 20 62 24 83 00 00 00 00 00 00 01 26 02 B9 F1 11 6A | -80 dBm
02:59:54.072 > [     10.670] DPL: waiting for inverter device information to be available
02:59:54.243 > RX Period End
02:59:54.243 > Success
02:59:54.251 > TX AlarmData Channel: 40 --> 15 74 20 62 24 80 17 14 32 80 11 00 64 AC A9 89 00 00 00 00 00 00 00 00 36 2B D2 
02:59:54.412 > Interrupt received
02:59:54.421 > RX Channel: 75 --> 95 74 20 62 24 74 20 62 24 03 71 02 00 00 00 00 41 2F 00 2B 64 A9 2D C9 0A C9 4A | -80 dBm
02:59:54.451 > Interrupt received
02:59:54.459 > RX Channel: 40 --> 95 74 20 62 24 74 20 62 24 04 02 1B 70 7C 00 2D 5F E1 60 12 00 00 00 00 70 7C 69 | -80 dBm
02:59:54.501 > Interrupt received
02:59:54.508 > RX Channel: 3 --> 95 74 20 62 24 74 20 62 24 05 00 55 6C 6A 6C 81 00 00 00 00 60 7C 01 0F 6D D4 85 | -80 dBm
02:59:54.551 > Interrupt received
02:59:54.560 > RX Channel: 75 --> 95 74 20 62 24 74 20 62 24 06 33 04 00 00 00 00 40 7C 01 11 33 10 33 5C 00 00 C4 | -80 dBm
02:59:54.601 > Interrupt received
02:59:54.608 > RX Channel: 61 --> 95 74 20 62 24 74 20 62 24 07 00 00 40 7C 01 13 33 67 34 6E 00 00 00 00 60 7C AE | -80 dBm
02:59:54.641 > Interrupt received
02:59:54.650 > RX Channel: 23 --> 95 74 20 62 24 74 20 62 24 08 02 21 5C 4C 35 81 00 00 00 00 40 7C 02 23 61 32 54 | -80 dBm
02:59:54.691 > Interrupt received
02:59:54.700 > RX Channel: 23 --> 95 74 20 62 24 74 20 62 24 09 61 5F 00 00 00 00 20 7C 02 36 47 2D 00 00 00 00 A0 | -80 dBm
02:59:54.762 > Interrupt received
02:59:54.778 > RX Channel: 40 --> 95 74 20 62 24 74 20 62 24 0A 00 00 80 02 02 AB 01 58 01 58 00 00 00 05 80 02 33 | -80 dBm
02:59:54.803 > Interrupt received
02:59:54.810 > RX Channel: 75 --> 95 74 20 62 24 74 20 62 24 0B 02 AC 01 8D 01 8D FF FF FF FB 80 02 02 AD 01 98 80 | -80 dBm
02:59:54.841 > Interrupt received
02:59:54.853 > RX Channel: 61 --> 95 74 20 62 24 74 20 62 24 8C 01 98 00 00 00 05 6E AC 47 | -30 dBm
02:59:55.018 > RX Period End
02:59:55.018 > Middle missing
02:59:55.018 > Request retransmit: 1
02:59:55.018 > TX RequestFrame Channel: 61 --> 15 74 20 62 24 80 17 14 32 81 37 
02:59:55.071 > Interrupt received
02:59:55.080 > RX Channel: 3 --> 95 74 20 62 24 74 20 62 24 01 00 01 80 01 00 01 1C 39 1C 39 00 00 00 00 40 7C 29 | -80 dBm
02:59:55.126 > RX Period End
02:59:55.127 > Middle missing
02:59:55.127 > Request retransmit: 2
02:59:55.127 > TX RequestFrame Channel: 75 --> 15 74 20 62 24 80 17 14 32 82 34 
02:59:55.242 > RX Period End
02:59:55.242 > Middle missing
02:59:55.242 > Request retransmit: 2
02:59:55.242 > TX RequestFrame Channel: 3 --> 15 74 20 62 24 80 17 14 32 82 34 
02:59:55.346 > RX Period End
02:59:55.346 > Middle missing
02:59:55.346 > Request retransmit: 2
02:59:55.346 > TX RequestFrame Channel: 23 --> 15 74 20 62 24 80 17 14 32 82 34 
02:59:55.456 > RX Period End
02:59:55.456 > Middle missing
02:59:55.456 > Request retransmit: 2
02:59:55.456 > TX RequestFrame Channel: 40 --> 15 74 20 62 24 80 17 14 32 82 34 
02:59:55.514 > Interrupt received
02:59:55.524 > RX Channel: 75 --> 95 74 20 62 24 74 20 62 24 02 00 16 1C 51 43 FD 00 00 00 00 40 7C 00 18 70 EA CC | -80 dBm
02:59:55.581 > RX Period End
02:59:55.581 > Success
[... 89 lines cut ...]
03:00:09.857 > Request retransmit: 2
03:00:09.857 > TX RequestFrame Channel: 75 --> 15 74 20 62 24 80 17 14 32 82 34 
03:00:09.966 > RX Period End
03:00:09.966 > Middle missing
03:00:09.966 > Request retransmit: 2
03:00:09.966 > TX RequestFrame Channel: 3 --> 15 74 20 62 24 80 17 14 32 82 34 
03:00:10.076 > RX Period End
03:00:10.076 > Middle missing
03:00:10.076 > Request retransmit: 2
03:00:10.076 > TX RequestFrame Channel: 23 --> 15 74 20 62 24 80 17 14 32 82 34 
03:00:10.182 > RX Period End
03:00:10.182 > Middle missing
03:00:10.182 > Retransmit timeout
03:00:10.210 > TX DevInfoAll Channel: 40 --> 15 74 20 62 24 80 17 14 32 80 01 00 64 AC A9 97 00 00 00 00 00 00 00 00 46 BB 3C 
03:00:10.271 > Interrupt received
03:00:10.283 > RX Channel: 3 --> 95 74 20 62 24 74 20 62 24 81 27 18 07 E4 02 C6 04 EA 00 68 00 00 00 00 5D EB 3C | -80 dBm
03:00:10.338 > Received MQTT message on topic: smartmeter/powerRateLimiter
03:00:10.338 > PowerMeterClass: TotalPower: 245.00
03:00:10.430 > RX Period End
03:00:10.430 > Success
03:00:10.449 > TX DevInfoSimple Channel: 61 --> 15 74 20 62 24 80 17 14 32 80 00 00 64 AC A9 97 00 00 00 00 00 00 00 00 87 BB FC 
03:00:10.481 > Interrupt received
03:00:10.489 > RX Channel: 23 --> 95 74 20 62 24 74 20 62 24 81 27 18 10 11 40 00 01 00 0A 00 20 01 00 00 8F A5 6A | -80 dBm
03:00:10.662 > RX Period End
03:00:10.662 > Success

schlimmchen, 2023-07-11:
Thanks for providing the log.

> 03:00:08.227 > [ 10.702] DPL: waiting for inverter device information to be available

The DPL is dormant since the inverter cannot provide `_inverter->DevInfo()->getMaxPower()`. You should habe worse problems than the inverter not being restarted, the DPL is not working at all?!

However, I see DevInfoSimple commands being sent and two answers being received. What inverter are you using? It is possibly different than the Hoymiles HM Series and may not provide its maximum power output capability. In that case the DPL must deal with that situation instead of insisting on the value to be available.

schlimmchen, 2023-07-11:
@1-am-r00t It seems unlikely that your inverter does not provide `getMaxPower()`, because the calculated power limit is always limited to `getMaxPower()`, at least since 2023-04-04. Have you upgraded from a version older than the beginning of April and started noticing problems just now?

schlimmchen, 2023-07-11:
That's weird... My device processes nearly two power meter updates per second from three topics (phases). You seem to subscribe to only one and you suggest that more updates than one every five seconds is too much. Do you use MQTT with authentication? Retain flag enabled? TLS enabled? Home Assistent auto discovery enabled?

@helgeerbe is correct, you should update to the most recent release, but I have doubts that this will fix your issue. Is your battery empty at 3 in the morning, when you expect the restart to happen? How many inverters do you watch with this OpenDTU?

1-am-r00t, 2023-07-11:
> That's weird... My device processes nearly two power meter updates per second from three topics (phases). You seem to subscribe to only one and you suggest that more updates than one every five seconds is too much.

I've made this assumption based on my observations ever since I started using OpenDTU on battery.

> Do you use MQTT with authentication? Retain flag enabled? TLS enabled? Home Assistent auto discovery enabled?

No to all of your questions

> you should update to the most recent release, but I have doubts that this will fix your issue.

I'm doubting it too, but for what its worth, I can give it a try.

> Is your battery empty at 3 in the morning, when you expect the restart to happen?

Actually is not empty because in DPL I've specified a low voltage limit of 23.5V which forces the HM to stop producing.

> How many inverters do you watch with this OpenDTU?

Only my HM-800 with both channels attached to the battery.

1-am-r00t, 2023-07-11:
> Okay, so your battery is not empty, but let's say it is depleted, and the inverter is stopped always or often at the time around which the inverter shall restart?

The battery usually is empty by midnight. I've chosen to set the restart of the inverter to 4:00 because it's a sweetspot where the battery will be definitely "empty" and sunrise is about to happen.

> If that's the case, then the nullptr access is probably the reason that your inverter is not restarted: shutdown() is called in every DPL loop that makes it to the calcPowerLimit() at all and triggers the bug every time. The ESP restarts every time before the DPL loop is executed a second time. Please upgrade.

For a strange reason the inverter was no longer being polled at all when its not producing (i.e. lower voltage limit). I then proceeded and removed the inverter from the polling list and re-added it using copy/paste of the SN followed by a quick restart. It was still not polling any data at all from the inverter. I then proceeded and manually disabled all polling options in the GUI followed again by a quick restart. It still didnt want to poll any data (like in my logfiles from yesterday). I then proceeded with the suggested upgrade and guess what; the SN suddenly did not match with what I entered/pasted before (should be 114174206224, but was in fact something like 114174206116). Once I corrected the SN to the correct one, the polling immediately is working again and also the powerlimiterclass is now showing up in logging:

22:49:27.614 > [PowerLimiterClass::loop] ******************* ENTER **********************
22:49:27.614 > [PowerLimiterClass::loop] dcVoltage: 24.00 Voltage Start Threshold: 25.50 Voltage Stop Threshold: 23.50 inverter->isProducing(): 0
22:49:27.614 > [    805.451] DPL: calculated power limit undercuts configured lower limit
22:49:27.614 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
22:49:27.614 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 805478[    805.502] DPL: the system is stable, the last power limit is still valid
22:49:27.622 > RX Channel: 75 --> 95 74 20 62 24 74 20 62 24 03 6A F7 00 00 00 00 20 7C 00 37 3E 07 00 00 00 00 59 | -80 dBm
22:49:27.650 > RX Period End
22:49:27.650 > Middle missing
22:49:27.650 > Retransmit timeout
22:49:27.799 > [PowerLimiterClass::loop] ******************* ENTER **********************
22:49:27.799 > [PowerLimiterClass::loop] dcVoltage: 24.00 Voltage Start Threshold: 25.50 Voltage Stop Threshold: 23.50 inverter->isProducing(): 0
22:49:27.799 > [    805.638] DPL: calculated power limit undercuts configured lower limit
22:49:27.799 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
22:49:27.799 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 805654[    805.677] DPL: the system is stable, the last power limit is still valid
22:49:27.985 > [PowerLimiterClass::loop] ******************* ENTER **********************
22:49:27.985 > [PowerLimiterClass::loop] dcVoltage: 24.00 Voltage Start Threshold: 25.50 Voltage Stop Threshold: 23.50 inverter->isProducing(): 0
22:49:27.985 > [    805.815] DPL: calculated power limit undercuts configured lower limit
22:49:27.985 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
22:49:27.985 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 805831[    805.854] DPL: the system is stable, the last power limit is still valid
22:49:28.142 > [PowerLimiterClass::loop] ******************* ENTER **********************
22:49:28.142 > [PowerLimiterClass::loop] dcVoltage: 24.00 Voltage Start Threshold: 25.50 Voltage Stop Threshold: 23.50 inverter->isProducing(): 0
22:49:28.142 > [    805.985] DPL: calculated power limit undercuts configured lower limit
22:49:28.142 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
22:49:28.142 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 806011[    806.034] DPL: the system is stable, the last power limit is still valid
22:49:28.318 > [PowerLimiterClass::loop] ******************* ENTER **********************
22:49:28.318 > [PowerLimiterClass::loop] dcVoltage: 24.00 Voltage Start Threshold: 25.50 Voltage Stop Threshold: 23.50 inverter->isProducing(): 0
22:49:28.318 > [    806.173] DPL: calculated power limit undercuts configured lower limit
22:49:28.318 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
22:49:28.318 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 806189[    806.212] DPL: the system is stable, the last power limit is still valid
22:49:28.495 > [PowerLimiterClass::loop] ******************* ENTER **********************
22:49:28.495 > [PowerLimiterClass::loop] dcVoltage: 24.00 Voltage Start Threshold: 25.50 Voltage Stop Threshold: 23.50 inverter->isProducing(): 0
22:49:28.495 > [    806.350] DPL: calculated power limit undercuts configured lower limit
22:49:28.495 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
[... cut ...]
22:49:31.329 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
22:49:31.329 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 809167[    809.224] DPL: the system is stable, the last power limit is still valid

Honestly I dont know what was going on, but I suppose the error is gone now since the polling is working again. I will report back tomorrow morning/noon if the restart at 4:00 was a success (which I suppose). Thanks for your help/input so far!

1-am-r00t, 2023-07-12:
Just a quick follow up: restart worked as intended. Somehow I'm still asking how that thing with the SN happened. I'll wait for the next update and open a new issue in case it'll happen again. Thanks for your contribution/help :)