hoylabs/OpenDTU-OnBattery #305: [Request] On/Off Toggle for Huawei PSU in webinterface

indie89, 2023-07-10:
### Is your feature request related to a problem? Please describe.

I love the option to turn on/off the PSU via Huawei slot-detect and use it heavily over mqtt in order to control my battery charging. Unfortunately there is no option to "manually" toggle this over the webinterface.

### Describe the solution you'd like

I'd love to have a on/off toggle located in the voltage/power control for the PSU in the webinterface.

MalteSchm, 2023-07-17:
Hi @indie89 

I recently implemented automatic power control and made this the default. So when you set a current over 1A this should enable the PSU. Make sure to set huawei/cmd/mode to "2" in your MQTT scripts if you want to use this.
If you set the online current to <1A the PSU will shutdown after 60 seconds (if I recall correctly)

Let me know if you have any issues.

indie89, 2023-07-18:
Thanks, Malte! As I decided to eliminate the slot detect off switch, I can't test that, but I trust you, that it works and that is a sufficient solution for me (and others).

closed 2023-07-18 by indie89