hoylabs/OpenDTU-OnBattery #308: Stability issue

helgeerbe, 2023-07-12:
### What happened?

I see 1-2 unplanned restarts per day.

See prometheus log:
[image]

This time I catched  the error in the log. Or to be more precisely,  there was not log output at all. Nearly 15 minutes no output to the log. Not even the reboot. Only uptime indicates a restart.

### To Reproduce Bug

No idea. Just wait

### Expected Behavior

No restart

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

1d559c1

### Relevant log/trace output

```Shell
03:59:03.730 > RX Period End
03:59:03.736 > All missing
03:59:03.736 > Nothing received, resend count exeeded
03:59:03.860 > [PowerLimiterClass::loop] ******************* ENTER **********************
03:59:03.866 > [PowerLimiterClass::loop] dcVoltage: 22.80 Voltage Start Threshold: 26.40 Voltage Stop Threshold: 21.60 inverter->isProducing(): 0
03:59:03.877 > [  22568.766] DPL: calculated power limit undercuts configured lower limit
03:59:03.883 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
03:59:03.894 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 22568783
03:59:03.899 > [PowerLimiterClass::loop] ******************* Leaving PL, PL set to: 0, SP: 0, Batt: 0, PM: 208.000000
03:59:03.910 > TX SystemConfigPara Channel: 75 --> 15 81 01 96 07 80 16 36 20 80 05 00 64 AE 08 E0 00 00 00 00 00 00 00 00 8F A1 0D 
03:59:03.961 > [  22568.857] DPL: the system is stable, the last power limit is still valid
03:59:04.031 > [PowerLimiterClass::loop] ******************* ENTER **********************
03:59:04.037 > [PowerLimiterClass::loop] dcVoltage: 22.80 Voltage Start Threshold: 26.40 Voltage Stop Threshold: 21.60 inverter->isProducing(): 0
03:59:04.237 > [  22569.133] DPL: calculated power limit undercuts configured lower limit
03:59:04.244 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
03:59:04.255 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 22569140
03:59:04.260 > [PowerLimiterClass::loop] ******************* Leaving PL, PL set to: 0, SP: 0, Batt: 0, PM: 207.000000
03:59:04.272 > RX Period End
03:59:04.272 > All missing
03:59:04.272 > Nothing received, resend whole request
03:59:04.277 > TX SystemConfigPara Channel: 3 --> 15 81 01 96 07 80 16 36 20 80 05 00 64 AE 08 E0 00 00 00 00 00 00 00 00 8F A1 0D 
03:59:04.329 > [  22569.225] DPL: the system is stable, the last power limit is still valid
03:59:04.398 > [PowerLimiterClass::loop] ******************* ENTER **********************
03:59:04.404 > [PowerLimiterClass::loop] dcVoltage: 22.80 Voltage Start Threshold: 26.40 Voltage Stop Threshold: 21.60 inverter->isProducing(): 0
03:59:04.415 > [  22569.302] DPL: calculated power limit undercuts configured lower limit
03:59:04.420 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
04:15:42.713 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, Star�[PowerLimiterClass::loop] ******************* ENTER **********************
04:15:42.765 > [PowerLimiterClass::loop] dcVoltage: 22.80 Voltage Start Threshold: 26.40 Voltage Stop Threshold: 21.60 inverter->isProducing(): 0
04:15:42.776 > [    626.337] DPL: calculated power limit undercuts configured lower limit
04:15:42.782 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
04:15:42.792 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 626354
04:15:42.797 > [PowerLimiterClass::loop] ******************* Leaving PL, PL set to: 0, SP: 0, Batt: 0, PM: 378.000000
04:15:42.808 > [    626.384] DPL: the system is stable, the last power limit is still valid
04:15:42.940 > [PowerLimiterClass::loop] ******************* ENTER **********************
04:15:42.946 > [PowerLimiterClass::loop] dcVoltage: 22.80 Voltage Start Threshold: 26.40 Voltage Stop Threshold: 21.60 inverter->isProducing(): 0
04:15:42.957 > [    626.518] DPL: calculated power limit undercuts configured lower limit
04:15:42.964 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
04:15:42.974 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 626544
04:15:42.979 > [PowerLimiterClass::loop] ******************* Leaving PL, PL set to: 0, SP: 0, Batt: 0, PM: 378.000000
04:15:42.997 > [    626.564] DPL: the system is stable, the last power limit is still valid
04:15:43.118 > [PowerLimiterClass::loop] ******************* ENTER **********************
04:15:43.123 > [PowerLimiterClass::loop] dcVoltage: 22.80 Voltage Start Threshold: 26.40 Voltage Stop Threshold: 21.60 inverter->isProducing(): 0
04:15:43.135 > [    626.695] DPL: calculated power limit undercuts configured lower limit
04:15:43.140 > [PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 0
04:15:43.151 > [PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 80, StopTH: 20, LastUpdate: 626712
04:15:43.157 > [PowerLimiterClass::loop] ******************* Leaving PL, PL set to: 0, SP: 0, Batt: 0, PM: 378.000000
04:15:43.173 > [    626.742] DPL: the system is stable, the last power limit is still valid
04:15:43.201 > RX Period End
04:15:43.201 > All missing
```

spcqike, 2023-07-12:
is this the output of webui console?

do you have several of these restarts in your influx database? if so, can you check the last free heap value before the restart occurs? it looks like there is a measurement for heap at ~4:07, but not for uptime. if both values are logged from mqtt output, mqtt.loop() must have run at ~4:07.  this freeheap value looks like its the lowest within the record.
the free heap from 4:07 looks a little lower than from ~3:25. maybe there is some kind of threshold where there is not enough heap for a certain task?

i played around with the httppowermeter and increased the ```    char response[2000],``` to ```4000``` which also resulted in permanent crashes, whenever ```bool HttpPowerMeterClass::updateValues()```was called. even tho an additional ```Serial.println(ESP.getFreeHeap());``` showed ~150kB free heap.

helgeerbe, 2023-07-12:
> is this the output of webui console?

No, its the USB serial log

Log is taken from the prometheus interface, not MQTT. All data is pulled by telegraf in a single call every 5s.

Due to memory fragmentation a freeHeap of 150kB doesn't guarantees that you can get a free block of 4000 bytes. That's the reason why I limited the max supported inverters from 10 to 5. The json structure for the live view and prometheus response become so large, that after a while, no free memory block at this size could be requested.

schlimmchen, 2023-07-13:
That is a valid point and would explain why no error message is produced before the device restarts. My guess, however, is that @helgeerbe has that under control. Since his device is connected to a host PC to capture the logs, the only faulty bit in this setup might be the USB cable. There are indeed plenty of very bad cables around.

I can only add that my openDTU (fusion v2 board) is running fine for 47 hours now.

helgeerbe, 2023-07-14:
To be honest. At the moment I have no idea what went wrong. I use the same setup nearly a year and never had seen something like this.  

My power supply is already stabilized with an onboard capacitor and in the past everything was very stable. Even if there are any power stability issues I would expect to see the start log on my usb serial log. 

But the log stopped and started again without any hint. In between I see that the uptime was resetted.

I put some `heap_caps_check_integrity_all(true)`calls into the code to see if I can find some heap corruptions. But since 18 hours it's up and running.

The only thing I noticed that my fritz box and wifi repeater did updates and the wifi became unstable.  I will keep this issue open for the rest of the day. If nothing happens I will close it as a one time, unreproducible hick up.

MalteSchm, 2023-07-18:
I have a similar observation. Currently the system restarts maybe once a day. Yesterday the uptime was 1d and a few hours but I had a restart tonight.
I'm guessing here on the reasons for this but wifi stability could be an issue in my case

helgeerbe, 2023-07-18:
Is it just me and @MalteSchm?  Malte are you capable to catch a log from usb serial?

schlimmchen, 2023-07-18:
I still have no issues, uptime is 3d 13h. I have no WiFi issues (using an external antenna on the opendtufusionv2).

spcqike, 2023-07-18:
I guess one must also compare the used components when talking about stability, mustn’t one? 

I have one DTU with nrf24l01+ (small PCB without external Antenne) and an oled soldered to a perfboard. Version https://github.com/helgeerbe/OpenDTU-OnBattery/commits/3af9360. Using just WiFi and one HM-600. No mqtt, no battery, no power meter or limiter. Uptime 18 days. Rarely “error 2” in the log or heap errors or unusable live view (heap and live view only occurred once or twice) 

I have another DTU on perfboard with nrf24l01+ with external antenna and oled. Using power meter (https+json) and MQTT. No battery or limiter right now. Version https://github.com/helgeerbe/OpenDTU-OnBattery/commits/3ac6075. Uptime 4days. Last restart was random reset on it’s own. 
frequent error 2, sometimes heap and live view problems. Stable for the last 4 days.

indie89, 2023-07-18:
> Is it just me and @MalteSchm? Malte are you capable to catch a log from usb serial?

I also have restarts. Still, with adding a capacitor to the power supply, the stability increased enormously! And the restarts don't lead to errors for it is obviously up and running again quickly enaugh. 
I just control a Huawei R4850 with "on battery" by now.

MalteSchm, 2023-07-18:
> Is it just me and @MalteSchm? Malte are you capable to catch a log from usb serial?

Well I could try to set this up on Thursday. It is not straightforward as the stuff is in the garage.

Manos1966, 2023-07-22:
Hi,
I want to share something that I noticed yesterday by coincidence. I did not collect any information as I did not understand that it was a problem, until it was too late.
I am using Firmwareversion / git Hash | 344498d and for one week it has been running without problems.
I have no ELKO, KEKO installed on my OpenDTU as I am still testing.

Yesterday I returned from a trip and noticed that for the past day 20th July), the inverter had stopped producing:
[image]
I thought it was an inverter problem so, I restarted the inverter. 
After the restart, I noticed a strange behaviour:
The inverter (HM-1500) was producing at max power 1448 W (despite the fact my Dynamic Power Limiter has max 950W and min 10W)
The absolute limit was at 9W
The relative limit was at 100%
Then I realised that the absolute limit value (9W) was not updating (see the last date on the right column).
[image]

After a software reboot of the OpenDTU everything  started working again normally.

I am not complaining, I just share this information in case it is helpful

Feldsalat, 2023-07-24:
Ich habe auch solche Neustarts. Gerade erst, mit seltsamer Auswirkung:
[image]

Limit 46W / 15%  aber Inverterleistung nur 15W. 
Das blieb etwa eine halbe Stunde so, dann ging die Leistung auf 50W wie mittlerweile gefordert. Dazwischen wieder ein Neustart der DTU.

Desweiteren bleibt der Limiter im Solar-Passthrough, wo er gar nicht sein dürfte. Das ist seit heute Morgen der Fall. Start steht auf 27,3V und Stop auf 26,9V

14:41:48.323 > [PowerLimiterClass::loop] ******************* ENTER **********************
14:41:48.324 > [PowerLimiterClass::loop] dcVoltage: 26.80 Voltage Start Threshold: 27.00 Voltage Stop Threshold: 25.60 inverter->isProducing(): 1
14:41:48.324 > [PowerLimiterClass::loop] Consuming Solar Power Only -> adjustedVictronChargePower: 112, powerConsumption: 368
14:41:48.324 > [PowerLimiterClass::loop] newPowerLimit: 112
14:41:48.324 > [PowerLimiterClass::setNewPowerLimit] reusing old limit: 46 W, diff: 4, hysteresis: 5
14:41:48.324 > [PowerLimiterClass::loop] Status: SolarPT enabPowerMeterClass: TotalPower: 400.04
14:41:49.024 > [   1494.188] DPL: the system is stable, the last power limit is still valid

Das ist so extrem, seit ich viel an der Konfiguration geändert habe. Zufall?

helgeerbe, 2023-07-24:
@Feldsalat Das sieht in der Tat komisch aus. Machst du täglich einen Restart des Inversters?

Feldsalat, 2023-07-24:
Ja, das steht auf 0:00.

Feldsalat, 2023-07-26:
Hatte vorhin einen Totalausfall, mir war aufgefallen, das seit längerer Zeit keine Daten mehr gesendet werden. DTU war auch bei der Fritzbox nicht mehr angemeldet. Erst als ich den Rechner angestöpselt habe ist die DTU wieder gestartet. War so etwa 40 Minuten lang ausgefallen. Knappe Stunde später der nächste Neustart:

[PowerLimiterClass::setNewPowerLimit] reusing old limit: 70 W, diff: 2, hysteresis: 5
[PowerLimiterClass::loop] Status: SolarPT enabled 1, Drain Strategy: 0, canUseDirectSolarPower: 1, Batt discharge: 0
[PowerLimiterClass::loop] Status: StartTH 0, StopTH: 1, loadCorrectedV 26.772202
[PowerLimiterClass::loop] Status Batt: Ena: 0, SOC: 0, StartTH: 70, StopTH: 40, LastUpdate: 3042772
[PowerLimiterClass::loop] ******************* Leaving PL, PL set to: 68, SP: 1, Batt: 0, PM: 26.000000
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40120822  PS      : 0x00060c30  A0      : 0x800d8ce9  A1      : 0x3ffb0e40
A2      : 0x3ffb0f4c  A3      : 0x00000004  A4      : 0x3ffe2bcc  A5      : 0x3ffb81e8
A6      : 0x3f458721  A7      : 0x00000001  A8      : 0x80125998  A9      : 0x3ffb0e20
A10     : 0x3ffd6620  A11     : 0x80000001  A12     : 0x80092840  A13     : 0x3ffd8800
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000015  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000040  LBEG    : 0x4008a6b4  LEND    : 0x4008a6ca  LCOUNT  : 0xffffffff

Backtrace: 0x4012081f:0x3ffb0e40 0x400d8ce6:0x3ffb0e60 0x400d978d:0x3ffb1230 0x400e4f6d:0x3ffb2230 0x400e539f:0x3ffb2250 0x4010183f:0x3ffb2270 0x40128c61:0x3ffb2290

ELF file SHA256: 2a029f7d0b4e492f

E (1443) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (700) esp_core_dump_flash: No core du▒▒partition found!
E (700) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... done
Reading PinMapping... found valid mapping done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... WiFi disconnected
done
Initialize Display... done
Initialize LEDs... done
Check for default DTU serial... done
done
Initialize Hoymiles interface... NRF: Connection successful
  Setting radio PA level...
  Setting DTU serial...
  Setting poll interval...
  Setting verbosity...
  Adding inverter: 112184709155 - HM-300 done
done
Initialize ve.direct interface...
ve.direct rx = 22, tx = 21
done
Initialize Pylontech battery interface...
Pylontech Battery rx = 27, tx = 14
done
Initialize Huawei AC charger interface...
Huawei AWiFi connected
C-charger miso = 12, mosi = 13, clk = 26, irq = 25, cs = 15, power_pin = 33
done
Switch to WiFi mode
Setting Hostname... done
WiFi got ip: 192.168.0.92
Network connected
Connecting to MQTT...
Configuring WiFi STA using existing credentials... [   484][E][WiFiClient.cpp:268] connect(): socket error on fd 48, errno: 113, "Software caused connection abort"
Disconnected from MQTT.
Disconnect reason:TCP_DISCONNECTED
done
Configuring WiFi STA DHCP IP... done
[      0.512] DPL: waiting for inverter device information to be available
WiFi got ip: 192.168.0.92
Network connected
Connecting to MQTT...
Connected to MQTT.
Connecting to MQTT...
PowerMeterClass: TotalPower: 27.88
[     10.516] DPL: waiting for inverter device information to be available
Fetch inverter: 112184709155
Request SystemConfigPara
Middle missing
Request retransmit: 1
Success
Middle missing
Request retransmit: 4
Middle missing
Request retransmit: 5
Success
Success
PowerMeterClass: TotalPower: 27.96
[     20.520] DPL: waiting for inverter device information to be available
Fetch inverter: 112184709155
Request device info
PowerMeterClass: TotalPower: 27.49
[     30.533] DPL: waiting for inverter device information to be available
Success

Ich hatte nur das Putty-Fenster offen, glaube das loggt nicht automatisch mit? Der Ausschnitt sollte aber ja genügen. Vielleicht hilft das?

helgeerbe, 2023-07-26:
Vielen dank für das Log. Ohne ein Log, gibt es kaum eine Chance den Fehler zu finden.

Es sieht so aus, als ob die CPU von einer illigalen Adresse lesen will (EXCVADDR: 0x00000040). Entweder wurde ein Zeiger nicht richtig initialisiert, oder es gibt ein Speicher-Fehler wo ein gültiger Zeiger überschrieben wird (z. B. durch einen Stack Overflow).

Ich schaue mir die Stelle um den letzten Log-Eintrag noch mal an. Wäre gut zu wissen, ob der Fehler immer an der selben Stelle auftritt. (Würde für nicht richtig initialisieren Zeiger sprechen).

schlimmchen, 2023-07-26:
@helgeerbe Ich hab schon gesehen, dass du so einen Backtrace auseinandergepflückt hast (Adressen in Funktionamen aufgelöst). Ich glaube es war der, der mich zu meinem nullptr Zugriff geführt hat. Wie hast du den übersetzen lassen?

helgeerbe, 2023-07-26:
@schlimmchen hatte nichts besonderes gemacht. Der Backtrace kommt bei normalen Exceptions im USB log.   Also die ganz normalen Einstellungen aus platformio.ini. Leider kann man kein Debug Build bauen, da das Image zu groß wird.

schlimmchen, 2023-07-26:
Achso, du hängst mit dem `pio monitor` dran, ggf. über VSCode, und der übersetzt die automagisch... Und ein Debug-Build muss es auch sein?! Das ist schon echt ein Problem, dann kann man keinen Backtrace von einem User entziffern, selbst wenn der eine Release-Firmware benutzt. Ich hatte auch sowas gehofft: `<debug_util> --input <trace.log> --elf <firmware.bin>` => Ausgabe menschenlesbarer Backtrace.

helgeerbe, 2023-07-26:
Was man machen könnte:

Adding regular calls to [heap_caps_check_integrity_all()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html#_CPPv429heap_caps_check_integrity_allb) or [heap_caps_check_integrity_addr()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html#_CPPv430heap_caps_check_integrity_addr8intptr_tb) in your code helps you pin down the exact time that the corruption happened. You can move these checks around to “close in on” the section of code that corrupted the heap.

Nicht initialisierte Pointer finde ich jetzt nicht.

helgeerbe, 2023-07-29:
So, ich war weg und hatte jetzt den pio Monitor 3 Tage und 18 Stunden laufen. Kein Absturz, nichts! 

Das einzige wo es Probleme gab und der esp32 neugestartet ist, als die Fritzbox sich aufgehängt hat und das Mesh Wifi in einem merkwürdigen Zustand war. Ferner verträgt der esp32 anscheinend es nicht, wenn das Stromsignal nicht ganz sauber ist.

Manos1966, 2023-07-29:
> So, ich war weg und hatte jetzt den pio Monitor 3 Tage und 18 Stunden laufen. Kein Absturz, nichts!
>
> Das einzige wo es Probleme gab und der esp32 neugestartet ist, als die Fritzbox sich aufgehängt [...]

Ich bin leider nicht so erfolgreich #336 
Ich habe aber auch (noch) kein ELKO/ KEKO installiert.

> Ferner verträgt der esp32 anscheinend es nicht, wenn das Stromsignal nicht ganz sauber ist.

Was wurde dagegen besser helfen? ein ELKO, oder ein KEKO (und ich weiss, du wirst sagen lieber beides)

MalteSchm, 2023-07-29:
Sehr das ähnlich wie @helgeerbe 
Ich habe gerade 40h uptime. Sollte mehr sein, d.h. es gab wohl mal einen Neustart, aber ist jetzt auch kein großes Problem mehr

schlimmchen, 2023-07-29:
> Was wurde dagegen besser helfen? ein ELKO, oder ein KEKO (und ich weiss, du wirst sagen lieber beides)

@Manos1966 Am besten hilft ein Board mit gutem Design run um dem 5V->3V3 Spannungsregler. Aber das ist natürlich keine Option bei einem fertigen Teil. Dann solltest du folgendes sicherstellen: Verwende ein gutes 5V Netzteil. Gut heißt: Es gibt mind. 1.0A Strom aus und die Spannung geht (gemessen) eher Richtung 5.1V statt 5.0V oder gar 4.9V. Ferner gibt es leider sehr viele schlechte USB-Kabel. Versuche eins zu finden, mit dem die messbare Spannung des 5V Rails an deinem Board möglichst hoch ist. Und dann lötest du mal einen 100µF Elektrolytkondensator an das 5V Rail dazu. Das sollte dann allerdings reichen.

Kannst du ausschließen, dass du zu viel Strom auf dem 3V3 Rail verbrauchst? Hast du einen Level-Shifter zwischen deinem Board und dem VE.Direct Produzenten (deinem Victron MPPT)?

Manos1966, 2023-07-29:
@schlimmchen 
Netzteil ist sehr gut.
USB Kabel - guter Punkt, muss ich messen.
ELKO war zu erwarten :-) muss ich, ich habe bis jetzt gewartet weil ich einen CANbus fuer die Pylontech ins laufen haben will, und einen CANbus fuer den HUAWEI R4850G2. 

Ich habe ein USB Spannung-Messgeraet zwischen Netzteil und USB Kabel angeschlossen (ich werde Bilder demnaechst posten).
Mist, dass deckt den Fall aber nicht ab, falls das USB Kabel schlecht ist... :-( daran habe ich nicht gedacht.

**Ich habe keinen Level-Shifter zwischen dem Board und dem Victron MPPT** Hmmm...

Vielen Dank fuer die guten Tipps 👍

schlimmchen, 2023-07-29:
> Ich habe keinen Level-Shifter zwischen dem Board und dem Victron MPPT

Dann hat dein Board 5V auf einem 3V3 Pin gesehen, nämlich MPPT TX bzw. ESP32 RX. Nach allem, was ich im Kopf habe, ist das schlecht. Steht auch so in https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf Seite 47: "High-level input voltage" Maximum ist VDD+0.3V, also 3V6. Das musst du bitte korrigieren, bevor wir weiter nach Problemen suchen. Ehrlich gesagt solltest du mind. einen anderen RX Pin benutzen und den bisher gebrauchten dauerhaft stilllegen, oder noch besser den ganzen Chip bzw. das ganze Boards zur Seite legen.

helgeerbe, 2023-07-29:
Hallo @schlimmchen  ganz so ist es nicht: https://www.qworqs.com/2021/05/19/are-the-esp32-and-esp8266-5v-tolerant-yes-they-officially-are/

Habe es auch selbst ausprobiert und keine Probleme. Aber inzwischen habe ich trotzdem einen level shifter.

Manos1966, 2023-07-29:
@schlimmchen @helgeerbe 
Ich habe eine ganze Packung level shifters zu Hause, hasse ich aber das Loetten :-)

Und da helgeerbe damals gepostet hatte "mann koenne vorerst ohne Levelshifter auskommen", habe ich gewartet bis ich ein Board entgueltig zusammenstellen kann.
Und ich bin weiterhin unentschlossen, ob ich nicht den ADUM1201 LevelsShifter mit galvanische Trennung installieren sollte.

schlimmchen, 2023-07-29:
@helgeerbe Ich hab diesen Beitrag, den du verlinkt hast, auch gesehen. Das war als ich vorhin prüfen wollte, dass ich keinen Unfug schreibe. Da war ich verwundert und hab nochmal ins Datenblatt geschaut, wo konservative Angaben gemacht werden.

Ich wäre da sehr vorsichtig. Vielleicht gilt diese Aussage für neuere Chips nicht mehr? Es ist scheinbar kein Feature, die 5V Toleranz, sonst würde man mehr dazu lesen können?

Wenn das mein Board wäre und ich festgestellt hätte, dass ich versehentlich 5V Logik an einen 3V3 SoC angesteckt hätte, würde ich den SoC mit deutlicher Markierung zur Seite legen und ihn erst wieder anrühren, wenn meine Probleme mit einem anderen Board auch auftauchen und aus einem gut verstandenen Grund verschwinden.

Ich hab schon viel Zeit mit "wackeliger" Hardware verbracht und verschwendet. Daher tausche ich bei "merkwürdigen" Fehlern gerne großzügig aus.

Manos1966, 2023-07-30:
@helgeerbe @schlimmchen 
Eingentlich mache ich mein Leben unnötig kompliziert.
Ich habe ja Heleerbes Platine mit KEKO, ELKO, Spannungsregler und Levelshifter.
Ich werde sie anschliessen, dann werden wir sehen ob alle Neustarts beseitigt sind (oder nicht)...
[image]

Manos1966, 2023-07-30:
Nebenbei, alte Platine, 
seit gestern ca 13:00 keine Neustarts.
Das Wetter hier ist besch... der Victron produziert nichts, die Batterie ist leer.
[image]

peff74, 2023-07-31:
Moin,
interessant, bin damit also nicht ganz alleine.
Bin für kurzem wieder vom einem "normalen" ESP32 auf einen WT32-ETH01 umgezogen.
Natürlich schaut man dann etwas häufiger, wie es denn so läuft.
Was mir dabei auffiel, je häufiger man auf der Website des ESPs geht, je eher schmiert die Maschine ab.
Seitdem ich die Website auf allen Geräten geschlossen habe, ist Ruhe und das Ding läuft seit 24h durch.
Vorher hatte ich Neustarts quasi im Halbstundentakt.
Die Online Konsole scheint besonders dieses Neustarts zu provozieren, zumindest bei mir.

Vielleicht ne Anmerkung am Rande, AsyncTCP und der AsyncWebserver sind ja leider kein Garant für lange Uptimes.
In einem eigenen Projekt habe ich mich lange damit herumgeschlagen und bin dann wieder zurück zum "normalen" Webserver.
Seitdem habe ich Uptimes...^^
[image]

Feldsalat, 2023-07-31:
>  Seitdem ich die Website auf allen Geräten geschlossen habe, ist Ruhe und das Ding läuft seit 24h durch. 
> 

Bei mir auch so, läuft jetzt seit vier Tagen ohne Neustart. Habe nur gelegentlich Spannungswerte angepasst und sonst bewusst den Aufruf von Webseiten vermieden.

Manos1966, 2023-07-31:
> Seitdem ich die Website auf allen Geräten geschlossen habe, ist Ruhe und das Ding läuft seit 24h durch.

Oh, guter Tipp, muss ich probieren! ich habe einen Server da, wo die Webseite immer offen ist... :-o

Manos1966, 2023-07-31:
Just two restarts, I would call this very stable (considering that I have nothing on this board - no ELKO, KEKO, Levelshifter etc).
[image]

helgeerbe, 2023-08-01:
Ich würde das Ticket jetzt schliessen. Alle meine Restarts kann ich auf Power Probleme und Netzwerk Probleme zurückführen. Im Dauertest habe ich keine Auffälligkeiten wie Memoryleaks gesehen.

Für weitere  Stabilitätsthemen bitte ein neues Ticket aufmachen

closed 2023-08-01 by helgeerbe

Manos1966, 2023-08-05:
@helgeerbe 
Ich habe gestern meine Pylotech US5000 angeschlossen.
Es ist das erste Mal, dass ich ein SN65HVD230 CAN-Board angeschlossen habe.
Alles funktioniert perfekt, **aber** ich habe den Eindruck, dass ich wieder viele Neustarts habe (siehe Bild).
Firmwareversion / git Hash | 43e836a
Immer noch kein ELKO, KEKO, Level Shifter. Ich glaube ich werde mein Raspi wieder anschliessen um die Grunde der Neustarts aufzunehmen...
[image]

Manos1966, 2023-08-07:
Seit heute habe ich @helgeerbe PCB Platine installiert (ELKO, KEKO, Levelshifter).
Jetzt werde ich sehen ob Neustarts auch mit dieser Platine passieren.
<html>
<body>

Firmwareversion / git Hash | 43e836a
-- | --

</body>
</html>
Firmwareversion / git Hash	[43e836a](https://github.com/helgeerbe/OpenDTU-OnBattery/commits/43e836a)

schlimmchen, 2023-08-07:
Gibt's einen bestimmen Grund, warum du mit einer älteren Firmware als die zuletzt veröffentlichte unterwegs bist? Also ich jedenfalls werde mich nicht mit Neustarts von Firmware auseinandersetzen, die womöglich mit neuerer Firmware bereits ausgebügelt wären.

Veröffentlichst du von einer anderen Maschine MQTT Topics, die von der OpenDTU verarbeitet werden?

Manos1966, 2023-08-07:
> Gibt's einen bestimmen Grund, warum du mit einer älteren Firmware als die zuletzt veröffentlichte unterwegs bist? Also ich jedenfalls werde mich nicht mit Neustarts von Firmware auseinandersetzen, [...]

Weil ich nie mehrere Parameter gleichzeitig aendere, wenn es um sich um Fehlererkennung handelt.
Stand der Dinge:
Gleiche Firmware:
1. Ich hatte keine Neustarts mehr...bis ich den CANbus angeschlossen habe.
2. Mit CANbus und Pylontech, jede Menge Neustarts.
3. Wenn ich jetzt ELKO, KEKO, LevelShifter und Spannungsregler anschliesse, bei gleicher Firmware, und meine NEustarts weg sind, dann weiss ich dass es an meiner HW liegt.

Haette ich beides ausgetauscht, waere ich nicht mehr sicher....

Jetzt scheint Stabil zu sein, keine Neustarts, es ist aber noch zu frueh 
[image]

schlimmchen, 2023-08-07:
> Weil ich nie mehrere Parameter gleichzeitig aendere, wenn es um sich um Fehlererkennung handelt.

Sehr guter Punkt. Das ist natürlich die richtige Vorgehensweise.

Carry on :wink:

Manos1966, 2023-08-07:
:-D

Ups! und es ist gerade passiert!
Interessant... Ein Neustart ist passiert, als ich mich mit EDGE auf die OpenDTU eingeloggt habe um die Einstellungen abzulesen (ohne zu speichern).
[image]
Vielleicht mache ich zu viele Abfragen?
Ich habe zwei JSON Abfragen (Rot markiert) die staendig die OpenDTU abfreagen.
[image]

schlimmchen, 2023-08-07:
Was ist EDGE? Und was genau wird da gemacht ("JSON Abfragen")? Welche URL wird an der OpenDTU gepollt -- falls du die HTTP-Schnittstelle dafür nutzt.

Manos1966, 2023-08-07:
Mit EDGE meinte ich den Internet Explorer....

Folgende URLs werden gepollt: 
defmod OpenDTUmemory JsonMod http://192.168.178.65/api/system/status
defmod VoltageThreshold JsonMod http://192.168.178.65/api/powerlimiter/status
(jede Minute)

schlimmchen, 2023-08-07:
Hm. Hast du die Möglichkeit, die Ausgaben auf dem UART des ESP32 mitzuloggen (WebConsole reicht nicht).

Manos1966, 2023-08-07:
> Hm. Hast du die Möglichkeit, die Ausgaben auf dem UART des ESP32 mitzuloggen (WebConsole reicht nicht).

Na ja, mein Raspi ist weiterhin angeschlossen **ABER** ich benutze helgeerbes Platine. Sie hat den Strom Eingang ohne USB.
Ich weiss nicht, ob ich gleichzeitig eine Strom Zufuhr durch den USB Eingang des ESP32 haben darf.
Jetzt das ich das Bild sehe, ich stelle fest, ich habe den ELKO **falsch** gelötet! 🥇  
[image]

Ich habe weiterhin viele Neustarts....
[image]

Manos1966, 2023-08-08:
Also, ich habe ein neuen ELKO korrekt geloettet.

Zusaetzlich, habe ich mein Raspi an die ESP32 angeschlossen.

Leider mein PuTTY log zeigt chinesisch egal ob ich 9600, 57600, 115200 Baud auswaehle :-(

> =~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2023.08.08 12:31:24 =~=~=~=~=~=~=~=~=~=~=~=

h&$4FHP(sQ(3J)R
AD_T SF$@
H@! AN pKzW[RUUZwdH%FS&U,;PAOP}@ AA|ORP'V^ E`O(bP$(HãY)@dg &)VsITBKUF!JB
(dVT'QJ$BA. J#T!TTN
@+F)GWw*=sTPD'J$!G*FXl{(*j(HHVPJAXG*N@u({(}PS$JNPuN(DkRV!:p@!VpTFPVlFd^|@!j&$
DQ*&&*  A+=@U*D);BkP`VUw
Ak*W $&
W(R$BP(=SyNJPw(dN"@
8 c}G[uIISnp, !QPA*b] O
T uGS%ypCw&D]Wa9&PH7{N
PBVkc:XZg]tw&{0Ei[FRmH(1OT@T*pVu[mj%uRn*p3w&@DiWA(&( T
%lN!]!APU$W ueVJ+ZD5VhV
R!F)}S &$'NJV&*Y(;;J(umYj'MjB0hku0WS)[FT[s7
(5%*TM!HgB@TJ,TPW0J[:eGRZFj]DuEcKPGN$=;!U[Dl'M{TsiP)8JvQZN$.A.
P)(Jju(PD/ t_yFRjIudn@UX&Pp}X5J"j(cUSj  A@T=up[Zwqj5QOEEt=*DQ*A
85l@*dWDT&@R@APG4Mt4Q)@4dz<5V*@<-<AKJg]VS1)xr` y~UnF*O;ZV;&[_THk GZKZ[bcX@ @

schlimmchen, 2023-08-08:
Schließ mal die OpenDTU nur mit dem Mikro-USB-Kabel an, also ohne den 5V Klemmblock. Vielleicht wirds dann besser. Andernfalls, muss ich sagen, ist deinem Aufbau bzw. dem Boards mit dem ESP32 nicht mehr zu trauen, und dann lohnt es sich auch nicht, die Ursache für die Neustarts mühevoll zu suchen.

Wenn du aber einen Backtrace organisieren kannst, haben wir zumindest gute Chancen, einen Grund zu finden.

spcqike, 2023-08-08:
@schlimmchen ich weiß nicht woran es liegt, aber mit 81864b3 läuft auch mein Modul selten länger als 3h. ab und an läuft es zwar auch mal für 1,5 Tage, aber eher selten.

Vorher lief er länger, hat sich aber regelmäßig aufgehangen (aller paar Tage) oder der Live-View ging nicht mehr und er brauchte einen externen Neustart (Stromversorgung aus / an), jetzt startet er sich unregelmäßig selber neu.

Da ich keine weitere Regelung nutze (keine Batterie, aktuell kein DPL), und ihn nur zum Protokollieren der Wechselrichterdaten hernehme, stört mich das nicht akut.

In den kommenden Tagen werd ich mal wieder auf die neuste FW updaten, gucken was die sagt.

Hardwareseitig hat sich seither nichts geändert.

peff74, 2023-08-14:
aktuelle: [026bca9](https://github.com/helgeerbe/OpenDTU-OnBattery/commits/026bca9)
[image]

- MQTT
- MQTT PowerMeter
- Solar Passthrough
- Victron

Auf nun fest verlöteter Platine / Kondensator und Funkmodul ohne Verstärker

Manos1966, 2023-08-15:
Firmware 2023.08.10.post1

Seit dem Update funktioniert es ohne Probleme.
Ein Neustart vielleicht jede 1,5 Tage 👍 
Version ohne KEKO, ELKO, Levelshifter, Victron+Pylontech

[image]