hoylabs/OpenDTU-OnBattery #316: Huawei R4850G2 - Sporadische CAN Verbindungsaussetzer

Phantomias2006, 2023-07-16:
### What happened?

Nach längerer Laufzeit des ESP gibt es vermehrt Probleme beim auslesen der Daten über CAN.
Für die Analyse habe ich mir den „date_age“ Parameter in Grafana visualisiert und einen automatischen reboot des ESP eingerichtet wenn die Verbindung länger als 50Sekunden ausgefallen ist. Sobald der ESP neu startet funktioniert es wieder für ca. 1-2h ohne Probleme.
Diese Neustarts sind im Chart gut sichtbar.

[image]

### To Reproduce Bug

Aktiviere „AC Ladegerät“ im Webinterface und kontrolliere das Logfile nach 1-2h.
Alternativ kann auch der „date_age“ Parameter extern gespeichert und visualisiert werden.

### Expected Behavior

keine bzw. seltene Verbindungsaussetzer.

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

1a34398

### Relevant log/trace output

```Shell
[HUAWEI********************* Sending request
07:32:26.707 > PowerMeterClass: TotalPower: 28.00
07:32:30.804 > Received MQTT message on topic: powermeter/saldo
07:32:30.804 > PowerMeterClass: TotalPower: 31.00
07:32:31.111 > [HUAWEI********************* Sending request
07:32:36.127 > [HUAWEI********************* Sending request
07:32:36.743 > PowerMeterClass: TotalPower: 31.00
07:32:40.850 > Received MQTT message on topic: powermeter/saldo
07:32:40.850 > PowerMeterClass: TotalPower: 28.00
07:32:41.146 > [HUAWEI********************* Sending request
07:32:41.170 > [HuaweiCanClass::onReceive] In:  231.06V, 0.00A, 0.00W
07:32:41.170 > [HuaweiCanClass::onReceive] Out: 2.36V, 0.00A of 0.00A, 0.00W
07:32:41.170 > [HuaweiCanClass::onReceive] Eff: 0.0%, Temp in: 40.0C, Temp out: 37.0C07:32:46.164 > [HUAWEI********************* Sending request
07:32:46.186 > [HuaweiCanClass::onReceive] In:  231.00V, 0.00A, 0.00W
07:32:46.186 > [HuaweiCanClass::onReceive] Out: 2.36V, 0.00A of 0.00A, 0.00W
07:32:46.186 > [HuaweiCanClass::onReceive] Eff: 0.0%, Temp in: 40.0C, Temp out: 37.0C07:32:46.779 > PowerMeterClass: TotalPower: 28.00
07:32:50.776 > Received MQTT message on topic: powermeter/saldo
07:32:50.776 > PowerMeterClass: TotalPower: 25.00
07:32:51.284 > [HUAWEI********************* Sending request
07:32:56.181 > [HUAWEI********************* Sending request
07:32:56.736 > PowerMeterClass: TotalPower: 25.00
07:33:00.714 > Received MQTT message on topic: powermeter/saldo
07:33:00.714 > PowerMeterClass: TotalPower: 28.00
07:33:01.217 > [HUAWEI********************* Sending request
07:33:06.234 > [HUAWEI********************* Sending request
07:33:06.246 > [HuaweiCanClass::onReceive] In:  231.00V, 0.00A, 0.00W
07:33:06.246 > [HuaweiCanClass::onReceive] Out: 2.36V, 0.00A of 0.00A, 0.00W
07:33:06.246 > [HuaweiCanClass::onReceive] Eff: 0.0%, Temp in: 40.0C, Temp out: 37.0C07:33:06.850 > PowerMeterClass: TotalPower: 28.00
07:33:10.843 > Received MQTT message on topic: powermeter/saldo
07:33:10.843 > PowerMeterClass: TotalPower: 27.00
07:33:11.355 > [HUAWEI********************* Sending request
07:33:11.368 > [HuaweiCanClass::onReceive] In:  231.00V, 0.00A, 0.00W
07:33:11.368 > [HuaweiCanClass::onReceive] Out: 2.36V, 0.00A of 0.00A, 0.00W
07:33:11.368 > [HuaweiCanClass::onReceive] Eff: 0.0%, Temp in: 40.0C, Temp out: 37.0C07:33:16.372 > [HUAWEI********************* Sending request
07:33:16.394 > [HuaweiCanClass::onReceive] In:  231.00V, 0.00A, 0.00W
07:33:16.394 > [HuaweiCanClass::onReceive] Out: 2.36V, 0.00A of 0.00A, 0.00W
07:33:16.394 > [HuaweiCanClass::onReceive] Eff: 0.0%, Temp in: 40.0C, Temp out: 37.0C07:33:16.885 > PowerMeterClass: TotalPower: 27.00
07:33:20.878 > Received MQTT message on topic: powermeter/saldo
07:33:20.879 > PowerMeterClass: TotalPower: 25.00
07:33:21.390 > [HUAWEI********************* Sending request
07:33:26.340 > [HUAWEI********************* Sending request
07:33:26.356 > [HuaweiCanClass::onReceive] In:  231.00V, 0.00A, 0.00W
[... 75 lines cut ...]
07:35:11.906 > [HuaweiCanClass::onReceive] In:  231.19V, 0.00A, 0.00W
07:35:11.906 > [HuaweiCanClass::onReceive] Out: 2.36V, 0.00A of 0.00A, 0.00W
07:35:11.906 > [HuaweiCanClass::onReceive] Eff: 0.0%, Temp in: 40.0C, Temp out: 37.0C07:35:16.900 > [HUAWEI********************* Sending request
07:35:17.032 > PowerMeterClass: TotalPower: 27.00
07:35:20.792 > Received MQTT message on topic: powermeter/saldo
07:35:20.792 > PowerMeterClass: TotalPower: 25.00
07:35:21.918 > [HUAWEI********************* Sending request
07:35:21.940 > [HuaweiCanClass::onReceive] In:  231.19V, 0.00A, 0.00W
07:35:21.940 > [HuaweiCanClass::onReceive] Out: 2.36V, 0.00A of 0.00A, 0.00W
07:35:21.940 > [HuaweiCanClass::onReceive] Eff: 0.0%, Temp in: 40.0C, Temp out: 37.0C07:35:26.937 > [HUAWEI********************* Sending request
07:35:27.060 > PowerMeterClass: TotalPower: 25.00
07:35:30.830 > Received MQTT message on topic: powermeter/saldo
07:35:30.830 > PowerMeterClass: TotalPower: 22.00
07:35:31.959 > [HUAWEI********************* Sending request
07:35:36.970 > [HUAWEI********************* Sending request
07:35:37.081 > PowerMeterClass: TotalPower: 22.00
07:35:40.863 > Received MQTT message on topic: powermeter/saldo
07:35:41.066 > PowerMeterClass: TotalPower: 21.00
07:35:41.990 > [HUAWEI********************* Sending request
07:35:47.006 > [HUAWEI********************* Sending request
07:35:47.097 > PowerMeterClass: TotalPower: 21.00
07:35:50.795 > Received MQTT message on topic: powermeter/saldo
07:35:50.795 > PowerMeterClass: TotalPower: 21.00
07:35:51.992 > [HUAWEI********************* Sending request
```

MalteSchm, 2023-07-18:
Hey

Das Netzteil schickt alle Parameter einzeln. Dazu wird es durch den ESP alle 5s aufgefordert

Der Data Age Wert wird nur zurückgesetzt wird wenn ein Output Strom oder Output Power Wert empfangen wurde. Das schienen mir die wichtigsten Werte zu sein.

Normalerweise sollte also Data age maximal 5s sein. Ich sehe aber auch gelegentlich höhere Werte, jedoch nicht so extrem wie bei dir.

Meine Vermutung sind hier Paketverluste. Mit kommen dazu zwei Dinge in den Kopf:

Bei dem Netzteil müsste man mal messen ob der can Bus terminiert ist. Ich komme da leider nur schwer ran. Evtl geht das bei dir besser

Der CAN request wird glaube ich auch acknowledged. Das wird im Code gerade nicht geprüft. Evtl lässt sich das nutzen

Phantomias2006, 2023-07-18:
Hi @MalteSchm,
danke für das Feedback. Aktuell ist alles noch bei mir auf der Werkbank aufgebaut, also komme ich sehr gut zu den Komponenten.
Ich werde den Bus Widerstand messen und Bescheid geben. 

Ich hatte bis jetzt einen Memory Leak in Verdacht. Wenn es die Verbindungsaussetzer gibt, geht der Heap auf 49% und plötzlich geht er wieder auf 47% und die Daten kommen wieder regelmäßiger. Das ist aber nur mal ein Verdacht... Leider werden die Heap Werte nicht mittels MQTT übertragen. Ich habe jetzt die API angebunden und werde mir die Daten auch loggen um zu schauen wie sich der Heap zu meinem Problem verhält.

Gibt es erweiterte debug ausgaben die ich aktivieren kann? Im Logfile sehe ich ja nur das der Request gesendet wird aber keine Antwort kommt. Ob generell etwas vom Bus retour kommt sehe ich ja nicht. Zur Not muss ich mir das Oszi mit nachhause nehmen.

LG

Phantomias2006, 2023-07-18:
@MalteSchm so wie es aussieht ist da nichts Terminiert am Netzteil. Gesamtwiderstand ist 120Ohm. Sollte aber 60Ohm sein. Ich werd mal einen Widerstand suchen und anschließen. Eventuell löst es meine Probleme :)

Danke für den wink. Ich hab mich da zu sehr auf die Infos im Netz verlassen und da niemand auf der Seite des Netzteils einen Widerstand hatte wäre ich da sicher nicht so schnell darauf gekommen…

MalteSchm, 2023-07-18:
Berichte mal bitte ob das hilft. Ist vermutlich auch bei mir der Fall und funktioniert dann einfach zufällig so gut das es mir nicht aufgefallen ist.

MalteSchm, 2023-07-19:
Ich versuche am Wochenende mal da reinzuschauen

Phantomias2006, 2023-07-19:
Danke, ich hoffe du findest etwas.
Ich hab jetzt dataAge vom Pylontech und Huawei + heap in Grafana.
[image]

heap geht schon ziemlich hoch und auch wieder runter sobald es wieder aufhört. Auch am Pylontech sieht man zur selben Zeit Aussetzer.
Denkst du es gibt nen memory leak?

Phantomias2006, 2023-07-19:
[image]

Phantomias2006, 2023-07-20:
Ich habe jetzt das AC Netzteil im WebUI deaktiviert und den rest laufen lassen:
[image]
Hier gibt es in der letzten Stunde keine Auffälligkeiten. Weder beim Pylontech CAN noch beim Heap...

MalteSchm, 2023-07-23:
Schon komisch. Bei mir ist es gerade so das das alles extrem instabil ist auch wenn das Netzteil inaktiv ist.

Phantomias2006, 2023-07-23:
Hast du einen Verdacht was da schuld sein könnte?

MalteSchm, 2023-07-23:
@Phantomias2006 
Ich hab da heute vermutlich 6h drüber gesessen und ziemlich geflucht. Zur Instabilität die ich gesehen habe kann ich nichts sagen. 

Beim Netzteil hat sich folgendes Problem herauskristallisiert:  
- Der MCP2515 hat nur einen sehr kleinen Lesepuffer von zwei Nachrichten. Neue Nachrichten überschreiben dabei die alten Nachrichten
- Das Lesen des MCP2515 Puffers war bisher per polling gelöst. Das Problem dabei ist das die Routine die das tut nur recht selten aufgerufen wird. Der ESP hat halt noch genug anderes zu tun.
- -> Daher wird ein Großteil der Nachrichten überschrieben.

Ich habe dann lange versucht das Lesen der Nachrichten per Interrupt zu erledigen. Problem dabei ist:
- Die Interruptroutine muss scheinbar komplett im SRAM liegen. Das lässt sich nur mit Verrenkungen erreichen. Das Problem dabei war aber das, wenn der Interrupt aktiviert war, OTA Updates nicht mehr gingen. Wenn da jemand Tips hat -> gerne her damit. Ansonsten habe ich diesen Ansatz auf Eis gelegt.
- Mein Ansatz war es jetzt die loop() Routine zu splitten und deutlich häufiger aufzurufen. Außerdem werden die Parameter häufiger abgefragt.

Den Zusammenhang zwischen dem Heap und dem Data Age hab ich so noch nicht hergestellt. Ich bin nicht sicher was da Ursache und Wirkung ist.

Phantomias2006, 2023-07-23:
@MalteSchm, danke für deine Analyse und Zeit die du mit der Fehlersuche heute verbracht hast.

Nutzt das Projekt beide Cores? Ich habe mir das noch nicht angesehen. Eventuell hilft es wenn OTA im anderen Core ausgelagert wird?

Wenn du willst kannst du mir deinen SW stand zukommen lassen. Ich flashe ihn drauf und schaue wie es sich heute nacht verhält.

LG

spcqike, 2023-07-23:
@Phantomias2006 falls nur 1 Kern genutzt wird, wäre es nicht einfacher dann die ISR kurz zu halten (flag setzen) und die eigentliche Routine zum abholen in separaten Core laufen zu lassen? Und den Rest (OTA) so zu belassen wie es ist? 

Oder allgemein die ganzen Loops (power Meter / power limiter / ….) auf den anderen Kern auszulagern, damit der Main-Loop nicht 10-20 Sekunden braucht in worst case?

Phantomias2006, 2023-07-23:
Hab kurz mal nachgelesen,
man kann es komplett im anderen core laufen lassen inkl. Interrupt.
Wäre eventuell ein möglicher Ansatz was auf den anderen Core auszulagern…

MalteSchm, 2023-07-23:
@Phantomias2006 
Code liegt hier: https://github.com/MalteSchm/OpenDTU-OnBattery/tree/huawei_interval_fix

Phantomias2006, 2023-07-23:
Danke @MalteSchm 
Wenn ich mir jedoch die letzten paar Minuten ansehe dann sehe ich hier keine Verbesserung.
Bzw. 16Sekunden in den ersten paar Minuten nach einem neustart hatte ich vorher nicht…
Ich lasse es über Nacht laufen und dann sehen wir mal wie es morgen aussieht.

[image]

MalteSchm, 2023-07-23:
Ja ist so nicht unerwartet .
Deswegen schreib ich oben etwas von Zufall
Warum ist das Timing hier so wichtig?

Phantomias2006, 2023-07-23:
Was meinst du bezüglich „Timing“?

MalteSchm, 2023-07-24:
Das Data Age klein ist

Vielleicht zum Verständnis: die Werte werden in meinen branch jetzt alle 2.5s abgefragt (vorher alle 5s). Abfragen geht nur en Block, resultiert aber in mehreren Nachrichten
Data age wird nur zurückgesetzt wenn ein Output Strom oder ein Output Power Wert empfangen wurde

Phantomias2006, 2023-07-24:
Guten Morgen,

wie du in meinen Charts siehst, schaukelt sich das komplette System bei mir auf mit der Laufzeit. Da ich anfangs nur das Netzteil verbunden hatte (weder WR noch sonstiges) habe ich begonnen den Fehler im Huawei CAN Teil zu suchen. Auffällig war/ist das die CAN-Verbindung augenscheinlich Aussetzer hat. Das schaukelt sich soweit auf bis ich bei ~120Sekunden Verbindungslücke bin. Abhilfe schafft hier nur ein Restart des ESP. Das ist auch soweit eingerichtet aktuell bei mir. Sobald es Aussetzer für >20 Sekunden gibt wird der ESP jetzt neu gestartet. Mit diesem Workaround komme ich aktuell zurecht, schön finde ich es jedoch nicht. CAN ist ein sehr robustes Protokoll und ich bin/war es einfach nicht gewöhnt, wenn ich alle x Sekunden daten auslese das ich diese nicht auch im selben Intervall bekomme. 

Hast du den Code noch von deinem ersten Lösungsansatz? Der hat grundsätzlich funktioniert nur OTA ging nicht mehr, oder? Ich würde den gerne testen, ob es eine Veränderung bei mir im System bringt. Aktuell komme ich gut zum System und kann auch nicht mittels Kabel flashen.

Phantomias2006, 2023-07-24:
Hi,

ich habe einen Pylontech Speicher welcher ausschließlich über das Huawei Netzteil geladen wird. Die Steuerung selber mache ich mittels NodeRed und falls dataAge beim Speicher oder Netzteil zu groß wird muss ich zwangsweise den Ladevorgang abschalten.
Ob ich jetzt die Werte alle 5 Sekunden oder alle 20 Sekunden bekomme spielt hier keine große rolle. Ob es für die Batterie relevant ist wenn ich während dem Ladevorgang jede Stunde den Ladevorgang kurz unterbreche für einen neustart bzw. er unterbrochen wird weil ein grenzwert überschritten wird kann ich nicht beurteilen, falls möglich würde ich es jedoch vermeiden.

Grundsätzlich gebe ich dir recht, ob diese instabilität auf das Netzteil bzw. den Code des Netzteils zurückzuführen ist kann ich nicht beurteilen, da nichts anderes bei mir verwendet wurde außer MQTT und Netzteil und ich davon ausging das die User die ein Huawei Netzteil verwenden eher weniger sind und zum Zeitpunkt des Bugreports keine nenneswerten Stabilitäts Issues offen waren war dies naheliegend für mich.

Gestern am Abend habe ich wieder den aktuellen Code aufgespielt und den Restart deaktivert.
Seit 12:30Uhr hat der ESP aufgehört zu senden und ist auch über das Webinterface nicht mehr erreichbar. 
Ich war aber noch nicht zuhause um nachzusehen was passiert ist...

MalteSchm, 2023-07-24:
Ok danke
Bei mir läuft das seit gestern ohne Neustart durch. Uptime ist irgendwas größer 20h

Kannst ja mal schreiben ob du da was gefunden hast

Phantomias2006, 2023-07-26:
@MalteSchm der Totalausfall war unabhängig von der Software. Mein kleiner hat sich anscheinend ein Eis aus dem Keller geholt und meinte man solle Stromsparen und den rot leuchtenden Hauptschalter ausschalten :-D

indie89, 2023-08-08:
Moin zusammen, seit der Version vom 1.8.23 gibt es bei mir tlws sehr lange Verbindungsunterbrechungen zum Huawei. Am Anfang (die ersten paar Minuten) pollt er noch alle 5 Sek, dann wird der Zeitraum länger (15-60 sek) und irgendwann scheint er gar keine Verbindung mehr zu bekommen (dann gehts in die 6000 Sek+). Das ist bei der 1.8. und 6.8. Version genau gleich und jeweils nach downgrade auf die letzte Juli Version funktioniert es dann wieder. Hab aktuell keine Muße dem näher nachzugehen (da die Juli Version super funktioniert bei mir) wollte es nur hier kurz melden. Vllt geht es jemand von euch ähnlich?

indie89, 2023-08-18:
Das kann sein, allerdings möchte ich auch die Daten, die vom Huawei kommen, nutzen (und sei es nur für Anzeige/Statistik).
Daher hab ich ein neues, eigenes Issue dafür aufgemacht: https://github.com/helgeerbe/OpenDTU-OnBattery/issues/387

Phantomias2006, 2023-08-18:
> Dieses Verhalten habe ich bei mir auch beobachtet. Die Anzeige zum Huawei in der Live-Ansicht funktioniert kaum noch bzw. nur mit teilweise stundenlangen Aussetzern.
> 
> Allerdings sieht es bei mir so aus, dass primär das **Auslesen** des Ist-Zustands vom HUAWEI nicht bzw. nur verzögert funktioniert. Das **Schreiben** von Spannung/Strom-Limits funktioniert anscheinend ohne Verzögerung.
> 
> Das sehe ich anhand meines in Node-Red implementierten Lade-Algorithmus. Dort werden die Änderungen recht schnell umgesetzt, sodass die Ladeleistung sich schnell an veränderte Werte von Bezug/Einspeisung anpasst.
> 
> Der Flow nutzt die folgenden MQTT-Topics zur Steuerung des HUAWEI:
> 
>     * /huawei/cmd/power
> 
>     * /huawei/cmd/limit_online_current
> 
>     * /huawei/cmd/limit_online_voltage
> 
> 
> Den aktuellen Bezug / Einspeisung bekommt der Flow über einen "Volkszähler"-Fühler am Stromzähler.

Hi,

kannst du mir deinen Flow für das Laden zeigen? 
LG

CaCu15, 2023-08-18:
> Hi,
>
> kannst du mir deinen Flow für das Laden zeigen? LG

Kein Problem: Ist aber etwas komplexer, da:
- Ich eine State-Machine für verschiedene Zustände (CHARGE, ABSORB, DISCHARGE, etc) gebaut habe, weil ich verhindern will, dass all zu schnell hin- und hergeschaltet wird. Außerdem will ich auch analog Victron eine ABSORB Phase nach dem eigentlichen Laden haben. 
- Der Flow wird über eine ganze Reihe von Parametern gesteuert, die als große JSON Struktur in dem CONFIG-Node enthalten sind. 
- Das Auslesen des Volkszaehlers & Schreiben in eine InfluxDB mache ich über einen separaten Flow, der dann eine Schnittstelle zur eigentlichen Ladesteuerung hat. Der Volkszahler läuft bei mir in ESPHome und ich nutze die EventSourceAPI zum Auslesen. 

Hier sind die Exports beider Flows. 
[Flows.zip](https://github.com/helgeerbe/OpenDTU-OnBattery/files/12378527/Flows.zip)

Bin gerade dabei, den ganzen Flow direkt in ESPHome und OHNE MQTT zu implementieren, weil ich dann das ganze System robuster bekomme. Dann läuft alles auf 2 ESP32 (1 mit OpenDTU, 1 mit ESPHome für Volkszähler und Pylontech-Konsole) mit statischen IPs. Dann keine Abhängigkeit mehr von anderen Server-Diensten (DNS, MQTT, Node-Red, ...).

MalteSchm, 2023-10-04:
Fixed in https://github.com/helgeerbe/OpenDTU-OnBattery/pull/454
I think we can close this issue

closed 2023-10-09 by helgeerbe