hoylabs/OpenDTU-OnBattery #332: PowerLimiter disable via MQTT is not working anymore.

Napizaki, 2023-07-21:
### What happened?

when I send the MQTT topic "powerlimiter/cmd/disable = 1" via MQTT, unfortunately nothing happens in the current version ("e06740f"). I have tested: the last working release was "52af52e".

Additional question: what is the topic "PV/powerlimiter/status/mode"?

[image]

### To Reproduce Bug

Send the MQTT topic  "powerlimiter/cmd/disable = 1" and watch "powerlimiter/status/disabled"

### Expected Behavior

PowerLimiter to become disabled.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

e06740f

sylvio69, 2023-07-22:
Not sure but i think this is powerlimiter/cmd/**mode** since some versions?

Napizaki, 2023-07-22:
Hello sylvio69,
that's exactly what it was !!! Thanks for the quick feedback. I had not noticed that this has changed.
Unfortunately, the docu is not yet adapted.

Phantomias2006, 2023-07-22:
@Napizaki 
https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/MQTT
The documentation here looks correct.

Napizaki, 2023-07-22:
Sorry, I had looked in the [Readme.md](https://github.com/helgeerbe/OpenDTU-OnBattery#readme) - here is still the old state in it! Changed it to "Mode" and everything works as it should.

Thanks for the great software and the quick help!!!

helgeerbe, 2023-07-22:
With the next release, I will replace the Readme with the wiki home page.

schlimmchen, 2023-07-28:
@helgeerbe I think this can be closed. https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/MQTT is up-to-date and the main README was reset to point to the wiki. However, `docs/MQTT_Topics.md` should be reset to the upstream version, so there is only one listing of the (-onbattery-specific) MQTT topics. At least that's my opinion. PR follows.

schlimmchen, 2023-07-28:
So the wiki is public-writable, okay... I just edited the MQTT wiki page and threw out all documentation which is already described in `MQTT_Topics.md` and instead explained, that topics which are provided through the upstream OpenDTU part of OpenDTU-OnBattery are described there. Do you @helgeerbe agree that this is a good way to go? My main reason is that copies of any documentation will inevitably drift apart. If OpenDTU changes topics, we will probably merge `MQTT_Topics.md` (if it is updated upstream), but we will most certainly not update the wiki page in a timely manner.

closed 2023-07-30 by Napizaki