hoylabs/OpenDTU-OnBattery #340: Systemcrash bei fehlerhafter Power-Meter MQTT Topic

peff74, 2023-07-27:
### What happened?

Der ESP32 rebootet in Endlosschleife wenn die MQTT Topic des Power-Meters falsch ist 

### To Reproduce Bug

Nicht existierende MQTT Topic im Power meter eingeben 

### Expected Behavior

ESP Crash

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

  43e836a

### Relevant log/trace output

```Shell
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (700) esp_core_dump_flash: No core duæž partition found!
E (700) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... done
Reading PinMapping... found valid mapping done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Check for default DTU serial... done
done
Initialize Hoymiles interface... NRF: Connection successful
  Setting radio PA level... 
  Setting DTU serial... 
  Setting poll interval... 
  Setting verbosity... 
  Adding inverter: 114181870469 - HM700 done
done
Initialize ve.direct interface... 
Invalid pin config
Initialize Pylontech battery interface... 
Invalid pin config
Initialize Huawei AC charger interface... 
Invalid pin config
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using new credentials... done
Configuring WiFi STA DHCP IP... done
[      0.501] DPL: waiting for valid date and time to be available
Fetch inverter: 114181870469
Request SystemConfigPara
WiFi connected
WiFi got ip: 192.168.0.225
Network connected
Connecting to MQTT...
Connected to MQTT.
Received MQTT message on topic: solar/huawei/cmd/power
Inverter not found
Received MQTT message on topic: solar/huawei/cmd/power
Inverter not found
Received MQTT message on topic: solar/huawei/cmd/power
Inverter not found
Received MQTT message on topic: solar/huawei/cmd/power
Inverter not found
Received MQTT message on topic: solar/huawei/cmd/power
Inverter not found
Received MQTT message on topic: solar/huawei/cmd/power
Inverter not found
Received MQTT message on topic: solar/huawei/cmd/limit_online_voltage
Received MQTT message on topic: solar/huawei/cmd/limit_online_current
Received MQTT message on topic: solar/huawei/cmd/limit_offline_voltage
Received MQTT message on topic: solar/huawei/cmd/limit_offline_current
Received MQTT message on topic: solar/huawei/cmd/mode
Received MQTT message on topic: solar/powerlimiter/cmd/mode
Power limiter enabled
Received MQTT message on topic: mqtt/0/zaehler/strom/energy_consumptio

abort() was called at PC 0x401b1d17 on core 1

Backtrace: 0x40083c8d:0x3ffd8490 0x4008f1dd:0x3ffd84b0 0x400956a9:0x3ffd84d0 0x401b1d17:0x3ffd8550 0x401b1d5e:0x3ffd8570 0x401b20a3:0x3ffd8590 0x401b2387:0x3ffd85b0 0x400e54ea:0x3ffd85d0 0x400e58cb:0x3ffd8610 0x401d2185:0x3ffd8650 0x401247df:0x3ffd8680 0x400e14b3:0x3ffd86e0 0x401d1f99:0x3ffd8720 0x4010fec9:0x3ffd8750 0x40110717:0x3ffd8840 0x40110893:0x3ffd8880 0x40110931:0x3ffd88b0

ELF file SHA256: f28c897288706a53

E (4310) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

schlimmchen, 2023-07-28:
> Nicht existierende MQTT Topic im Power meter eingeben

Reicht bei mir nicht, um das Problem zu reproduzieren. Was genau muss ich tun?

Ich kann mehrere Probleme in der Implementierung erkennen, aber ein Subscribe auf ein nicht-existierendes Topic ist keins davon, da passiert nichts (Schlimmes).

Was wohl schief geht ist, wenn man ein Topic ließt, in dem etwas drinsteckt, das sich nicht zu einen float konvertieren lässt. Kann das dein Problem erklären?

peff74, 2023-07-29:
Moin,

Vielleicht muss ich erwähnen, dass ich den IOBroker als MQTT Broker nutze, da gab es schon einmal merkwürdige Phänomene.

Also folgendes ist bei mir passiert.
Ich habe "mqtt/0/zaehler/strom/energy_consumptio" beim PowerMeter eingegeben anstatt "mqtt/0/zaehler/strom/energy_consumption".
Die Folge war das oben dargestellte.

Merkwürdigerweise habe ich nun einen Topic der so benannt ist.

[image]

schlimmchen, 2023-07-29:
Und dieses falsche Topic kann nirgendwo anders mal publiziert worden sein? Das würde ja heißen, dass ein Subscribe in deinem Broker ein Topic "angelegt" hat und dann ggf. `(null)` publiziert?

Begreife ich nicht. Ist aber auch egal, #342 würde dafür sorgen, dass deine OpenDTU dann nicht mehr wegen einer Exception neustartet. Da würde ja dann ein leerer String (oder von mir aus "(null)") in `std::stof` gefüttert werden, und dessen Exception wird in Zukunft abgefangen. Wenn du sicher bist, wie du das Problem reproduzieren kannst, merk dir das doch bitte eine kurze Weile und versuche es nochmal zu provozieren, nachdem #342 auf deiner OpenDTU enthalten ist.

peff74, 2023-07-29:
nein das verrückte ist ja der falsche Topic ist nie publiziert worden, war vorher nicht vorhanden.
Ist schon schräg, zumal ich Ihn jetzt noch nicht mal so einfach wieder aus dem IOBroker rausbekomme.
Ich sollte auf Moskito umsteigen.

Ich kann gerne noch mal nachdem die #342 im Release drin ist.

helgeerbe, 2023-07-31:
> Und dieses falsche Topic kann nirgendwo anders mal publiziert worden sein? Das würde ja heißen, dass ein Subscribe in deinem Broker ein Topic "angelegt" hat und dann ggf. `(null)` publiziert?

ioBroker macht glaube ich so einen Mist. Das kommt mir bekannt vor.

closed 2023-08-01 by helgeerbe

peff74, 2023-08-01:
> > Ich kann gerne noch mal testen nachdem die #342 im Release drin ist.
> 
> @peff74 Prüfst du es bei nächster Gelegenheit mit der 2023.08.01 und gibst Rückmeldung?

mache ich, sollte bis Ende der Woche möglich sein.

peff74, 2023-08-04:
Fehler ist nachweislich mit der neuen Version behoben.

Und ich kann bestätigen, dass:
  ein Subscribe im IOBroker ein Topic "angelegt" und dann null publiziert!

Ob das so richtig/gewollt ist... Der MQTT Broker vom IOBroker ist schon ein wenig merkwürden^^

helgeerbe, 2023-08-04:
@peff74 Danke für die Rückmeldung