hoylabs/OpenDTU-OnBattery #355: Power meter value not displayed

alexz707, 2023-08-04:
### What happened?

The third phase json path is not prefilled with the value which was set before.

If someone wants e.g. to switch off the logging and presses save it won't work because phase 3 json path must not be empty.
[image]

### To Reproduce Bug

Set up https+ json with one request for all phases.
Add for each phase a different json path.
Save it, reload the page

json path of the third phase is empty the other two are filled out

### Expected Behavior

all 3 json paths are filled out

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

81864b3

helgeerbe, 2023-08-07:
Thanks for reporting. I can confirm the error.

helgeerbe, 2023-08-08:
Should be fixed with release 2023.08.08. Can you test it and close the issue, if you can confirm the fix.

alexz707, 2023-08-09:
Thanks for the fast response!  i can confirm the fix :)

closed 2023-08-09 by alexz707