hoylabs/OpenDTU-OnBattery #356: Der Wert: "Inverter Gesamtertrag Heute" wird mehrmals am Tag auf 0 zurückgesetzt!

the-lonely-one, 2023-08-05:
### What happened?

Der Wert: "Inverter Gesamtertrag Heute" wird mehrmals am Tag auf 0 zurückgesetzt!

Siehe Chart:
(Y-Achse sind "Wh")
[image]

Laut dem Webinterface von meiner OpenDTU-OnBattery gibt es keinen Hinweis darauf, das meine DTU selbst noch mein Hoymiles-Inverter ständig einen Neustart gemacht haben!

Ich vermute mal, das mein Problem etwas mit der Web-API bzw. mit der Webserver-Implementierung zu tun, das aktuell bei OpenDTU als Pull Request https://github.com/tbnobody/OpenDTU/pull/1201 diskutiert wird!

Meine Daten:

- Firmwareversion: 81864b3
- Chip-Modell: ESP32 WROOM 32U
- Display: SSD1306
- Inverter: Hoymiles HM-800
- MQTT: aktiviert
- DTU-Abfrageintervall: 5 Sekunden
- VE.Direct: aktiviert
- Power Meter: aktiviert HTTP(S) + JSON
- Dynamic Power Limiter: deaktiviert
- Batterie CAN Bus Schnittstelle: deaktiviert
- AC Ladegerät: deaktiviert

Lasse über MQTT alle 20 Sekunden ein Limit für den Inverter mit dem Topic "limit_nonpersistent_absolute" setzen.

### To Reproduce Bug

Solange die OpenDTU-OnBattery unter den genannten Bedingungen läuft, ist das Problem vorhanden!

### Expected Behavior

Der Wert: "Inverter Gesamtertrag Heute" sollte nur dann auf 0 zurückgesetzt werden, wenn der Inverter über die DC-Eingänge keine Energie mehr bekommt (zum Beispiel Nachts ohne Batterieversorgung) oder bewusst ein Neustart vom Inverter angestoßen wird! Zum Beispiel über das Webinterface oder über MQTT.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

81864b3

schlimmchen, 2023-08-05:
Wir dein Invert nur von genau einer OpenDTU/Ahoy abgefragt? Wie kommen die Daten vom Inverter in diesen Graph, den du geteilt hast?

Schau mal auf den 02.08., da siehst du, dass die Energiemenge stetig steigen "möchte":

[image]

Du hast eine Batterie, korrekt?

the-lonely-one, 2023-08-05:
> Wir dein Invert nur von genau einer OpenDTU/Ahoy abgefragt?

Ja, mein Inverter wird nur von einer DTU (OpenDTU OnBattery) abgefragt!

> Wie kommen die Daten vom Inverter in diesen Graph, den du geteilt hast?

Die Daten für den Graphen werden von der OpenDTU OnBattery per MQTT mit einem Veröffentlichungsintervall von 5 Sekunden an meinen MQTT-Broker gesendet.
Der "MQTT-Broker" ist ein sogenannter "Adapter" unter ioBroker.
Die Daten selber werden mit dem "History"-Adapter unter ioBroker aufgezeichnet und ich kann diese Daten als Graphen mit dem "ECharts"-Adapter darstellen!
Von diesem stammt auch der beigefügte Graph! 

Ich denke nicht, das es ein Problem mit der Daten-Aufzeichnung per MQTT gibt, da auch in der Live-Ansicht von der OpenDTU OnBattery der Wert: "Inverter Gesamtertrag Heute"  mehrmals am Tag auf 0 zurückgesetzt wird bzw. deckt sich das auch mit der bisherigen Aufzeichnung!

Mir ist aber heute etwas anderes aufgefallen:

Habe heute morgen ab ca.9:45Uhr das Setzen vom Inverter-Limit im 20 Sekunden-Takt per MQTT komplett gestoppt und das Limit einmal fest auf 10 Watt gesetzt!

Ab diesem Zeitpunkt wurde der Wert: "Inverter Gesamtertrag Heute" nicht mehr auf 0 zurückgesetzt!

Hier der zugehörige Graph:

[image]

Um ca. 20:15Uhr habe ich dann das Setzen vom Limit im 20 Sekunden-Takt per MQTT (alle 20 Sekunden) wieder aktiviert.

So wie es aussieht kehrt damit das Problem wieder zurück!

> Schau mal auf den 02.08., da siehst du, dass die Energiemenge stetig steigen "möchte":

Ich bin mir nicht mehr zu 100% sicher, aber ich glaube das ich zu dem Zeitpunkt, den du grün im Graphen markiert hast, an meinem Node in "Node-RED" herumgebastelt habe, der das Inverter-Limit alle 20 Sekunden per MQTT an die OpenDTU-OnBattery weitergibt! 

> Du hast eine Batterie, korrekt?

Ja, richtig erkannt. Ich habe eine Batterie!

helgeerbe, 2023-08-07:
Also "Inverter Total Yield Day" wird dann zurückgesetzt, wenn der Inverter einen Restart Befehl bekommt, oder er an der DC Seite stromlos (z. B. in der Nacht) gemacht wird. Der Wert kommt vom Inverter und kann nicht verändert werden. Wenn die OpenDTU neu startet, wird der Wert nicht geändert.

Hast du denn etwas im Event-Log im Live View stehen?

the-lonely-one, 2023-08-07:
> Also "Inverter Total Yield Day" wird dann zurückgesetzt, wenn der Inverter einen Restart Befehl bekommt, oder er an der DC Seite stromlos (z. B. in der Nacht) gemacht wird. Der Wert kommt vom Inverter und kann nicht verändert werden. Wenn die OpenDTU neu startet, wird der Wert nicht geändert.

Danke für die Info!
Wusste nicht genau, ob der Wert in der OpenDTU-OnBattery berechnet wird oder direkt aus dem Inverter kommt!
Hat mir aber bei der Suche der Ursache geholfen!

> Hast du denn etwas im Event-Log im Live View stehen?

Laut der Ereignisanzeige in der Live-Ansicht gab es bisher keinen direkten Hinweis darauf, das der Inverter einen Neustart gemacht hat, wenn der Wert: "Inverter Gesamtertrag Heute" auf 0 gesetzt wurde.
Außer der Inverter hat über die DC-Eingänge keine Energie mehr bekommt (zum Beispiel Nachts ohne Batterieversorgung) oder es wurde bewusst ein Neustart vom Inverter angestoßen wird,
dann konnte man einen zugehörigen Eintrag in der Ereignisanzeige finden -> "Inverter start"

Ich denke aber, das ich die Ursache nun gefunden habe!

Vorab folgende Info`s:
Ich benutze Node-RED, um den Inverter über die OpenDTU-OnBattery per MQTT zu steuern und hatte mal zum Testen einen Flow erstellt mit dem ich per MQTT einen Neustart vom Inverter auslösen konnte!

Hierfür habe ich das Topic:

[serial]/cmd/restart

benutzt und einen Neustart vom Inverter mit dem Wert "1" ausgelöst!
Hat soweit funktioniert!

Zusätzlich habe ich auch das Topic:

[serial]/cmd/power

getestet, mit dem ich den Inverter per MQTT aus- oder einschalten konnte! 
Hat soweit auch funktioniert!

Zuletzt hatte ich hier den Wert "1" gesetzt! 

Und hier fing meiner Meinung nach auch die Ursache für das Problem an!

In meinem MQTT-Broker waren diese Werte immer noch gesetzt!

Das heißt:

[serial]/cmd/restart = 1
[serial]/cmd/power   = 1

Normalerweise sollte dies aus meiner Sicht keine Probleme verursachen,
da die Werte eigentlich nur dann an die OpenDTU-OnBattery per MQTT gesendet werden,
wenn die Werte erneut gesetzt oder geändert werden! -> Dachte ich!

Nun habe ich mir aber heute den Protokollauszug von meinem MQTT-Broker angesehen und festgestellt,
das es hier öfters Verbindungsabbrüche zwischen meinem MQTT-Broker und der OpenDTU-OnBattery gab!

Siehe hier - Protokollauszug Meldungen MQTT-Broker:

```
2023-08-07 01:07:21.094 - info: mqtt.0 (123456) Client [OpenDTU-123456] connection closed: Error: read ECONNRESET
2023-08-07 01:07:23.120 - info: mqtt.0 (123456) Client [OpenDTU-123456] connected with secret 1691363243113_626
2023-08-07 01:07:23.210 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/limit_persistent_relative" with regex /solar\.[^.]*\.cmd\.limit_persistent_relative/
2023-08-07 01:07:23.210 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/limit_persistent_relative" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.limit_persistent_relative/
2023-08-07 01:07:23.212 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/limit_persistent_absolute" with regex /solar\.[^.]*\.cmd\.limit_persistent_absolute/
2023-08-07 01:07:23.212 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/limit_persistent_absolute" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.limit_persistent_absolute/
2023-08-07 01:07:23.215 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/limit_nonpersistent_relative" with regex /solar\.[^.]*\.cmd\.limit_nonpersistent_relative/
2023-08-07 01:07:23.215 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/limit_nonpersistent_relative" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.limit_nonpersistent_relative/
2023-08-07 01:07:23.219 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/limit_nonpersistent_absolute" with regex /solar\.[^.]*\.cmd\.limit_nonpersistent_absolute/
2023-08-07 01:07:23.219 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/limit_nonpersistent_absolute" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.limit_nonpersistent_absolute/
2023-08-07 01:07:23.223 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/power" with regex /solar\.[^.]*\.cmd\.power/
2023-08-07 01:07:23.223 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/power" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.power/
2023-08-07 01:07:23.227 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/restart" with regex /solar\.[^.]*\.cmd\.restart/
2023-08-07 01:07:23.227 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "solar/+/cmd/restart" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.restart/
2023-08-07 01:07:23.229 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "mqtt.0.solar.huawei.cmd.limit_online_voltage"
2023-08-07 01:07:23.232 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "mqtt.0.solar.huawei.cmd.limit_online_current"
2023-08-07 01:07:23.236 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "mqtt.0.solar.huawei.cmd.limit_offline_voltage"
2023-08-07 01:07:23.240 - info: mqtt.0 (123456) Client [OpenDTU-123456] subscribes on "mqtt.0.solar.huawei.cmd.limit_offline_current"
[... 253 lines cut ...]

 

Habe zwischenzeitlich das Topic:

[serial]/cmd/restart

auf den Wert "0" gesetzt und seitdem kein Problem mehr das der Wert: "Inverter Gesamtertrag Heute" 0 Wh zurückgesetzt wird, wenn es einen MQTT-Verbindungsabbruch gibt!

Werde zukünftig, wenn ich das Topic:

[... cut ...]
Warum es diese MQTT-Verbindungsabbrüche nur dann gibt, wenn ich per MQTT alle 20 Sekunden ein Limit für den Inverter mit dem Topic "limit_nonpersistent_absolute" setzen lasse, kann ich aktuell nicht nachvollziehen!

Für mich ist aber das Problem erstmal soweit gehoben und die OpenDTU-OnBattery
macht genau das, was ich will! 

Danke nochmal für eure Unterstützung und tut mir leid, wenn ich eure Zeit mit meinem Problem geraubt habe!

Ist aus meiner Sicht kein Bug in der OpenDTU-OnBattery, sondern der Bug bin ich! :-(

OpenDTU-OnBattery find ich gut, macht weiter so!!!

schlimmchen, 2023-08-08:
Du kannst womöglich das "Retain-Flag" weglassen wenn du das Topic aus Node-Red veröffentlichst. Dann wird der Broker den Wert des Topics nur einmal an MQTT Clients zustellen und dann vergessen. So sollte es jedenfalls ohne "Retain" sein.

Schön, dass du das Problem gelöst hast, dann kannst du dieses Issue schließen.

closed 2023-08-08 by the-lonely-one