hoylabs/OpenDTU-OnBattery #359: [Request] Support for Volkszähler/vzlogger raw/timestamp MQTT data as power meter input (JSON parsing?)

TimeTravel-0, 2023-08-06:
### Is your feature request related to a problem? Please describe.

First of all, thanks to all contributors for this great piece of software!

I would like to use Volkszähler/vzlogger MQTT raw output as power meter input for OpenDTU-OnBattery.  The data format is incompatible (contains some json data).

This data format can be configured with (excerpt from vzlogger.conf):
```
  "mqtt": {
    "enabled": true,
    "host": "10.10.9.249",
    "port": 1883,
    "id": "vzlogger",
    "cafile": "",
    "capath": "",
    "certfile": "",
    "keyfile": "",
    "keypass": "",
    "keepalive": 30,
    "topic": "vzlogger/data",
    "id": "",
    "user": "",
    "pass": "",
    "retain": true,
    "rawAndAgg": true,
    "qos": 0,
    "timestamp": true
  },
```

and results in the following MQTT data:
`{ "timestamp": 1691336711731, "value": 84.800000000000011 }`

This data is reported every time the "smart" meter reports data via its optical interface (once per second). I've chosen this config to have accurate data available via MQTT  but only log data on much higher interval (once every n minutes) to the Volkszähler Database.

### Describe the solution you'd like

In PowerMeter.cpp / onMqttMessage in case parsing the whole string as float fails, add some code that can convert data in the format of `{ "timestamp": 1691336711731, "value": 84.800000000000011 }` to  `84.800000000000011`, e.g. split the string at the last ":" and parse the string from there as float. 

In case the data cannot be parsed as float to begin with, its unlikely that bad data will contain a ":" and a string that can be parsed as float afterwards, so unexpected side effects are unlikely, but this is kind of hidden unexpected functionality. Worst-case some power meter returns "error code: 32" in this field and it gets parsed as 32 Watts.

A more-universal and better solution would be to use some kind of json parser like its done in the Android App MQTTdash, but thats kind of overkill just for this use-case. As the pin config is provided as json there must be a runtime json parser present in the code, so maybe its just a matter of adding the possibility to specify a json path to extract the value from?

### Describe alternatives you've considered

I just wrote a small python script that runs as daemon on the Raspberry Pi that runs Volkszähler anyway (endless loop subscribed to the topic, does the string operation and publishes to another topic), but this is kind of ugly glue logic and adds another item in the data transport that adds delay and potential issues. Direct support for vzlogger output in OpenDTU-Onbattery would be way nicer.

spcqike, 2023-08-07:
if its of any help: 
you could try to use HTTP(S)+JSON parsing.

http://<ip or domain>/middleware/data/<uuid>.json

gives a json like

`{"version":"0.3","data":{"tuples":[[1691308865300,255.7,1]],"uuid":"xxxxx","from":1691308805492,"to":1691308865300,"min":[1691308865300,255.6999969],"max":[1691308865300,255.6999969],"average":255.7,"consumption":4.248,"rows":2}}`

JSON Path would look like
`data/tuples/[0]/[1]`

[image]

spacyt, 2023-11-20:
Hello i had some problem to get my vzlogger output to openDTU_on_Battery.
i am trying to use http+json. My vzlogger's http output looks like that.

{ "version": "0.8.2", "generator": "vzlogger", "data": [ { "uuid": "xxxxxxxxxxxxxxx", "last": 1700420816881, "interval": 0, "protocol": "d0", "tuples": [ [ 1700420816881, 22786.386299999998 ] ] } ] }

When i try to give this to my address "data/tuples/[0]/[1]" the output is empty.
I have try to find the solution with some JSON Formater online, but my output i always empty.

Hope someone can open my eyes.

Greeting
spacy

spacyt, 2023-11-21:
Thank you for yout answer.
This was what i try at the begininng but the result was.

When i open that link in webbrowser.
http://192.168.1.5:8084/0383faf0-bc61-11e5-a165-3d8426a81ff2/data/[0]/tuples/[0]/[1]

The Result was: 
{ "version": "0.8.2", "generator": "vzlogger", "data": [ ] }

Thx for help
spacy

spcqike, 2023-11-21:
i'm not sure if you can open the json as URL.

as shown here https://github.com/helgeerbe/OpenDTU-OnBattery/issues/359#issuecomment-1667391372 you have an URL like ```http://ip/middleware/data/UUID.json``` and you have a JSON path which should be ```data/[0]/tuples/[0]/[1]``` in your case.

spacyt, 2023-11-26:
Hi and Thank you for your help.
The answer was DNS. My OpenDtu was not working with my dns name of my vzlogger. 
Haven't  seen this, my mistake.
Everything else was ok and now it is working.

schlimmchen, 2024-07-01:
Please have a look at #1077. The requested feature (allow parsing of MQTT payload as JSON) is now implemented and will be part of the next release.

closed 2024-07-10 by schlimmchen