hoylabs/OpenDTU-OnBattery #361: Schalter "Ausführliche Protokollierung" bei "Dynamic Power limiter Settings" ohne Funktion --> immer ON

peff74, 2023-08-07:
### What happened?

Schalter "Ausführliche Protokollierung" bei "Dynamic Power limiter Settings" ohne Funktion --> immer ON

### To Reproduce Bug

schiebe den Switch auf Off, speichere, danach steht er wieder auf ON

### Expected Behavior

Nach dem umschalten auf OFF und speichern, steht der Schalter weiterhin auf ON und im Log werden die ausführlichen Infos zum Dynamic Power limiter ausgeben,

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

81864b3

schlimmchen, 2023-08-07:
Oh man, du hast recht, wie peinlich... Ich schau es mir sofort an.

helgeerbe, 2023-08-07:
Ich mach gleich mal eine Release 2023.08.07

closed 2023-08-07 by helgeerbe