hoylabs/OpenDTU-OnBattery #362: Firmware-Aktualisierung zeigt >Neue Version verfügbar! Zeige Änderungen!< obwohl keine neue verfügbar ist.

peff74, 2023-08-07:
### What happened?

Firmware-Aktualisierung zeigt >Neue Version verfügbar! Zeige Änderungen!< obwohl keine neue verfügbar ist.

### To Reproduce Bug

have a look into "System Informationen"

### Expected Behavior

Although the FW is up-to-date, it indicates that there is a newer one.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

81864b3

schlimmchen, 2023-08-07:
Naja, also https://github.com/helgeerbe/OpenDTU-OnBattery/compare/a6b849b...HEAD zeigt in der Tat jetzt Änderungen. Also in der OpenDTU-OnBattery ist jetzt "Neue Version verfügbar" zu sehen.

Obwohl ich glaube GIT recht gut zu beherrschen, weiß ich leider nicht, warum der HEAD des github Repos auf Branch `development` steht. Vielleicht, weil `development` inzwischen der Default-Branch ist.

Scheinbar kann man keinen abweichenden Default-Branch für Pull Requests einstellen.

@helgeerbe Vielleicht sollte `fetchUrl` in `webapp/src/views/SytemInfoView.vue` nicht mit `HEAD` vergleichen, sondern mit `master`? Das scheint nach manuellem Aufruf der URL ein funktionierender Ansatz zu sein. Aus allem, was in `master` landet, wird automatisch ein Release gebaut, korrekt? Andernfalls müsste man mal schauen, ob man nicht die Releases per API abfragen kann und daraufhin entscheidet, ob tatsächlich eine neue Version verfügbar ist.

peff74, 2023-08-08:
ich habe den Fall dann man wieder eröffnet.
Heute erscheint bei mir auch wieder: Neue Version verfügbar!

helgeerbe, 2023-08-08:
Den Abschnitt in der OpenDTU hatte ich mir noch nie angeschaut. Aber es macht Sinn, auf die entprechende Branches zu gehen. Eventuell erweiter ich es so, dass der installierte Branch angezeigt wird (z. B. "master" oder "development"). Dann kann man sehen, ob es weitere commits auf auf dem jeweiligen Branch gibt.

Eine Release wird nur dann automatisch erzeugt, wenn die Release Version als Git Tag angelegt wird.

commit 5a72d749 (2023-08-09): fix #362 system Info checks git head, instead of build branch (#374)
closed 2023-08-09 by helgeerbe