hoylabs/OpenDTU-OnBattery #364: Shelly PRO 3EM lässt sich nicht auslesen

Jens869, 2023-08-07:
### What happened?

ich kann den Shelly PRO 3EM nicht auslesen

der richtige Link ist bei mir:
http://shellypro3em-0cb815fd1c64.fritz.box/rpc/Shelly.GetStatus

 em:0  total_act_power
"em:0":"total_act_power" 
{"em:0":"total_act_power" } 

???

### To Reproduce Bug

?

### Expected Behavior

?

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

c950eb7

Jens869, 2023-08-07:
[image]

helgeerbe, 2023-08-07:
Schau mal hier: https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/247#discussioncomment-6353999

Jens869, 2023-08-07:
es geht:

http://shellypro3em-####.fritz.box/rpc/Shelly.GetStatus

/em:0/total_act_power

[image]

closed 2023-08-08 by helgeerbe

Omermann, 2023-10-13:
(der richtige Link ist bei mir:) 
Link wo ist es

Wie kann ich es finden? Der DTU-Bildschirm ist gelb und funktioniert nicht.

Omermann, 2023-10-13:
[image]

Jens869, 2023-10-13:
http://< IP Nummer des Shelly Pro EM>/rpc/Shelly.GetStatus

oder

[http://shellypro3em-<Zufallszahl des Shelly>.< Hostname des DNS Servers>/rpc/Shelly.GetStatus

Omermann, 2023-10-13:
http://192.168.1.69/rpc/Shelly.GetStatus
/em:0/total_act_power
Error(http://192.168.1.69/rpc/Shelly.GetStatus): connection refused
Ich habe jede Möglichkeit ausprobiert, aber es funktioniert nicht (Hostname des DNS Servers), ich konnte kein Video finden, das erklärt, wie man es macht.
ich hab 4 panel, 150/45 Victron MPPT,HM 1500 und /OpenDTU-OnBattery , shelly3em. Es gibt
MQTT ?  Muss das sein?
Tasmota ?   Muss das sein?
/
[image]

Bad HTTP code: 401

Jens869, 2023-10-14:
kannst du die Adresse http://192.168.1.69/rpc/Shelly.GetStatus im Chrome/Safari/Internet Explorer/Firefox aufrufen?
wenn Nein, dann hat du die falsche IP gewählt

Omermann, 2023-10-14:
[image]
[image]
Es gibt keine Chance, falsch zu liegen. Wenn es für Sie kein Problem ist, können Sie sich mit TeamViewer verbinden und einen Blick darauf werfen

Omermann, 2023-10-14:
IR Reader ?  Tasmota(ESP8266) Muss das sein?

Omermann, 2023-10-14:
http://192.168.1.69/Status 
/em:0/total_act_power
Als dieser Link verloren ging, tauchte dies auf
Wo ist der Fehler?

{"wifi_sta":{"connected":true,"ssid":"demirarslan","ip":"192.168.1.69","rssi":-63},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"15:16","unixtime":1697289409,"serial":25,"has_update":false,"mac":"485519D99FA9","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":71.54,"pf":0.62,"current":0.49,"voltage":232.79,"is_valid":true,"total":373.0,"total_returned":9.3},{"power":252.91,"pf":0.66,"current":1.61,"voltage":233.84,"is_valid":true,"total":3152.0,"total_returned":0.0},{"power":0.00,"pf":0.09,"current":0.02,"voltage":235.56,"is_valid":true,"total":557.2,"total_returned":3.3}],"total_power":324.45,"emeter_n":{"current":0.00,"ixsum":1.81,"mismatch":false,"is_valid":false},"fs_mounted":true,"v_data":1,"ct_calst":0,"update":{"status":"idle","has_update":false,"new_version":"20230913-114244/v1.14.0-gcb84623","old_version":"20230913-114244/v1.14.0-gcb84623"},"ram_total":49920,"ram_free":31248,"fs_size":233681,"fs_free":154867,"uptime":3044}

Jens869, 2023-10-14:
ich sprach vom "Shelly PRO 3EM" 
Link ist http://<IP>/rpc/Shelly.GetStatus

und du hast aber ein "Shelly 3EM"  
Link ist http://192.168.1.69/Status

du musst nach total Power suchen, kann ich jetzt nicht genau sagen