hoylabs/OpenDTU-OnBattery #369: [Request] Update/consolidate infos for Huawei AC charger

eu-gh, 2023-08-09:
### Describe the solution you'd like

Hi and thank you for this awesome project!

I bought a Huawei R4850G2 power supply for charging my Pylontech US2000C battery and stumbled upon some small problems:

- I bought a small SN65HVD230 CAN shield, but according to the pinout configuration in [platformio](https://github.com/helgeerbe/OpenDTU-OnBattery/blob/development/platformio.ini), this will only connect to the Pylontech CAN. For setting the Huawei unit, it needs to be a whole seperate CAN controller (which I ordered now and is called MCP2515/TIA1050)
- According [this picture](https://github.com/helgeerbe/OpenDTU-OnBattery/blob/development/docs/Wiring_ESP32_Symbol.png) inside the docs folder, another "Huawei slot detect" PCB and a 5V supply mod on the CAN shield is needed. Is this still necessary and, if so, which purpose does it serve?
- Setting the Huawei offline (=start) values [now seems possible](https://github.com/helgeerbe/OpenDTU-OnBattery/pull/315), which is nice. According to some comments I found, the unit always starts with maximum voltage/current, so maybe it would be a good practice to set the offline values to a save level (or even close to 0), until the online values are updated?
- The [wiki page](https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Huawei-AC-PSU) for this topic is still empty, I would like to change that, but only with verified information.

Best regards!

MalteSchm, 2023-08-13:
The CAN details and the slot detect info is described here: 
https://github.com/helgeerbe/OpenDTU-OnBattery/blob/development/README_onBattery.md
https://github.com/helgeerbe/OpenDTU-OnBattery/blob/development/docs/hardware_flash.md

Feel free to move the information over to the wiki. The wiki is a recent addition to the project

eu-gh, 2023-09-18:
Thank you! I added my share of knowledge to the [wiki page](https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Huawei-AC-PSU)

closed 2023-09-18 by eu-gh