hoylabs/OpenDTU-OnBattery #375: Bei upgrade des .bin Files Version 2023.08.09 wird in der System Info Seite keine Version angezeigt

XpertXXL, 2023-08-09:
### What happened?

hat seither immer einwandfrei funktioniert

### To Reproduce Bug

bei Update via OTA funktioniert die Anzeige und Abfrage der Firmware version nicht mehr.
Mit Visual Studio Code funktioniert das Ganze wie gehabt

### Expected Behavior

korrekte Anzeige der Version und Hinweis auf Neuerungen

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

2023.08.09 

### Anything else?

[image]

alexz707, 2023-08-09:
kann ich bestätigen

schlimmchen, 2023-08-09:
Strg + Shift +F5?

helgeerbe, 2023-08-09:
Danke für die Hinweise. War tatsächlich dran.
Ich muss mir die Actions anschauen. Eigentlich ermittelt ein Python Skript den Hash und den Branch und übergibt das als Define an den Compiler. 

Sieht so aus, als ob das Skript nicht ausgeführt wird.

helgeerbe, 2023-08-10:
Bin gerade etwas ratlos.

Leider sind die Actions schwer zu debuggen. Aktueller Stand der 2023.08.10.

- locales build funktioniert
````log
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: v23.8.8-18-g4873041
Firmware build on branch: master
````
- Action build auf Branch, der noch nicht das Release Tag hat funktioniert ebenfalls

````log
Firmware Revision: g4873041
Firmware build on branch: master
````

- setze ich aber das Release Tag z. B. 2023.08.10 dann geht es nicht
````log
Unexpected err=IndexError('list index out of range'), type(err)=<class 'IndexError'>
Firmware Revision: g4873041
Firmware build on branch: 
````

Die Doku sagt:
````TXT
dulwich.porcelain.active_branch(repo)
Return the active branch in the repository, if any.

Parameters
repo – Repository to open

Returns
branch name

Raises
[KeyError](https://docs.python.org/3/library/exceptions.html#KeyError) – if the repository does not have a working tree

[IndexError](https://docs.python.org/3/library/exceptions.html#IndexError) – if HEAD is floating
````

Warum ist der HEAD floating, wenn ein Tag zum commit gepusht wird? Zumindest bei einer Action. Lokal geht es ja.

Als Workaround könnte annehmen, dass die Basis als default immer der master branch ist?

Ideen oder Vorschläge?

helgeerbe, 2023-08-10:
OK, with 2023.08.10.post1 habe ich den default Branch auf master gesetzt. Kann dass jemand mal testen und feedback geben?

XpertXXL, 2023-08-10:
sieht gut aus :-)

[image]

Manos1966, 2023-08-13:
> sieht gut aus :-)
> 

nicht bei mir.... :-(
oder muss ich einen kompleten NEustart der OpenDTU machen?
[image]

helgeerbe, 2023-08-30:
Hi @Manos1966 sorry für die späte Antwort (ich habe mir mal 2 Wochen Urlaub gegönnt).

Bei dir fehlt die Zeile mit dem "Firmware Branch", daher vermute ich, dass du noch mal die Oberfläche laden musst. Der alte Code ist noch im Browser Cache.