hoylabs/OpenDTU-OnBattery #376: [Request] Add calculated output (battery) power & efficiency to mqtt

alexz707, 2023-08-09:
### Is your feature request related to a problem? Please describe.

I want to work (calc SOC) with the ouput power of the victron because I don't have access to my BMS.
Ofc I could calculate the power in HA but as the value gets already calculated on openDTU I thought it would be nice to also publish that value(s) over mqtt.

### Describe the solution you'd like

Publish calculated battery power and efficiency over mqtt

### Describe alternatives you've considered

Calcualte it in home assistant.

### Additional context

BTW: Great work! appreciate it! :)

schlimmchen, 2023-08-24:
Please specify in more detail what you want. What value in what unit should be published to what topic?

spcqike, 2023-08-24:
if i understand correctly he wants to calculate the SoC in HA based on victron charging power and hoymiles discharge power.

this would be possible in firmware, too. but i see several disadvantages as the systems only query the data once in a while. for static discharging, that wont be a problem. but IMO the charging current wont be reported frequent enough to get reliable SoC values.

smartshunts and BMS have dedicated ICs that only monitor current/voltage/power and calculate energyflow and SoC based on that, with several measurements and updates per second.

@alexz707 i guess your best (or more reliable) option would be a dedicated smartshunt, either a victron original or a [DIY ](https://github.com/kscholty/SmartShuntINA226)one

schlimmchen, 2023-08-24:
> if i understand correctly he wants to calculate the SoC in HA based on victron charging power and hoymiles discharge power.

IMHO that's futile and a waste of time.

Still, I think it's valid wanting the MPPT charging power to be available through MQTT, if that's what is asked. For that reason I wanted a list that specifies what is wanted, regardless of whether or not it makes sense do to the calculations that we are talking about.

swingstate, 2023-08-24:
my 2cents, using a dedicated shunt is much more convenient and integration into HA is very simple with an ESP/ESPhome and https://github.com/KinDR007/VictronMPPT-ESPHOME. I'D recommend taking that route. 

However, @schlimmchen enabling onBattery to receive the SOC information that HA has available (through whatever integration, MPPT-ESPHome is just one example) through MQTT would be nice, as long as onBattery can use that information similar to the Pylontech CAN SOC readings.

ele-dev, 2023-08-30:
Yes I would also very much appreciate a feature for openDTU on Battery to fetch SOC values over mqtt or http from any external source. In my special situation: I actually have a Pylontech Battery but for some reason that can't be explained, it's not sending out these valueable information from the internal BMS over the CAN port like it should and therefore I am not able to discharge the battery based on the actual SOC.
For all custom battery setups with other BMS than Pylontech it would be a very useful feature too if they could feed the SOC somehow into open dtu on Battery. For example over MQTT and ESPHome

schlimmchen, 2023-08-30:
See #293 (MQTT as Battery Data Provider).

alexz707, 2023-09-11:
Sorry for the late reply, I was on vacation :)

I was already thinking of buying a smart shunt for this but I was waiting on the JK BMS branch to be merged to have a look at it. I will use 2 BMS for 2 batteries in parallel so I'm not sure if openDTU can handle 2 jk bms's.

Currently with 1 battery I'm calculating the SOC in HA myself with the voltage and current values published by openDTU from the victron data. I saw that the power is already displayed in the GUI. So I thought it would be a good idea to just publish the output power as it is already calculated and I wouldn't need to do that in HA.
I know it's not accurate but since it is in the UI it why not just publish that value?

I would like to avoid to have severall esps running just for the battery, I would prefer to only have one esp which does everything. That's also a point which kept me from buying the smartshunt.

Is it already possible to get the data of a connected smart shunt out of the victron mppt directly?

schlimmchen, 2023-09-22:
> I'm not sure if openDTU can handle 2 jk bms's

Nope.

> since it is in the UI it why not just publish that value?

Yup, that's doable and easy.

> Is it already possible to get the data of a connected smart shunt out of the victron mppt directly?

No, but see #452: You can connect to a SmartShunt directly using a separate serial connection.

PR #475 (2023-10-04): Publish calculated MPPT metrics
The added values P, IPV, and E, are calculated by OpenDTU-OnBatery and part of the web application live view, but were previously not published through MQTT.

Closes #376.
commit f7abbdbe (2023-10-04): Publish calculated MPPT metrics (#475)

* VE.Direct MPPT MQTT: remove trailing whitespace

* VE.Direct MPPT MQTT: publish P, IPV and E to MQTT

those values are calculated by OpenDTU-OnBatery and are part of the web
application live view, but were previously not published through MQTT.

closes #376.
closed 2023-10-04 by helgeerbe