hoylabs/OpenDTU-OnBattery #378: MQTT victron topic contains not allowed characters for HA

alexz707, 2023-08-10:
### What happened?

HA complains about the name of the topic "mppt_day_sequence_number_(0...364)".
Only following chars are allowed according to https://www.home-assistant.io/integrations/mqtt/#discovery-topic

[a-zA-Z0-9_-]

### To Reproduce Bug

Just have a look into HA logs:

`Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:114
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 14:43:00 (3 occurrences)
Last logged: 18:20:21
Received message on illegal discovery topic 'homeassistant/sensor/dtu_victron_123/mppt_day_sequence_number_(0...364)/config'. The topic contains not allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic`

### Expected Behavior

Value of mppt_day_sequence_number is displayed in HA

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

Version of yesterday where the git hash is not displayed

madmartin, 2023-08-21:
I will investigate this and provide a PR

madmartin, 2023-08-22:
see #394

closed 2023-08-28 by helgeerbe